Kaldek/rat-ratgdo #1: great work lol ive done the same. what new in the new board?

tylermeredyk, 2023-10-25:
[image]

Kaldek, 2023-10-25:
Looks like you're using the dry contact setup, which this project doesn't cover.    What component are you using specificially for the connection between D4 and the red wire other than the 10k resistor?  Have you built and tested this?

tylermeredyk, 2023-10-25:
This is just a trace by trace copy. And all the smd labels are as they are on the board. The mosfet between the 10k labeled xoyl 4n. Is most likey a alphaomega, maybe a a03400a but I’m not 100%

tylermeredyk, 2023-10-25:
And 30$ for something like this, given the components and board design is a bit insane. Don’t you think?

CrazyCoder, 2023-10-25:
What would be the minimal schematics for the dry contact configuration mode like supported in 2.5 version? I have a stupid Genie opener and would like to DIY the board to control it. Thanks for sharing the information!

rlowens, 2023-10-25:
> What would be the minimal schematics for the dry contact configuration mode like supported in 2.5 version? I have a stupid Genie opener and would like to DIY the board to control it. Thanks for sharing the information!

I can't find the firmware source for the v2.5 Dry Contact firmware, and we haven't seen what the board looks like to trace any changes that help with the Dry Contact wiring method.  But the [wiring diagram for the v2.5 Dry Contact method](https://user-images.githubusercontent.com/4663918/277838851-e338c3bf-4eda-447a-9e79-737aa1a622a0.png) just uses the D5 Open and D6 Close dry contact inputs as limit switches, and they just have an SL 1N5819W schottky barrier diode between input and GPIO (which you can omit if you want, they are just there for extra protection). Note the warning to "Only connect obstruction sensors to ratgdio if their operating voltage is 5-7vdc!" And the [ratgdo models page](https://github.com/PaulWieland/ratgdo/wiki) mentions specifically "Genie | H8000D | Dry Contact |   | x | user accessible limit switch terminals, no obstruction sensing, voltage too high".

So you would wire up the Red Wire/Control and White Wire/Ground the same as for Security+ 1.0/2.0, wire the limit switches to D5/D6, and not connect the obstruction sensors. Then flash with the Dry Contact firmware and you should be good to go.

Let us know how it goes.

CrazyCoder, 2023-10-25:
Thanks! I also found https://github.com/PaulWieland/ratgdo/issues/145 with the discussion how to connect the obstruction sensor via the voltage divider. I'll have to check with the oscillograph how they work for my old Genie model. If they send pulses that are detected by the current algorithm, it should be possible to make them work.

There is 2.5 firmware for the esphome-ratgdo variant that I plan to use: https://github.com/ratgdo/esphome-ratgdo/tree/main/static, specifically this one:

https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp32_d1_mini.yaml

Kaldek, 2023-10-25:
> And 30$ for something like this, given the components and board design is a bit insane. Don’t you think? 

No, the price is of no concern to me. I'm also developing a project based on an Arduino which I'll charge more than that.  I don't mind if I get copied but would hope to pay off the effort spent on the research and development.  The issue that got me started on this is the time to delivery of ratgdo and the shipping costs.

CrazyCoder, 2023-10-25:
> The issue that got me started on this is the time to delivery of ratgdo and the shipping costs.

Same here. I'd also need 2 boards for 2 garage doors and I already have all the components needed for DIY version. myQ stopped working with HA, so I need a working solution ASAP. Can't wait a month for the boards to be delivered.

Kaldek, 2023-10-25:
>  [image]

I have to ask again if you built one or have just created schematics.  The 10k resistor in series to the gate of AO3400A is off to me.  I've built similar circuits using Arduino and always use the 10k as a pulldown so there is no risk of a floating voltage tripping the MOSFET.  10k in series is usually for control of inrush current which I just don't see being an issue here.

tmeredyk, 2023-10-25:
im still just prototyping, i haven't breadboarded anything. im pretty sure its inverted in code but the gpio2 pin needs to be pulled high during boot or it will fail on the esp-12f. for the 10k maybe its not necessary. but thanks for the critique

Kaldek, 2023-10-25:
> im still just prototyping, i haven't breadboarded anything. im pretty sure its inverted in code but the gpio2 pin needs to be pulled high during boot or it will fail on the esp-12f. for the 10k maybe its not necessary. but thanks for the critique

No that's fine.  We should clarify then that my current schematic assumes use of a D1 mini rather than a raw ESP8266.

tmeredyk, 2023-10-25:
all good, i was just using the default pins on the ratdgo 2.0, if you'd like me to change them for the 2.5, it's no problem. its just i haven't seen the latest pcb. not sure what's changed. happy to add to this project

CrazyCoder, 2023-10-25:
Some findings. Dry contact GDO support is currently only in this branch of ratgdo: https://github.com/PaulWieland/ratgdo/tree/2.0-secplus1.0.

It's not available in the ratgdo-esphome yet. Related issue: https://github.com/ratgdo/esphome-ratgdo/issues/62.

My Genie obstruction sensors are ~12V. This post may be helpful regarding how they work: https://forum.allaboutcircuits.com/threads/genie-garage-door-sensors-used-elsewhere.59838/#post-402692:

> The way they work is the receiver adds a very narrow pulse train to the 12 V. line when it is detecting the emitter. The circuitry inside the door opener detects this signal.

Will verify it with the oscilloscope shortly.

Kaldek, 2023-10-26:
> all good, i was just using the default pins on the ratdgo 2.0, if you'd like me to change them for the 2.5, it's no problem. its just i haven't seen the latest pcb. not sure what's changed. happy to add to this project

The only change from a circuitry perspective was the change to D1 (GPIO5) for TX away from D4 (GPIO2).

CrazyCoder, 2023-10-26:
Genie obstruction detection pulse when not obstructed (HIGH when obstructed):

[image]

Voltage goes up to 12.5V (12.7V max). It looks like voltage divider will be needed to get it compatible with the current board version.

According to the code:

> // If at least 3 low pulses are counted within 50ms, the door is awake, not obstructed and we don't have to check anything else

There should be close to 20 low pulses in 50ms with Genie, so it might work.

rlowens, 2023-10-26:
Looking thru the issues in ratgdo I noticed a link to https://github.com/mariusmuja/esphome-ratgdo where the have this schematic:
https://github.com/mariusmuja/esphome-ratgdo/blob/main/static/gdo_schematic.png

So, that's a comparison if needed.

Not sure if @mariusmuja has any insight into how well his schematic is working?

mariusmuja, 2023-10-26:
> Not sure if @mariusmuja has any insight into how well his schematic is working?

As I mentioned on the original issue, this schematic is working fine for me (tested on 2 security+ 2.0 GDOs). I don't have a ratgdo board yet (one is on its way).

Kaldek, 2023-10-26:
> Looking thru the issues in ratgdo I noticed [a link](https://github.com/ratgdo/esphome-ratgdo/issues/46#issuecomment-1721584172) to https://github.com/mariusmuja/esphome-ratgdo where they have this schematic: https://github.com/mariusmuja/esphome-ratgdo/blob/main/static/gdo_schematic.png [image]

I note that @mariusmuja isn't using any form of pulldown resistor or series resistor on the TX GPIO.  I'm curious about that and also why the 150 ohm resistor on the RX GPIO?

mariusmuja, 2023-10-26:
> I note that @mariusmuja isn't using any form of pulldown resistor or series resistor on the TX GPIO. I'm curious about that and also why the 150 ohm resistor on the RX GPIO?

I put it there to limit the current sourced by the GPIO when the MOSFET is open.

Kaldek, 2023-10-26:
> > I note that @mariusmuja isn't using any form of pulldown resistor or series resistor on the TX GPIO. I'm curious about that and also why the 150 ohm resistor on the RX GPIO?
> 
> I put it there to limit the current sourced by the GPIO when the MOSFET is open.

Paul Wieland doesn't do this (and I didn't either).  I don't think it is needed here, because that pin is in INPUT_PULLUP mode and is designed to be pulled low by, say, a switch (or in this case, a MOSFET).

mariusmuja, 2023-10-26:
> Paul Wieland doesn't do this (and I didn't either). I don't think it is needed here, because that pin is in INPUT_PULLUP mode and is designed to be pulled low by, say, a switch (or in this case, a MOSFET).

Yes, I think that's correct.

Kaldek, 2023-10-26:
> > Paul Wieland doesn't do this (and I didn't either). I don't think it is needed here, because that pin is in INPUT_PULLUP mode and is designed to be pulled low by, say, a switch (or in this case, a MOSFET).
> 
> Yes, I think that's correct.

OK, I'm not trying to ask that you change anything.  My main goal here is to make sure I update my own schematic if there are egregious errors in it.

gcormier, 2023-10-30:
Here's a design I have abandoned. The goal was to siphon power off the remote control lines, but there is just not enough power on those lines really.

It might be possible with a coin cell or something, so maybe later in the winter I will give it another shot.

Regarding the resistors
1. A gate resistor is sometimes used in case the MOSFET fails - it can prevent the microcontroller exploding upstream.
2. It could be for BOM optimization. The value is not important. In this case if a 10k is used elsewhere, it might be fine to use that on the MOSFET as well if the rise time isn't very demanding.

[image]

mariusmuja, 2023-10-30:
It's easy to bring a 3.3V wire out from inside the GDO, just unscrew the side with the wire contacts, unplug a couple connectors and take the board out. On some models 3.3V is marked on the board, otherwise, look for the voltage regulator chip & measure to confirm.

gcormier, 2023-10-30:
Sure, but it's even easier to just connect the 2 wires from your remote control :)

akohlsmith, 2023-11-01:
> And 30$ for something like this, given the components and board design is a bit insane. Don’t you think? [image]

It's not insane at all.

You can't look at BOM cost alone and derive an appropriate price point. He's created something new, made a PCB, assembled, tested and packaged for shipping, not to mention developed the software and provide an amount of support, not just for those who've bought one.  In that context, $30 for that isn't too much to ask at all. If you can DIY it then by all means, go for it.

That said, I'm happy that this was published because Mr. Wieland's obviously backlogged and having to wait an indeterminate amount of time to receive a pre-assembled board and ESP module that I could build myself in the span of an hour is where my issue is, not with his price point.

tmeredyk, 2023-11-01:
I’ve made a board, $7.50 dollars for 3 pcbs. bom is under $5 per board. Given it doesn’t have screw terminals, not pre-assembled but includes the esp. price would only go down at scale. Sure packing and shipping would also go into the price. I just think somebody trying to profit 15 or so dollars per board with essentially two mosfets is alittle much. Why are the schematics so secret in the first place?

Kaldek, 2023-11-01:
> I’ve made a board, $7.50 dollars for 3 pcbs. bom is under $5 per board. Given it doesn’t have screw terminals, not pre-assembled but includes the esp. price would only go down at scale. Sure packing [...]

I don't argue with his prices as it's worth recouping your R&D costs.

jgosnell56, 2023-11-06:
I am seeing an issue and wondered if anyone saw this or what could be causing my issue. When I use an external power source for my Ratgdo, everything works properly. When I pull the 3.3v source from the board following the diagram here (very bottom) [https://paulwieland.github.io/ratgdo/03_wiring_soldered.html](https://github.com/ratgdo/esphome-ratgdo/issues/url), my remotes drastically lose distance. They are pretty much useless unless I am standing inside the garage when pressing a button. I assume I am drawing too much power from the board causing the remote receiver to be underpowered. Is there another 3.3v source on the board I could solder to and try?

tmeredyk, 2023-11-06:
i haven't tried this but i would try and add ur own regulator off of the 12v rail

Kaldek, 2023-11-06:
> I am seeing an issue and wondered if anyone saw this or what could be causing my issue. When I use an external power source for my Ratgdo, everything works properly. When I pull the 3.3v source from [...]

Yeah don't pull power from the garage door opener's board.  Either use an external supply or - if your GDO has it - the battery backup wires. These are 12v high power.  We have notes here in this repo on using a voltage regulator for this.

rlowens, 2023-11-12:
FYI: Paul has switched to an integrated ESP12F-based board for his new ratgdo v2.5i. It is "functionally the same as v2.5", but this is the first shot I've seen of the front of a v2.5+ board, if we want to try to understand what changed from v2.0.
[image]

rlowens, 2023-11-12:
I also found this shot of the ratgdo v2.5 without a D1 mini installed. I think I can confirm it has the same number of transistors and diodes as ratgdo v2.0 but 3 more resistors (2 more on the left and 1 more on the right):
[image]

My guess is that this is one more resistor for each of the AO3400A MOSFETs, probably a pull-down resistor as @Kaldek added, in addition to the series-resistor Paul already had on each of those.

foreverimagining, 2023-11-12:
If it helps any, I'm expecting my v2.5 to come in tomorrow (as long as USPS doesn't botch it again, since it's been in the area since 11/09). I was going to offer to provide any information/testing-I'm-capable-of you guys might be needing before I went ahead and installed it. I'm planning to make the second one I'll need for various reasons, and I'd likely rather base it off v2.5.

I can already tell you, from the picture of the v2.5i, that he's going to cheese off the 3D printing community again, though.

tmeredyk, 2023-11-12:
[image]

rlowens, 2023-11-12:
> offer to provide any information/testing-I'm-capable-of you guys might be needing before I went ahead and installed it. I'm planning to make the second one I'll need for various reasons, and I'd likely rather base it off v2.5.

That would be great. We'll need clear shots of the front and back without D1 mini installed, then probably some tracing with a continuity tester to confirm what connects to what.

foreverimagining, 2023-11-12:
> We'll need clear shots of the front and back without D1 mini installed, then probably some tracing with a continuity tester to confirm what connects to what.

I think I can manage that. Would you prefer I start a new issue with the pictures to make things a little more organized?

rlowens, 2023-11-12:
> Would you prefer I start a new issue with the pictures to make things a little more organized?

That sounds good.

Kaldek, 2023-11-12:
Re-opening for public discussion visibility.

Kaldek, 2023-11-12:
> FYI: Paul has switched to an integrated ESP12F-based board for his new ratgdo v2.5i. It is "functionally the same as v2.5", but this is the first shot I've seen of the front of a v2.5+ board, if we want to try to understand what changed from v2.0. [image]

If Paul is building a complete unit including CH340 USB-to-serial and power regulation, it likely means expectation of sales volume and the reduced wholesale prices that come with that.

Kaldek, 2023-11-28:
Closing due to multiple pull requests which have updated all of our schematics based on information gathered from other issues.

closed 2023-11-28 by Kaldek