Kaldek/rat-ratgdo #3: Setup with ESP WROOM 32

rohinrohin, 2023-10-26:
Hey @Kaldek! 

Great job on this repo and for constantly updating the instructions! I am in process of setting this up and would appreciate your help on this. 

I have a [ESP WROOM 32](https://www.amazon.com/ESP-WROOM-32-Development-Microcontroller-Integrated-Compatible/dp/B08D5ZD528?th=1), pinout [here](https://m.media-amazon.com/images/I/51FzgrVl-cL._AC_.jpg). (sorry about the amazon links)

I was wondering if there were any nuances that would make my installation steps a bit different?

I noticed on the pinout of this board vs the D1 mini
 - GPIO4 (SDA/I2C) in D1 Mini is similar to GPIO21 (SDA/I2C)
 - GPIO13 (MOSI) in ESP8266/ESP WROOM is the same

I was curious if I was on the right track with the pinout mapping and how this changes things. Also side note, I was not able to find the IRL mosfet easily, have ordered it but I am using the 2n7000 for now. 

Thanks and looking forward to your response!

donavanbecker, 2023-10-26:
@rohinrohin, I have the same board but with a usb-c. Can you share your setup once you have it complete?

rohinrohin, 2023-10-26:
Absolutely! Will do :)

rlowens, 2023-10-26:
ratgdo uses these pins for esp32 https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp32_d1_mini.yaml

  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7

Which looking at the D1 Mini ESP32 pin mapping at https://templates.blakadder.com/wemos_D1_Mini_ESP32.html translates to:

  uart_tx_pin: D1=GPIO22
  uart_rx_pin: D2=GPIO21
  input_obst_pin: D7=GPIO23

So, that's the mapping that it would default to expecting if you flash the stock ratgdo firmwares, but you can use pretty much any that you want if you edit the ESPHome .yaml after importing it.

rohinrohin, 2023-10-26:
Understood! Appreciate the quick and detailed response @rlowens. Will try it and keep you posted.

Kaldek, 2023-10-26:
> I have a [ESP WROOM 32](https://www.amazon.com/ESP-WROOM-32-Development-Microcontroller-Integrated-Compatible/dp/B08D5ZD528?th=1), pinout [...]

As this is using ESPHome, there are only two things I would do:

1. Validate that the ratgdo ESPHome installer will work on the ESP32 like it does for generic ESPHome installer
2. If #1 is confirmed, install ESPHome-ratgdo and map your pins to GPIO4, GPIO5, and GPIO13, as the code uses those pin identifiers internally.

Kaldek, 2023-10-26:
This issue is held open until testing has been performed by @rohinrohin and we have closure on any follow-up work.

rohinrohin, 2023-10-26:
thanks for the info @Kaldek! 

Here are some interim updates from my end:

1. The web installer (blue v2.5) throws up an error saying that the board is not supported. Do you know of an approach where I can compile and install the firmware to my board without using Paul’s web installer?
2. I tried flashing the older version 2.0 ESP32 and that seems to have flashed properly. Would this work with our existing schema?

Kaldek, 2023-10-26:
> 2. I tried flashing the older version 2.0 ESP32 and that seems to have flashed properly. Would this work with our existing schema?

Yes v2.0 will work the only difference is that TX is on GPIO2 on v2.0.

rlowens, 2023-10-26:
https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_d1_mini.yaml uses
uart_tx_pin: D4

Which[ on D1 mini ESP32](https://templates.blakadder.com/wemos_D1_Mini_ESP32.html) is GPIO16

sanjay900, 2023-10-26:
I tried this myself, and i was able to control my garage door but not receive its state, any idea what might cause that?

vaidehi-joshi, 2023-10-26:
Hey @sanjay900! Can you let me know which version you tried flashing onto the ESP WROOM 32? Was it v2 or v2.5? 
Also which pins did you use for tx,rx and input_obst_pin?

sanjay900, 2023-10-26:
I used `v25board_esp32_d1_mini.yaml`
Seems to be fully working now!
As for pins, 
```
uart_tx_pin: D1=GPIO22
uart_rx_pin: D2=GPIO21
input_obst_pin: D7=GPIO23
```

Kaldek, 2023-10-26:
Beautiful work, folks!

Kaldek, 2023-10-26:
> I tried this myself, and i was able to control my garage door but not receive its state, any idea what might cause that?

I know you fixed your isuse, but I also had this issue yesterday and it was because I had used a 2n7000 MOSFET that had previously been almost cooked by some of my stupid wiring efforts.  Overnight it eventually died and did the good old "MOSFET shorts to ground" failure. Whoops!

sanjay900, 2023-10-27:
> I know you fixed your isuse, but I also had this issue yesterday and it was because I had used a 2n7000 MOSFET that had previously been almost cooked by some of my stupid wiring efforts. Overnight [...]

I did think this might have been the case and replaced it a few times haha

Kaldek, 2023-10-27:
Hey folks once @rohinrohin has also confirmed theirs as working, someone just write a summary of what was needed for the ESP32 and I'll add that to the project in the FAQ or maybe a separate "supported ESP modules" document.

sanjay900, 2023-10-27:
> Hey folks once @rohinrohin has also confirmed theirs as working, someone just write a summary of what was needed for the ESP32 and I'll add that to the project in the FAQ or maybe a separate "supported ESP modules" document.

Summary wise, i think all we need is a diagram that references the correct GPIO pins
```
uart_tx_pin: D1=GPIO22
uart_rx_pin: D2=GPIO21
input_obst_pin: D7=GPIO23
```

Hopefully the web flasher will be updated to just directly allow programming the `v25board_esp32_d1_mini.yaml`

Kaldek, 2023-10-27:
> Summary wise, i think all we need is a diagram that references the correct GPIO pins
>
> ```
> uart_tx_pin: D1=GPIO22
> uart_rx_pin: D2=GPIO21
> input_obst_pin: D7=GPIO23
> ```

I have some images here of the ESP-WROOM-32 and I can't see pins labelled D7 or D1.  

https://github.com/Kaldek/rat-ratgdo/blob/main/ESP-WROOM-32%20board.jpg

Kaldek, 2023-10-27:
@sanjay900  also can you explain this process in more detail including how you told it to use a specific YAML file?

> I programmed on v2 using the web programmer, but then changed it to v2.5 from ESPHOME as the web flasher didn't let me program v2.5

Kaldek, 2023-10-27:
I have created a "Supported Boards" page with lots of details.  Just waiting on that information about updating to v2.5 from within ESPHome @sanjay900

sanjay900, 2023-10-27:
> I have created a "Supported Boards" page with lots of details. Just waiting on that information about updating to v2.5 from within ESPHome @sanjay900

If you edit the yml from ESPHome, you can just change the packages line to the following
```yml
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp32_d1_mini.yaml@main
```

sanjay900, 2023-10-27:
> > > Hey folks once @rohinrohin has also confirmed theirs as working, someone just write a summary of what was needed for the ESP32 and I'll add that to the project in the FAQ or maybe a separate "supported ESP modules" document.
> > 
> > 
> > Summary wise, i think all we need is a diagram that references the correct GPIO pins
> > ```
> > uart_tx_pin: D1=GPIO22
> > uart_rx_pin: D2=GPIO21
> > input_obst_pin: D7=GPIO23
> > ```
> 
> I have some images here of the ESP-WROOM-32 and I can't see pins labelled D7 or D1.
> 
> https://github.com/Kaldek/rat-ratgdo/blob/main/ESP-WROOM-32%20board.jpg
> 
> They just appear to be labelled D21, D22, D23.

Sorry yeah, those were just the labels in the yml, since its using the d1 mini pinouts, which end up mapping to those specific GPIO pins

Kaldek, 2023-10-27:
> > I have created a "Supported Boards" page with lots of details. Just waiting on that information about updating to v2.5 from within ESPHome @sanjay900
> 
> If you edit the yml from ESPHome, you can just change the packages line to the following
> 
> ```yaml
> packages:
>   ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp32_d1_mini.yaml@main
> ```

Sorry mate, I was specifically referring to the complete process.  That is, how to force it to use the specific YAML.  I'm really vague on this, sorry.  I know it's to do with the ESP web installer and the manifest files that it uses, but not much more than that.

sanjay900, 2023-10-27:
> Sorry mate, I was specifically referring to the complete process. That is, how to force it to use the specific YAML. I'm really vague on this, sorry. I know it's to do with the ESP web installer and [...]

Yeah sorry I have no idea how to change the web installer, I just modified the yaml of it manually after programming V2.

Kaldek, 2023-10-27:
> Yeah sorry I have no idea how to change the web installer, I just modified the yaml of it manually after programming V2.

Yes, but ummm, how?  Frome within Home Assistant?  From the ratgdo web interface?

Kaldek, 2023-10-27:
I just realised I don't have the ESPHome Add-On installed in Home Assistant.

sanjay900, 2023-10-27:
Right yeah, so adopt it into ESPHome, then hit the edit button.
[image]

Kaldek, 2023-10-27:
> Right yeah, so adopt it into ESPHome, then hit the edit button. [image]

And then click the install button, right?

sanjay900, 2023-10-27:
Yeah, click install

Kaldek, 2023-10-27:
> Yeah, click install

OK.  But cludgy I suppose but I can see the process now.  I had not installed ESPHome Add-On into HA, as HA has the discovery part of it already included.  All the management stuff is in the ESPHome Add-On.

sanjay900, 2023-10-27:
> > Yeah, click install
> 
> OK. But cludgy I suppose but I can see the process now. I had not installed ESPHome Add-On into HA, as HA has the discovery part of it already included. All the management stuff is in the ESPHome Add-On.

Hopefully someone will send the project an image and then they can add esp32 v2.5 support to the installer.

Kaldek, 2023-10-27:
> Hopefully someone will send the project an image and then they can add esp32 v2.5 support to the installer.

For the ESP-32 I can just tell people to use v2.0 and that pinout.  There's no specific we have to use v2.5 - at least not with what I'm trying to achieve.  Your thoughts?

rlowens, 2023-10-27:
> I have some images here of the ESP-WROOM-32 and I can't see pins labelled D7 or D1.
> 
> https://github.com/Kaldek/rat-ratgdo/blob/main/ESP-WROOM-32%20board.jpg
> 
> They just appear to be labelled D21, D22, D23.

The https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_d1_mini.yaml and https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp32_d1_mini.yaml use the "wemos_d1_mini32" board, [which looks like this](https://templates.blakadder.com/wemos_D1_Mini_ESP32.html) and is an expanded version of the ESP8266-based D1 mini board, made to be able to fit into boards that take a ESP8266 D1 mini board, but still offer the extra pins of the ESP32. So for the original D1 mini pins it allows use of the original pin names. All other ESP32-based boards I've seen just label the pins with the actual GPIO pin numbers.

> For the ESP-32 I can just tell people to use v2.0 and that pinout. There's no specific we have to use v2.5 - at least not with what I'm trying to achieve. Your thoughts?

I agree. And highlight that the ESPHome-firmware is the same between v2.0 hardward and v2.5 hardware, only the one pin is used differently.

Kaldek, 2023-10-27:
> I agree. And highlight that the ESPHome-firmware is the same between v2.0 hardward and v2.5 hardware, only the one pin is used differently.

What's the mapping of D4 in their YAML to an ESP-32 pin?  I've, uh, had a couple of beers already as it's Friday arvo here.

Kaldek, 2023-10-27:
Is it GPIO16?

sanjay900, 2023-10-27:
> Is it GPIO16?

Should be, I'll give it a go in a bit

Kaldek, 2023-10-27:
OK looks like it is indeed GPIO16
[image]

sanjay900, 2023-10-27:
Does seem like gpio16 works, annoyingly i took my breadboard apart to put it into a PCB, which then didnt work and after putting back on a breadboard im back to RX not working lol

sanjay900, 2023-10-27:
Oh, the [ESP32 Schematic](https://github.com/Kaldek/rat-ratgdo/blob/main/ratgdo%20open%20source%20ESP-32_schem_v1.png) you uploaded is wrong, youve flipped the source and drain.

After flipping that, it works perfectly with V2.0, and D16, D21 and D23

CrazyCoder, 2023-10-27:
@sanjay900 all the schematics that are currently uploaded (including the breadboard working version photo) say "source to ground and drain to RX". Are you sure it should be flipped? If so, why only ESP32 schematic is wrong and not all of them (since they all are the same)?

sanjay900, 2023-10-27:
> @sanjay900 all the schematics that are currently uploaded (including the breadboard working version photo) say "source to ground and drain to RX". Are you sure it should be flipped? If so, why only ESP32 schematic is wrong and not all of them (since they all are the same)?

oh your absolutly right, im just blind

vaidehi-joshi, 2023-10-29:
quick update on @rohinrohin 's project:
we have followed the circuit diagram mentioned [here](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v3.png) but the setup with our ESP8266 is still not working. We tried replacing transistors, tested the resistors, and even the continuity in the connections. So I am guessing we made an error in the circuit.  It's a little hard to follow through the connections between the 2 transistors and the control wires. Can someone please tell me where the Gate, Drain, and Source for each transistor are going? Appreciate any help!

vaidehi-joshi, 2023-10-29:
Please correct me if I am wrong:
IRLB8721:
Gate- D1
Source- Ground
Drain- Red control wire

2n 7000 n-channel:
Gate- Red control wire
Source- Ground
Drain- D2

Kaldek, 2023-10-29:
> Please correct me if I am wrong: IRLB8721: Gate- D1 Source- Ground Drain- Red control wire
> 
> 2n 7000 n-channel: Gate- Red control wire Source- Ground Drain- D2

Updated schematic with labels for Source/Gate/Drain:
https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v4.png

Kaldek, 2023-10-30:
@vaidehi-joshi I have updated the FAQ with extra info in case you are connecting to the wrong terminals on your GDO.  Some GDOs do not use the Red/White/Black colour scheme for their wires.

https://github.com/Kaldek/rat-ratgdo/blob/main/FAQ.md

sanjay900, 2023-10-30:
> @vaidehi-joshi I have updated the FAQ with extra info in case you are connecting to the wrong terminals on your GDO. Some GDOs do not use the Red/White/Black colour scheme for their wires.
> 
> https://github.com/Kaldek/rat-ratgdo/blob/main/FAQ.md

Thanks for that, my unit is one of these ones

vaidehi-joshi, 2023-10-31:
Latest update on the issue:

We have tried to test our connection with the circuit diagram multiple times. Thank you for the SGD labels @Kaldek . Even followed [this](https://github.com/Kaldek/rat-ratgdo/issues/10#issuecomment-1783921301) and we are pretty certain of our connections. We have done everything in our power to test the rest of it as well(checked continuity, replaced transistors, tested resistors). 
Here's what we see this in the esphome UI:
[image]
But no movement of the door. We are using the blue board config from [here](https://ratgdo.github.io/esphome-ratgdo/) to flash the ESP 

Here's how are GDO setup looks like:
[image]

 Listing the circuit board connections:
 IRLB8721:
Gate- D1
Source- Ground
Drain- Red control wire

2n 7000 n-channel:
Gate- Red control wire
Source- Ground
Drain- D2

Can someone please help us with the verification of the setup and connections?

Kaldek, 2023-10-31:
@vaidehi-joshi the "Collision Detected, waiting to send packet" means that the red signal wire has been pulled low when the TX line is wanting to transmit.  As it is half duplex, you cannot transmit whilst the red wire has been pulled low.

You should **never** see the collision message.  Since I've had the PCB working, I have never seen it.

> Thanks for that, my unit is one of these ones

You didn't attach an image.  What model GDO do you have?  Are you sure it's not a simple dry contact based unit that doesn't even use the communication protocol?

vaidehi-joshi, 2023-10-31:
@sanjay900 First off thank you, this is something we hadn't crossed off the list. 
Our model is Chamberlain Whisper Drive Plus (Model: 547445). It is Security+ not sure how to check if it's Security+ 1.0 or Security+ 2.0.
We have referred to [this](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v4.png) which is the D1, D2, D7 pinout.

Can we still run it on our ESP8266?

rlowens, 2023-10-31:
> It is Security+ not sure how to check if it's Security+ 1.0 or Security+ 2.0.

If it doesn't say 2.0, it is 1.0. Also, [this page says](https://support.chamberlaingroup.com/s/article/Determine-the-Color-of-the-Learn-Button-on-Your-Garage-Door-Opener) your purple Learn button indicates Security+ 1.0.

> Can we still run it on our ESP8266?

Yes, just not the ESPHome version of the firmware (yet). Have to use the MQTT version of the firmware, since they haven't added the Security+ 1.0 and Dry Contact code to the ESPHome firmware (Paul mentioned they are still ironing out some bugs).

rohinrohin, 2023-10-31:
> @vaidehi-joshi the "Collision Detected, waiting to send packet" means that the red signal wire has been pulled low when the TX line is wanting to transmit. As it is half duplex, you cannot transmit whilst the red wire has been pulled low.
> 
> You should **never** see the collision message. Since I've had the PCB working, I have never seen it.
> 
> > Thanks for that, my unit is one of these ones
> 
> You didn't attach an image. What model GDO do you have? Please check Paul Wieland's Wiki article on supported models. Yours might be Security+ 1.0. https://github.com/PaulWieland/ratgdo/wiki
> 
> If yours is Security+ 1.0 you won't be able to use ESPHome and will have to run native ratgdo. [image]
> 
> For Security+ 1.0 you apparently must use the D1, D2, D7 pin layout (not D2, D4, D7).

Sorry for being MIA! Thanks a ton @Kaldek, @rlowens, @sanjay900 for your updates. 

This is possible the issue, I have confirmed that we are indeed on Security+ 1.0, will loop back here once we verify with the native firmware. 

Once done, @vaidehi-joshi and I will summarize all of our learnings till now (including compiling with ESPHome, Security.., and many other nuances that we came across).

sanjay900, 2023-10-31:
> If it doesn't say 2.0, it is 1.0. Also, [this page says](https://support.chamberlaingroup.com/s/article/Determine-the-Color-of-the-Learn-Button-on-Your-Garage-Door-Opener) your purple Learn button [...]

FYI my unit didnt mention anything about what security version it uses, and it was one of those ones with the green serial port and was in fact security+ 2.0

ccutrer, 2023-11-01:
I have an ESP32-DevKitC (pinout: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/_images/esp32-devkitC-v4-pinout.png). I built my rat-ratgdo yesterday with the recommended 2n7000 and IRLB8721 transistors, the flashed it. I had to adjust the pins from https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_lolin_s2_mini.yaml slightly to match the wiring diagram at https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v4.png. And I also had to just completely download the [base.yml](https://github.com/ratgdo/esphome-ratgdo/blob/main/base.yaml) and modify it since I don't actually use home assistant, and I didn't feel like configuring the dry contact relay pins unnecessarily. I'm using USB power for now, with an 8500W jackshaft opener (should be Security+ 2.0). Hooking it up this morning and enabling debug logging, it fails to sync, and I can't send any commands to the opener (logs indicate the command was received from MQTT, and the rolling_code_counter increases, but nothing happens), but I get full feedback from actions triggered by other means - current door state, motion detected by control panel, button pressed at control panel, calibrated current position percentage, etc. So it seems that writing to the bus is failing for me. I suspect I didn't connect the IRLB8721 correctly. I'm pretty good at soldering and using resistors for a few things, but transistors elude me, and definitely am unsure how to read them in a circuit diagram. I was going by [the datasheet](https://www.infineon.com/dgdl/Infineon-IRLB8721-DataSheet-v01_01-EN.pdf?fileId=5546d462533600a40153566056732591) showing (with the heat sync oriented towards the back) pin 1 as gate (connects to TX pin - GPIO16 for me, along with 10k pulldown resistor to ground), pin 2 as drain (to red wire from opener), and pin 3 as source (ground). Attaching photo of my completed board with behind-the-board connections drawn on , as well as my ESPhome config:

[image]
[image]

single_car_gdo.yaml:
```yaml
substitutions:
  name: single-car-gdo
  friendly_name: Single Car Garage Door Opener
  id_prefix: gdo
  uart_tx_pin: GPIO16
  uart_rx_pin: GPIO21
  input_obst_pin: GPIO23
  status_door_pin: GPIO5
  status_obstruction_pin: GPIO12

packages:
  base: !include common/base.yaml
  ratgdo: !include common/ratgdo.yaml

ota:
  password: "ce275759362f3215b59d1fdb1365ffd9"

wifi:
  ap:
    ssid: Single Car GDO
    password: "..."

logger:
  level: DEBUG
```

common/base.yaml:
```
esphome:
  name: $name
  friendly_name: $friendly_name

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: INFO

mqtt:
  broker: mqtt.local
  username: esphome
  password: !secret mqtt_password
  discovery_unique_id_generator: mac
  topic_prefix: esphome/$name

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: $friendly_name Setup

time:
  - platform: sntp
    servers: ntp.local
    timezone: America/Denver

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Strength"
[... 135 lines cut ...]
    entity_category: diagnostic
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).query_openings();

  - platform: template
    id: ${id_prefix}_sync
    name: "Sync"
    entity_category: diagnostic
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).sync();

  - platform: template
    id: ${id_prefix}_toggle_door
    name: "Toggle door"
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).toggle_door();
```

Any help would be appreciated.

rohinrohin, 2023-11-01:
@Kaldek, I can confirm that we have a Security+ GDO. [manual](https://cdn.shopify.com/s/files/1/0090/2591/2887/files/114a4328.pdf?v=1687180694) 

We have flashed ratgdo native and things are still not working as expected. 
 - In the logs for native, it says wall panel detected. 
 - Obstruction sensor status's work
 - Up/Down/Stop/Status updates are still not working
 - I also noticed, then when I pulled the red wire to GND manually. The GDO is able to toggle the garage door to open/close (not sure if this info is useful)

Any idea what we maybe missing out on?

Another side note, I noticed from this [wiki](https://github.com/PaulWieland/ratgdo/wiki) that since I have the purple learning button it only supports ratgdo v2.5 board, but you recommended that I flash firmware for the 2.0 board.

sanjay900, 2023-11-01:
> I have an ESP32-DevKitC (pinout: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/_images/esp32-devkitC-v4-pinout.png). I built my rat-ratgdo yesterday with the recommended ` and IRLB8721 transistors, the flashed it. I had to adjust the pins from https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_lolin_s2_mini.yaml slightly to match the wiring diagram at https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v4.png. And I also had to just completely download the [base.yml](https://github.com/ratgdo/esphome-ratgdo/blob/main/base.yaml) and modify it since I don't actually use home assistant, and I didn't feel like configuring the dry contact relay pins unnecessarily. I'm using USB power for now, with an 8500W jackshaft opener (should be Security+ 2.0). Hooking it up this morning and enabling debug logging, it fails to sync, and I can't send any commands to the opener (logs indicate the command was received from MQTT, and the rolling_code_counter increases, but nothing happens), but I get full feedback from actions triggered by other means - current door state, motion detected by control panel, button pressed at control panel, calibrated current position percentage, etc. So it seems that writing to the bus is failing for me. I suspect I didn't connect the IRLB8721 correctly. I'm pretty good at soldering and using resistors for a few things, but transistors elude me, and definitely am unsure how to read them in a circuit diagram. I was going by [the datasheet](https://www.infineon.com/dgdl/Infineon-IRLB8721-DataSheet-v01_01-EN.pdf?fileId=5546d462533600a40153566056732591) showing (with the heat sync oriented towards the back) pin 1 as gate (connects to TX pin - GPIO16 for me, along with 10k pulldown resistor to ground), pin 2 as drain (to red wire from opener), and pin 3 as source (ground). Attaching photo of my completed board with behind-the-board connections drawn on , as well as my ESPhome config:
> 
> [image] [image]
> 
> single_car_gdo.yaml:
> 
> ```yaml
> substitutions:
>   name: single-car-gdo
>   friendly_name: Single Car Garage Door Opener
>   id_prefix: gdo
>   uart_tx_pin: GPIO16
>   uart_rx_pin: GPIO21
>   input_obst_pin: GPIO23
>   status_door_pin: GPIO5
>   status_obstruction_pin: GPIO12
> 
> packages:
>   base: !include common/base.yaml
>   ratgdo: !include common/ratgdo.yaml
> 
> ota:
>   password: "ce275759362f3215b59d1fdb1365ffd9"
> 
> wifi:
>   ap:
>     ssid: Single Car GDO
>     password: "..."
> 
> logger:
>   level: DEBUG
> ```
> 
> common/base.yaml:
> 
> ```
> esphome:
>   name: $name
>   friendly_name: $friendly_name
> 
> esp32:
>   board: esp32dev
>   framework:
>     type: arduino
> 
> logger:
>   level: INFO
> 
> mqtt:
>   broker: mqtt.local
>   username: esphome
>   password: !secret mqtt_password
>   discovery_unique_id_generator: mac
>   topic_prefix: esphome/$name
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   ap:
>     ssid: $friendly_name Setup
> 
> time:
>   - platform: sntp
>     servers: ntp.local
>     timezone: America/Denver
> 
> captive_portal:
> 
> sensor:
>   - platform: wifi_signal
[... 140 lines cut ...]
>         lambda: !lambda |-
>           id($id_prefix).query_openings();
> 
>   - platform: template
>     id: ${id_prefix}_sync
>     name: "Sync"
>     entity_category: diagnostic
>     on_press:
>       then:
>         lambda: !lambda |-
>           id($id_prefix).sync();
> 
>   - platform: template
>     id: ${id_prefix}_toggle_door
>     name: "Toggle door"
>     on_press:
>       then:
>         lambda: !lambda |-
>           id($id_prefix).toggle_door();
> ```
> 
> Any help would be appreciated.

The pinouts assume you are using the `ESP32 d1 mini` yml file, hence why the pins are totally different to the schematic when you pick the `lolin s2 mini` board as your base.
If you are receiving data but not able to send, that sounds like something is wrong with just the TX portion of your circuit. Check your `2n7000` isn't dead, I have had a few of those fail on me and when they fail, source to drain is just shorted permanently.

sanjay900, 2023-11-01:
@rohinrohin 
Yeah so with the ESPHome firmware, there is no difference between 2.0 and 2.5 besides pinouts, hence why we just use the 2.0 pinouts as the web flasher doesn't support 2.5 with esp32s.

Only the MQTT firmware supports security+ 1. 2.5 uses pin GPIO22, while 2.0 uses GPIO16. In saying that, i didn't think the mqtt firmware had offical support for the ESP32, so did you build that yourself?

Kaldek, 2023-11-01:
> I have an ESP32-DevKitC (pinout: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/_images/esp32-devkitC-v4-pinout.png). I built my rat-ratgdo yesterday with the recommended 2n7000 and IRLB8721 transistors, the flashed it. 

We're starting to suspect that - even though the specs are correct - the IRLB8721 is just not working for anyone!  Please try with another 2n7000 and get back to us as soon as you can!

Kaldek, 2023-11-01:
CRAP.  Confirmed it looks like the lack of current is what stops the IRLB8721 from switching:
https://forums.adafruit.com/viewtopic.php?p=446184

vaidehi-joshi, 2023-11-02:
> @rohinrohin Yeah so with the ESPHome firmware, there is no difference between 2.0 and 2.5 besides pinouts, hence why we just use the 2.0 pinouts as the web flasher doesn't support 2.5 with esp32s. The ESPHome firmware doesn't support security+ 1 at all.
> 
> Only the MQTT firmware supports security+ 1. 2.5 uses pin GPIO22, while 2.0 uses GPIO16. In saying that, i didn't think the mqtt firmware had official support for the ESP32, so did you build that yourself?

Thanks @sanjay900! 

@rohinrohin and I purchased a new D1Mini ESP8266MOD - [this exact one](https://www.amazon.com/MakerFocus-NodeMcu-Development-ESP8266-Compatible/dp/B07KW54YSK?th=1) and we are running all experiments on this for now. 

@Kaldek 

To summarize, 
1. We are using the D1Mini ESP8266MOD
2. We have a security 1+ GDO (purple learn button)
3. We are using 2x 2n7000n Transistors
4. Obstruction sensor is working
5. We used native ratgdo and flashed `ratgdo v2.26, Security + 2.0 (secplus 1.0 alpha)`
6. Logs show `Wall panel detected` and when I try to open the door, it says `Door is not moving`

😞

sanjay900, 2023-11-02:
> > @rohinrohin Yeah so with the ESPHome firmware, there is no difference between 2.0 and 2.5 besides pinouts, hence why we just use the 2.0 pinouts as the web flasher doesn't support 2.5 with esp32s. The ESPHome firmware doesn't support security+ 1 at all.
> > Only the MQTT firmware supports security+ 1. 2.5 uses pin GPIO22, while 2.0 uses GPIO16. In saying that, i didn't think the mqtt firmware had official support for the ESP32, so did you build that yourself?
> 
> Thanks @sanjay900!
> 
> @rohinrohin and I purchased a new D1Mini ESP8266MOD - [this exact one](https://www.amazon.com/MakerFocus-NodeMcu-Development-ESP8266-Compatible/dp/B07KW54YSK?th=1) and we are running all experiments on this for now.
> 
> @Kaldek
> 
> To summarize,
> 
>     1. We are using the D1Mini ESP8266MOD
> 
>     2. We have a security 1+ GDO (purple learn button)
> 
>     3. We are using 2x 2n7000n Transistors
> 
>     4. Obstruction sensor is working
> 
>     5. We used native ratgdo and flashed `ratgdo v2.26, Security + 2.0 (secplus 1.0 alpha)`
> 
>     6. Logs show `Wall panel detected` and when I try to open the door, it says `Door is not moving`
> 
> 
> Any help greatly appreciated! 😞

Might be worth spinning up another issue for this, this one was more about the esp32

[1 more reply]

closed 2023-11-02 by Kaldek