Kaldek/rat-ratgdo #5: Dry contact configuration schematics (for older/stupid GDO like Genie)

CrazyCoder, 2023-10-26:
Ratgdo 2.5 added support for dry contact configuration:

[image]

The code supporting this mode is in a separate branch at the moment: https://github.com/PaulWieland/ratgdo/tree/2.0-secplus1.0.
ratgdo-esphome doesn't support it yet.

In this mode, the door is opened/closed by pilling the `OUTPUT_GDO` (TX) low, see https://github.com/PaulWieland/ratgdo/blob/e9a49eb5bab7eb3bbb08113d99b5805be8854abe/src/ratgdo.cpp#L620.

So, it looks like we can omit the RX pin components completely as only TX pin is used? Maybe the TX GPIO connection can be also simplified for this scenario?

We also have open/close limit switches.

In my Genie 8600IC, the voltage between the control button terminals is ~6V.
The voltage between the open/close limit switch terminals is ~8V.
The voltage between the obstruction sensors is 13V.

ESP32 board will need to be protected from the high voltage and also be able to read the pulse signal (~3V-13V) of the obstruction sensor (convert to 3.3V CMOS logic):

[image]

I'm a noob in electronics, so the suggestions for the best schematics to support all the above are welcome.

To sum up, the schematics for the following would be appreciated:

- TX GPIO to short 6V button switch line to toggle door open/close
- 2 x INPUT GPIOs to sense open/close limit switches at 8V
- 1 x INPUT GPIO to sense 3-13V pulses at 3.3 CMOS logic level

Kaldek, 2023-10-26:
I can't speak for the open/close limit switches yet but the existing TX logic should work without alteration as should the obstruction sensor.  I've updated the schematic to show that there are two 10k resistors (set up as a voltage divider) so that the obstruction sensor voltages reach the ESP8266 GPIO pin at ~3.3v.

I won't lie; this is annoying tricky stuff when you're really new to electronics.  There's no easy way to remotely simplify it.  As much as you need a solution **soon** you might have to resort to buying a kit from Paul Wieland and just waiting for delivery.

CrazyCoder, 2023-10-26:
> I've updated the schematic to show that there are two 10k resistors (set up as a voltage divider) so that the obstruction sensor voltages reach the ESP8266 GPIO pin at ~3.3v.

An input voltage range of 3V to 13V would result in an output voltage range of 1.5V to 6.5V for the obstruction sensor. Am I right? 6.5V will fry the ESP.

> you might have to resort to buying a kit from Paul Wieland and just waiting for delivery

I would gladly buy the 2.5 version if it properly supported the obstruction sensors I have. But it doesn't and would need the voltage divider as well. The next version supporting wider voltage ranges is several months away.

I see no reason to buy 2.5 board now since I'd still need to do some modifications. Why not do it from scratch?

Kaldek, 2023-10-26:
> 
> An input voltage range of 3V to 13V would result in an output voltage range of 1.5V to 6.5V for the obstruction sensor. Am I right? 6.5V will fry the ESP.
> 

Your setup might require using a MOSFET to act as a level shifter. I am sorry to admit that without a similar setup here that I could test and validate, I can't do the work on your behalf.  Hopefully there will be another person who has the electronics experience and the same setup.

rlowens, 2023-10-26:
> An input voltage range of 3V to 13V would result in an output voltage range of 1.5V to 6.5V for the obstruction sensor. Am I right? 6.5V will fry the ESP.

You would just need to adjust the resistor value for the voltage divider. To get 13V max down to 3.3V, using https://ohmslawcalculator.com/voltage-divider-calculator and https://www.rfcafe.com/references/electrical/resistor-values.htm you need resistors of 10k for White Wire to ESP and 3.3k for ESP to GND. Will take the 13V max to 3.226V.

CrazyCoder, 2023-10-26:
@rlowens Thanks. That is what I figured with the calculator myself. 

So, I can control the switch with the original schematics using a MOSFET, can read the obstruction sensor voltage via the voltage divider.

What about the contacts for open/close? Also a voltage divider or a zenner diode?

Kaldek, 2023-10-26:
> 
> What about the contacts for open/close? Also a voltage divider or a zenner diode?

Was that question answered in this comment? https://github.com/Kaldek/rat-ratgdo/issues/1#issuecomment-1780011138

CrazyCoder, 2023-10-26:
@Kaldek Sorry, missed that:

> they just have an SL 1N5819W schottky barrier diode between input and GPIO (which you can omit if you want, they are just there for extra protection)

Still don't understand why it can be omitted if the voltage is ~8V on the open/close switches.

Also, how exactly this diode should be wired and what would be its effect? How will it protect the GPIO from high voltage? 

Thanks for your help and sorry for the stupid questions =)

rlowens, 2023-10-26:
> Still don't understand why it can be omitted if the voltage is ~8V on the open/close switches.

Because I didn't understand that these would be wired to the GDO too, not only to the ratgdo. You do need the diodes to protect the ESP from the GDO's voltage there.

> Also, how exactly this diode should be wired and what would be its effect? How will it protect the GPIO from high voltage?

They are wired facing away from the ESP. So the circuit would go: ESP GPIO Pin to diode anode end, diode cathode end to positive wire to limit switch (also connected to GDO's ~8V), limit switch connects to ground when pressed.
[image]
Thus, when the limit switch is open, the ~8V from the GDO is blocked from entering the ESP GPIO, and no energy flows out of the GPIO so it remains HIGH and the ESP knows the switch is open. When the switch is closed, the ~8V from the GDO is connected to ground so the GDO knows the switch is closed and the ~8V changes to ~0V (since it is grounded) and energy flows out of the GPIO pulling it LOW so the ESP knows the switch is closed.

Make sense?

CrazyCoder, 2023-10-26:
> Make sense?

@rlowens This is a nice explanation and it makes sense to me. I forgot that GPIOs in input mode are pulled up via the internal pull-ups. Thank you again.

Kaldek, 2023-10-26:
Gent, let me know what constitutes this issue being resolved.  I'm happy to leave it open if the answer is "somebody adds to the schematic for dry-contact GDO systems" and we can just work the problem as a collective.

CrazyCoder, 2023-10-26:
@Kaldek Please keep it open until I build and verify that everything works according to the suggestions above. Feel free to add this to the schematics too, I'm sure many users will find it useful.

Kaldek, 2023-10-29:
@CrazyCoder can you take a look at the new Kicad_files folder and let me know if the needed content is in there for you?

CrazyCoder, 2023-10-29:
@Kaldek These files do not include 1N5819 diodes, traces and terminals for dry contact open/close inputs and light output.
Several comments above is the relevant part of the scheme: https://github.com/Kaldek/rat-ratgdo/issues/5#issuecomment-1780422981.

CrazyCoder, 2023-10-31:
@rlowens, @Kaldek I have an issue with the dry contact mode setup for the open/close control itself. Maybe you have any ideas what might be wrong?

Everything is wired correctly according to the schematic and I'm using (IRLB8721)[https://www.amazon.com/dp/B083TGVFNQ] MOSFET for TX control.

I verified with the multimeter continuity test that MOSFET DRAIN and GROUND have a connection when I press a button to close or open the door in Home Assistant (for the configured 500ms). In other words, MOSFET drain is pulled to ground as expected which should emulate the button press.

I also tested that if I short the red and ground button wires, the GDO triggers door open/close.

The problem is, that as soon as I connect the red door button wire to MOSFET DRAIN and try to trigger the door via Home Assistant, it does not work and the multimeter no longer beeps for continuity.

As soon as the red button wire is disconnected from the scheme, multimeter beeps as it should on door toggle via HA.

It looks like MOSFET is not able to pull the button control logic to ground (it's not a simple button, it has a LED and some small PCB). But if I short the wires manually, it works.

I thought that maybe 500ms is not enough for pulling it down to ground and tried with 2000ms, but it didn't help.

I'm pretty sure I can use a relay instead of the MOSFET, connect it to D1 GPIO, change the code logic to pull it up to VCC instead of pulling it down so that it triggers the relay and connects button ground and red wires via the the NO terminals of the relay.

What could be the problem and what are the other possible options? I know the original schematic was not designed to control stupid doors by shortening the contacts and this mode was added as a hack only recently. It looks like a change to the scheme is needed as well, not just to the code, at least for my specific GDO model.

Thanks for any ideas.

CrazyCoder, 2023-10-31:
I tested it with a LAB power supply set to 0.005A and 3V and measured the voltage between the source and the drain of the MOSFET (to simulate a button with the LED circuit). Normally it's 3V and the power supply doesn't show any load. When HA toggles a door, the voltage between the source and the drain starts to drop slowly. Power supply is showing that a bit of current flows, but the voltage never drops to 0V like it does on the short circuit. Is it expected behavior?

Kaldek, 2023-10-31:
This sounds awfully like your MOSFET isn't being switched fully.  So, funny thing with MOSFETs, they have this narrow window where you can stop them from acting like an on/off switch and kind of use them as an analog control where it behaves as you describe.

Make sure you probe what the voltage coming off the TX PIN is when you activate it.  It needs to be at a suitable voltage to fully trigger the gate of the MOSFET, which we thought we have confirmed as being suitable with the IRLB8721!

CrazyCoder, 2023-10-31:
@Kaldek I'm not running the 10K resistor in series, and I already tested the voltage at the gate, and it's a healthy 3.3V. I also swapped a couple of MOSFETs and they all behave the same.

Kaldek, 2023-10-31:
> @Kaldek I'm not running the 10K resistor in series, and I already tested the voltage at the gate, and it's a healthy 3.3V. I also swapped a couple of MOSFETs and they all behave the same.

Can you try this with a 2n7000 if not already?

CrazyCoder, 2023-10-31:
@Kaldek Yes, was already planning to do that. So, I upped the simulated button power to 5V and 0.1A and tried it with 2n7000. It already works much better and is able to pass more current. When the button is toggled, the voltage between drain and source drops from 5V to 0.4V. But it probably won't be enough for the GTO to trigger (I may test it tomorrow since too lazy to go to the garage now). 

Still thinking about going with a relay since it will work 100%. I already have a dozen of (JQC-3F)[https://www.amazon.com/dp/B00IIDY8JU] that can be installed on a breadboard if you bend the common pin a bit.

Kaldek, 2023-10-31:
> @Kaldek Yes, was already planning to do that. So, I upped the simulated button power to 5V and 0.1A and tried it with 2n7000. It already works much better and is able to pass more current. When the [...]

Do you think that maybe Paul Wieland hasn't successfully completed testing of this?  I can go and test my own system with an oscilloscope to see how log the TX line goes when it's being switched.

CrazyCoder, 2023-10-31:
I think I probably just have a bad batch of IRLB8721. Tested all of them, and they behave the same (except one, which doesn't pass any current at 3.3V). They should open fully at 3.3V unlike 2n7000, but with my testing they behave much worse.

rlowens, 2023-10-31:
> Do you think that maybe Paul Wieland hasn't successfully completed testing of this? I can go and test my own system with an oscilloscope to see how log the TX line goes when it's being switched.

Note that we don't actually know at all what Paul Wieland is using on the ratgdo v2.5 board which supports Dry Contact mode.

Kaldek, 2023-10-31:
Right I've tested red wire to GND with my oscilloscope when it's being switched and it's absolutely pulling the +12v line to GND fully when switching.

CrazyCoder, 2023-10-31:
@rlowens That is true.

On another note, I just tried it with [RFP30N06LE](https://www.digikey.com/en/products/detail/fairchild-on-semiconductor/RFP30N06LE/458769) that I had from [this kit](https://www.amazon.com/gp/product/B09K3T5LSN/), and they are working perfectly!

I'll use RFP30N06LE in my design.

Kaldek, 2023-10-31:
Image from my Hantek on switching voltages driven by the 2n7000.
[image]

Kaldek, 2023-10-31:
> I just tried it with [RFP30N06LE](https://www.digikey.com/en/products/detail/fairchild-on-semiconductor/RFP30N06LE/458769) that I had from [this kit](https://www.amazon.com/gp/product/B09K3T5LSN/), and they are working perfectly!

Let me go compare the datasheets for the critical stats of these three MOSFETs.

Kaldek, 2023-10-31:
Here we go.  

| Parameter     | 2N7000          | IRLB8721        | RFP30N06LE     |
|---------------|-----------------|-----------------|----------------|
| V_{DS}        | 60V             | 30V             | 60V            |
| V_{GS(th)}    | 0.8V to 3V      | 1V to 2.35V     | 1V to 2V       |
| R_{DS(on)} (at 3.3V V_GS) | ~5Ω (approx., and at a much lower current) | ~8.7mΩ (at 15A) | ~35mΩ (at 20A, with V_GS at 5V; might be a bit higher at 3.3V) |

Kaldek, 2023-10-31:
> I think I probably just have a bad batch of IRLB8721. Tested all of them, and they behave the same (except one, which doesn't pass any current at 3.3V). They should open fully at 3.3V unlike 2n7000, but with my testing they behave much worse.

Would love to see what the exact code is on your IRLB8721 parts, or even a nice close up picture.

CrazyCoder, 2023-10-31:
@Kaldek 

[image]

Kaldek, 2023-10-31:
Well they certainly **look** correct.  I can't even theorise you wired them wrong because the RFP30N06LE  is also TO-220 with the same G/D/S pin layout from left to right.

CrazyCoder, 2023-10-31:
Indeed, I'm also surprised all 10 from the bag didn't work according to the specs.

Anyway, I'm glad we sorted it out. Thanks a lot for your help.

My breadboard design is working perfectly, sensing the obstruction and open/close limit switches. Final tests tomorrow, then converting to the permanent proto board.

[image]

Prepared the wiring. Spliced the limit switch wires with Wago connectors:

[image]

Kaldek, 2023-10-31:
I checked all the datasheet capacitance charts for these MOSFETs.  They're within the same range, so it's not like the 10k resistor is too much.  I mean, if you have the *time* you could try a 1k pulldown.

CrazyCoder, 2023-10-31:
I even tried without the resistor, and it didn't change anything.

closed 2023-10-31 by Kaldek

CrazyCoder, 2023-11-02:
Here is the final working minimal design for the dry contact mode (no security, no TX line), 10K and 3.3K resistors for the voltage divider (to be compatible with 12V obstruction sensor Genie uses):

[image]

And here is the real working device:

[image]

[image]

Connected, installed and working:

[image]

Mounted using the [Scotch Self Mating Reclosable Fasteners](https://www.amazon.com/gp/product/B09FYB7J4H/).

CrazyCoder, 2023-11-02:
Running slightly modified 2.5 MQTT firmware from https://github.com/ratgdo/mqtt-ratgdo with the fix for this issue: https://github.com/PaulWieland/ratgdo/issues/168.

CrazyCoder, 2023-12-07:
@dbrewer-wdc I'm using the original ratgdo schematic here with the control via D1 which is pulled down to the ground with the 10-Ohm resistor, so it will never trigger a door on reset. I did test it. If you are using some other pin, it may go HIGH on reset depending on the board you are using and its configuration, or you need to make sure you pull this pin down.

rlowens, 2023-12-07:
> Most all the outputs go high/active on reset. Some IOs stay high longer than others.

Another thing to look into if you are using ESPHome with ESP8266 is setting early_pin_init: false (ref https://esphome.io/components/esp8266.html#configuration-variables ). My smart plugs and light switches often had a moment of their relay clicking for a moment on reboot/update without that.

dbrewer-wdc, 2023-12-07:
@CrazyCoder , yeah, you're right about the 10k, that will work. I've had to move things around so much to get them working that, I've forgotten where I've come from. Glad you tested it. The problem I have is that I just have a ton of these 2n7000 mosfets and I used those to bridge the switches on the wall switch. However, these mosfets have a higher threshold voltage which make them a poor choice for what I'm doing. The way the wall switch works, the low side of the button switch is already 10k up from GND which means I either need a lower threshold N channel mosfet or I higher gate input. I had some level shifters so I used those to boost the voltage to the gate of the mosfet...but the consequence is that the level shifter I use cannot have a pull down on its input or it doesn't work; so those went away. Anyway, that's what I did because of parts-on-hand and it was working great till I did a reset. That was a surprise; but with more 2n7000s, problem solved. 

Best Regards