Kaldek/rat-ratgdo #8: Convert v2 board to v2.5?

tfcarlin, 2023-10-27:
I'd like to update a v2.0 shield to v2.5. It is as simple as removing the R3 resistor and adding a 10k resistor between d1 and the "output" side of R3?

THANKS!

rlowens, 2023-10-27:
As far as we know, yes. Either remove R3 (so that D4 is not connected) and connect a 10k resistor between D1 and the "output" pad of R3, or cut the trace between R3 and D4 and add a bodge wire from the "input" pad of R3 to D1.

We haven't seen any pictures of the front of ratgdo v2.5 to trace where anything new may have been added. From the programming side, changing from D4 to D1 is the only change we know of.

Reference for others reading this discussion, this is the ratgdo v2.0:
[image]
[image]

Kaldek, 2023-10-28:
@tfcarlin since you have a board in hand can you do some probing with a multimeter in resistance mode for us?

Please do the following two probes and provide the resistance values here if you can!
[image]

tfcarlin, 2023-10-28:
Both the green and red arrows report 0.

Kaldek, 2023-10-28:
> Both the green and red arrows report 0.

Can you do the green one again sorry but from the other side of the resistor?  So it's from the white wire terminal to the left side of that resistor.

Kaldek, 2023-10-31:
Paging @tfcarlin !

tfcarlin, 2023-10-31:
Unfortunately I have since converted my 2.0 shield to have a 10k resistor between d1 and the output of R3 (and thus removed the 10k resistor surface mounted at R3). So my readings might be corrupt. With that caveat, both Red and Green arrows read 0L on my Fluke 179.

Kaldek, 2023-10-31:
Cheers, thanks for putting in the time anyway!

closed 2023-10-31 by Kaldek