Kaldek/rat-ratgdo #9: Schematic - Layout Error

clumsyCoder00, 2023-10-27:
The schematic is showing the Red/Control wire as being connected to ground, but the breadboard layout shows it as only being connected to the two FETs. I'm guessing the breadboard layout is correct and the Control wire should not be shorted to ground. Thanks for putting this together!

rlowens, 2023-10-27:
That's a cross-over not a junction. Junctions have a dot. Here's my version of the schematic:
[image]

Note that Kaldek's version has that top R1 connecting from gate to ground instead of in-series gate-R1-D1, but this is how the ratgdo has it wired.

Kaldek, 2023-10-28:
> 
> Note that Kaldek's version has that top R1 connecting from gate to ground instead of in-series gate-R1-D1, but this is how the ratgdo has it wired.

Aren't we still debating this? 😁  Every time I go looking into the in-series resistor as a "gate resistor" the ranges suitable for the resistor **top out** at around 1kohm, and the faster the gate needs to be switched the lower the ohms (like down to 10 ohms even).

When I tried to use the in-series 10k resistor, the 2n7000 would not switch.  We really need someone to wire their circuit this way and confirm behaviour.

Kaldek, 2023-10-28:
>  Here's my version of the schematic: [image]
> 

I am keen to switch to the Ki-Cad version too once we've put this in-series resistor question to bed empirically.  I'm sure it will make it easier for people to then use all that to send files to PCBWay etc to have their board made on their behalf.

Kaldek, 2023-10-28:
Here's my latest question posed to ChatGPT on this.  I keep running this query through it using various prompts to avoid it hallucinating and getting confused.  Regardless of that, it always comes back with the same answer.  Here's that latest one:

> You're correct in your observation, and I can provide some insights based on the given information:

> 1. **AO3400A MOSFET**: The AO3400A is an N-channel, logic-level MOSFET. Its typical gate threshold voltage (Vgs(th)) is around 1.0V to 2.1V, which means it can be fully turned on with a 3.3V logic signal from the ESP8266. Given its common use in low-voltage logic applications, it's likely suitable for direct driving by the ESP8266.

> 2. **10k Resistor**: A 10kΩ resistor is quite large to be in series for switching a MOSFET, especially at 9600 bits per second (bps). If it were in series, the RC time constant (with the gate capacitance) would indeed make the switching times very long. This would be very inefficient and could potentially damage the MOSFET if it lingers too long in the linear region due to the slow switching.

>  Given this, it seems more plausible that the 10kΩ resistor is a pull-down resistor. A pull-down resistor on the gate of the MOSFET is a common design choice to ensure the MOSFET remains off during times when the driving GPIO might be in a high-impedance state, such as during microcontroller boot-up.

> 3. **Switching at 9600bps**: If you're modulating the gate of the MOSFET at 9600bps (like in a UART communication signal), then the gate needs to switch on and off very quickly. A high-value series resistor would drastically slow down this switching and distort the signal. A pull-down resistor would not interfere with the switching speed when the GPIO is driving the gate, as the resistor is effectively out of the circuit (only providing a path to ground when the GPIO isn't driving the gate).

> In conclusion, based on your description and the requirements of the application (switching at 9600bps), it's more likely that the 10kΩ resistor is functioning as a pull-down resistor for the gate of the AO3400A MOSFET, not as a series resistor.

Kaldek, 2023-10-28:
I should add that even if the original board is wired in series I'm suspicious of that design and, at the moment, I think it's wrong.  Maybe let's ask that fellow who posted in the other issue about altering his 2.0 board and get him to probe the circuit with a multimeter.

clumsyCoder00, 2023-10-28:
> That's a cross-over not a junction. Junctions have a dot. Here's my version of the schematic: [image]
> 
> Note that Kaldek's version has that top R1 connecting from gate to ground instead of in-series gate-R1-D1, but this is how the ratgdo has it wired.

Thanks for your response. It's not a cross-over. There's a junction at conductor 2 of J1, connecting to a junction at the negative side of the DC supply, D1. The alternate schematics are an improvement, even if the original (mostly) gets the job done.

Kaldek, 2023-10-28:
There's a junction at conductor 2 of J1, connecting to a junction at the negative side of the DC supply, D1. The alternate schematics are an improvement and are omitting the ground connection correctly, even if the original (mostly) gets the job done.

Argh stupid Fritzing, let me fix that.  It keeps doing this automatically.

Kaldek, 2023-10-28:
I'm going to pivot to Ki-Cad.  For now I'll have to alter the design to use the pull down as I still think that's correct.  I'll work on this today.

closed 2023-10-28 by Kaldek

rlowens, 2023-10-28:
> I'm going to pivot to Ki-Cad. For now I'll have to alter the design to use the pull down as I still think that's correct. I'll work on this today.

Emailed you this.
[image]

clumsyCoder00, 2023-10-28:
Thanks everyone, nice work on the new layouts.