Kaldek/rat-ratgdo #10: Wired and remote isn't turning on, what am I doing wrong?

mikekay1, 2023-10-28:
This is what I followed -- 
[image]

the black wire leads to R on the remote (confirmed)

in the next post is my wiring at GDO

mikekay1, 2023-10-28:
[image]

Kaldek, 2023-10-28:
Hi mate, it's possible you followed one of my busted-ass schematics which Fritzing keeps borking.  Check the v7 schematic and make sure your design follows that.  Most importantly make sure you don't have the red wire shorted to ground like I did in the v6 diagram.   

If this was the problem I am really sorry.

mikekay1, 2023-10-28:
Hey Kal, thanks for the quick reply -- Ah shit I did follow v6 I thought that was the newest one. but still shouldn't remote work even with the esp32 off?

mikekay1, 2023-10-28:
for additional reference, I followed this from the supported boards readme:

https://github.com/Kaldek/rat-ratgdo/blob/main/ratgdo%20open%20source%20ESP-32_schem_v1.png

Kaldek, 2023-10-28:
> Hey Kal, thanks for the quick reply -- Ah shit I did follow v6 I thought that was the newest one. but still shouldn't remote work even with the esp32 off?

If the red wire is grounded the wall remote won't get any power, regardless of the ESP32.  Don't feel too bad about this BTW, when I was first reverse engineering this earlier in the week I grounded out the red wire like ten frickin' times.

Kaldek, 2023-10-28:
> for additional reference, I followed this from the supported boards readme:
> 
> https://github.com/Kaldek/rat-ratgdo/blob/main/ratgdo%20open%20source%20ESP-32_schem_v1.png

I'm re-checking that diagram now and it looks correct. The other thing to be aware of is whether you may have the MOSFET pins flipped around.  Ignore the pin *numbering* on that diagram, it's Fritzing having the icons totally wrong.  

Let me upload one with Source, Gate, and Drain labelled.

mikekay1, 2023-10-28:
but regardless, the remote should turn on even if the esp32 isn't on right? like doesn't this whole system work without anything turned on because it runs in parallel?

I had a buddy whos an electrical engineer check my work. Im pretty sure I am good with the mosfets, I got them back to back. Let me take a pic.

Kaldek, 2023-10-28:
Updated image:
https://github.com/Kaldek/rat-ratgdo/blob/main/images/ratgdo%20open%20source%20ESP-32_schem_v2.png

> but regardless, the remote should turn on even if the esp32 isn't on right? like doesn't this whole system work without anything turned on because it runs in parallel?

Not quite.  If the passive circuit design is wrong the whole Red wire is permanently grounded.

mikekay1, 2023-10-28:
Its impossible to take a picture of breadboard its so jam packed because I wanted to make aure I could fit it on a d1

Kaldek, 2023-10-28:
> Its impossible to take a picture of breadboard its so jam packed because I wanted to make aure I could fit it on a d1

Have a look at my D1 Mini breadboard that was my first working prototype and let me know if yours follows this?
https://github.com/Kaldek/rat-ratgdo/blob/main/Breadboard_working.png

mikekay1, 2023-10-28:
Yea followed it to a T, i assumed this green wire leads to the center pin...

[image]

Kaldek, 2023-10-28:
> Yea followed it to a T, i assumed this green wire leads to the center pin...

Yep, sure does.  The pins of the MOSFET are Source/Gate/Drain so the middle pin is Gate.

Check your MOSFET to make sure it's not shorted Gate to Ground.  It is the *classic* MOSFET failure mode.

mikekay1, 2023-10-28:
Hey bro, totally ripper, I looked at your schematic and actually lined up the ends pins on the n7000's back to back. what I outta do is short it to the center pin, because soo as I unplugged the chip it stopped pulling to ground and remote turned on. I think I need another drink haha!

Kaldek, 2023-10-28:
Also make sure with your ESP-32 setup that you check our updated notes.  We're pivoting back to using the v2.0 web installer option for anyone with an ESP-32 because the ESPHome flasher web page doesn't have the v2.5 ESP-32 settings available, and Paul Wieland hasn't even started offering the v2.5 settings on his downloader.

On an ESP-32, TX should therefore be connected to GPIO16!

mikekay1, 2023-10-28:
Yea I used 2.0, read that in the other thread, no need for 2.5 because I dont need dry contact support or additional(?) trip sensors... I will put a ultrasonic sensor attached to this esp32 for car detection as well...

mikekay1, 2023-10-28:
> Also make sure with your ESP-32 setup that you check our updated notes. We're pivoting back to using the v2.0 web installer option for anyone with an ESP-32 because the ESPHome flasher web page doesn't have the v2.5 ESP-32 settings available, and Paul Wieland hasn't even started offering the v2.5 settings on his downloader.
> 
> On an ESP-32, TX should therefore be connected to GPIO16!
> 
> I mean, you _could_ try to use ESPHome and setup a v2.5 wiring layout but it's currently, ah, "[fun](https://github.com/Kaldek/rat-ratgdo/issues/3)"?

So yea, I actually followed that thread for pinouts but followed the schematics from v 2.0 is that not what I am supposed to be doing?

as for GPIO16, I went with 23,22 and 21 is that also not correct?

Kaldek, 2023-10-28:
> as for GPIO16, I went with 23,22 and 21 is that also not correct?

Yeah I'm sorry this is a bit confusing but it's IO16/GPIO16 rather than GPIO21.  I hadn't gotten around to updating the schematic.

mikekay1, 2023-10-28:
kk noted, I will update! and try again 👍

Kaldek, 2023-10-28:
> kk noted, I will update! and try again 👍

Give me 5 minutes to make sure I get it right.  I've got a rotten hangover.

Kaldek, 2023-10-28:
Righto I think I've fixed this up.
https://github.com/Kaldek/rat-ratgdo/blob/main/images/ratgdo%20open%20source%20ESP-32_schem_v3.png

mikekay1, 2023-10-28:
let me give it a whirl

mikekay1, 2023-10-28:
one quick question, if using 2n7000 x 2 configuration, are the gates in the same orientation as the schematic provided? I'm  waiting on the additional transistor from amazon.

Kaldek, 2023-10-28:
> one quick question, if using 2n7000 x 2 configuration, are the gates in the same orientation as the schematic provided? I'm waiting on the additional transistor from amazon.

They should be correct as per the schematic.  On the RX line, Gate is connected to the Red wire whereas on TX the Drain is connected to it.

mikekay1, 2023-10-28:
Yea sorry im a bit of a noob qhen it comes to bill of materials, i dont experience with these transitors.

Soon as i plih in my circuit the damn wall unit turns off aomething is wrong on my board and I cant pinpoint it

Kaldek, 2023-10-28:
> Yea sorry im a bit of a noob when it comes to bill of materials, i dont experience with these transitors.
> 
> Soon as i plih in my circuit the damn wall unit turns off aomething is wrong on my board and I cant pinpoint it

Did you validate that neither of your MOSFETs have shorted Gate to Source?  Remove them from the board and do a continuity test.  That's step #1.  After that it's just that slow process of tracing each wire.  It's particularly tricky on breadboards.

Common issues I've had with my breadboard setups over the years:

- Accidentally connecting to the ground plane
- Not seeing that I've got two separate circuits connected to each other
- Getting parts installed backwards

sanjay900, 2023-10-28:
> > Yea sorry im a bit of a noob when it comes to bill of materials, i dont experience with these transitors.
> > Soon as i plih in my circuit the damn wall unit turns off aomething is wrong on my board and I cant pinpoint it
> 
> Did you validate that neither of your MOSFETs have shorted Gate to Source? Remove them from the board and do a continuity test. That's step 1. After that it's just that slow process of tracing each wire. It's particularly tricky on breadboards.
> 
> Common issues I've had with my breadboard setups over the years:
> 
>     * Accidentally connecting to the ground plane
> 
>     * Not seeing that I've got two separate circuits connected to each other
> 
>     * Getting parts installed backwards

The amount of times i had MOSFETs with shorted Gate to Sources yesterday....

mikekay1, 2023-10-28:
Good tips here, ill try pulling the mosfets and tracing back.

Can someone give me an true to esp32 wiring diagram? Is that the auper updated one provided by @Kaldek?

poldim, 2023-10-28:
Can you guys confirm that schematics/ratgdo open source_schem_v7.png is accurate?  I can share the easyeda layout:

[image]

rlowens, 2023-10-28:
> Can you guys confirm that schematics/ratgdo open source_schem_v7.png is accurate? 

Accurate to what? It does match your layout here. In ratgdo v2.0 the 10k resistor on the gate of IRLB8721 is in-series to D1 where we think it makes more sense to have it as a pull-down between gate and ground, as you have matched here.

> I think the matches the schematic just a bit cleaner.

Very nice, but I'd also like the Source/Gate/Drain labeled on the MOSFETs for those of us that aren't great at reading them.
Mine:
[image]

> Personally, I'd like to use D5/D6/D7 for the connections as they allow you to keep D1/D2 for hardware I2C.
That sounds fine to me. Just modify your device .yaml to use that.

rlowens, 2023-10-28:
> Good tips here, ill try pulling the mosfets and tracing back.
> 
> Can someone give me an true to esp32 wiring diagram? Is that the auper updated one provided by @Kaldek?

https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v3.png  looks great

Kaldek, 2023-10-31:
> Can you guys confirm that schematics/ratgdo open source_schem_v7.png is accurate? I can share the easyeda layout
> I think the matches the schematic just a bit cleaner.

I like it mate, it's very clean and easy to read; easier than Ki-Cad even.  Can you do a pull request for creation of an easyeda folder with the files in it please.

mikekay1, 2023-10-31:
> I like it mate, it's very clean and easy to read; easier than Ki-Cad even. Can you do a pull request for creation of an easyeda folder with the files in it please.

I also used this schematic but damn thing still aint working... Ill get back to it. If you are wanting for close this issue. It has more than enough revelent info in it.

Thanks for everyone's call to arms!

poldim, 2023-10-31:
> I like it mate, it's very clean and easy to read; easier than Ki-Cad even. Can you do a pull request for creation of an easyeda folder with the files in it please.

I got a bit carried away  😛 

[image]

I'll happily share but I want to build one of these things before I upload it.

Kaldek, 2023-10-31:
> I got a bit carried away 😛
>
> [image]
> I'll happily share but I want to build one of these things before I upload it.

You know, I just **knew** someone who really liked laying out boards would come along.  This is quite the PCB.

Kaldek, 2023-10-31:
> I also used this schematic but damn thing still aint working... Ill get back to it. If you are wanting for close this issue. It has more than enough revelent info in it.
>
> Thanks for everyone's call [...]

@mikekay1 do you want to try with 2n7000 if not already, or an RFP30N06LE?  Another guy just tried ten different IRLB8721 and none of them worked, which is...very weird.
https://github.com/Kaldek/rat-ratgdo/issues/5#issuecomment-1786406085

I am also very curious what your ESPHome console log is saying.  Is it seeing status updates (RX working) but can't send commands (TX not working), or getting lots of errors like "Collision detected"?

sanjay900, 2023-10-31:
> > > > Can you guys confirm that schematics/ratgdo open source_schem_v7.png is accurate? I can share the easyeda layout
> > > > I think the matches the schematic just a bit cleaner.
> > > 
> > > 
> > > I like it mate, it's very clean and easy to read; easier than Ki-Cad even. Can you do a pull request for creation of an easyeda folder with the files in it please.
> > 
> > 
> > I also used this schematic but damn thing still aint working... Ill get back to it. If you are wanting for close this issue. It has more than enough revelent info in it.
> > Thanks for everyone's call to arms!
> 
> @mikekay1 do you want to try with 2n7000 if not already, or an RFP30N06LE? Another guy just tried ten different IRLB8721 and none of them worked, which is...very weird. [#5 (comment)](https://github.com/Kaldek/rat-ratgdo/issues/5#issuecomment-1786406085)
> 
> I am also very curious what your ESPHome console log is saying. Is it seeing status updates (RX working) but can't send commands (TX not working), or getting lots of errors like "Collision detected"?

I myself used two 2n7000s because they didn't have IRLB8721s in stock at my local electronics shop

closed 2023-10-31 by Kaldek

mikekay1, 2023-11-01:
Im still trying to solve this, sorry guys thought I had it figured out! Thanks for all the comments, Ive double checked the circuit a million times and rebuilt countless times. My remote on the wall just shuts off. I have a model with batter backup do I need to bypass that for this to work? Are my grounds getting messed up? I wish I could send a picture of the board juat really hard to see anywl yall available on discord. This is driving me nuts!

I know I have the red wire correct, obstruction sensors connected. Even swapped to dual 2n7000 just to see if I could make it the same as Kaldeks protype board, might try and make it exactly! The same just for shits and giggles...