Kaldek/rat-ratgdo #11: Convert all schematics to Ki-Cad

Kaldek, 2023-10-28:
All schematics need to pivot from Fritzing to Ki-Cad as Fritzing is killing me with its routing bugs which I'm often not catching. 🤬

mikekay1, 2023-10-28:
Oh dude, totally just found a few errors in my breadboard:

[image]

these are a lifesaver -- but I'm still buying kaldek a brewski.. rage against the ratgdo, if I could buy one I would. just release the schematics dude :[

kudos to you man! doing great work here --

mikekay1, 2023-10-28:
to add, I'll help with drawing out some of the stuff soon as I get mine going...

bjhiltbrand, 2023-10-28:
I just started a PR with my stab at a KiCAD rendition.

https://github.com/Kaldek/rat-ratgdo/pull/13

Kaldek, 2023-10-29:
Accepted pull request to add the Ki-Cad files as they are in their own folder structure.  Time for me to open/review these in Ki-Cad, and then we can close this issue.

Kaldek, 2023-10-29:
@bjhiltbrand let's remove all of the power supply circuitry from the design; I was a bit cart-before-the-horse on this and it turns out there's not enough power available.  Other than that these look like we can use as the basic schematic for the repo.

bjhiltbrand, 2023-10-29:
Sounds good. I have removed those components, you want me to do another PR?

Kaldek, 2023-10-29:
> Sounds good. I have removed those components, you want me to do another PR?

Yep, hit it.

Kaldek, 2023-10-30:
@bjhiltbrand I'm going to learn how to use Ki-Cad shortly (promise!) but can you remove J2 from the schematic?   Unless it's there for other reasons of course.

bjhiltbrand, 2023-10-30:
@Kaldek, I put the 2 terminal blocks on there to give people options. In KiCAD, I usually design the schematic first, and then have the program put the component footprints from the schematic on the PCB view when developing a PCB. So, both of these terminal blocks link pins together, so that when someone builds this board, they can choose to use a terminal block with a 5mm pitch between pins (typical screw terminal), or a 2.5mm pitch (typical phoenix or dupont connector). I can remove one of the terminals from the schematic, but leave the footprint on the PCB design, if that would help clear things up?

Kaldek, 2023-10-31:
>  I can remove one of the terminals from the schematic, but leave the footprint on the PCB design, if that would help clear things up?

Yes it's probably worth doing that given there's already been some struggles with folks getting their prototypes up.  If needed we can work on basic vs. advanced schematics for the different audiences.   I honestly expected only the super technical folks to even consider trying this but I guess the backlog of ratgdo boards is making people desperate.

bjhiltbrand, 2023-10-31:
Alright. I removed the terminal block from the schematic, but not from the board design. I also changed the folder structure to have a self-contained kicad project for each variation on the board (D1 Mini - ESP32 vs. ESP8266). I have a couple more variations for bare ESP 8266 (12F) modules and ESP32 (WROOM) modules, but they're not ready to merge yet. I need to brush up one what pull-up/pull-down resistors I need and where they need to be placed first.

mikekay1, 2023-10-31:
Please shared the esp32 version when you have a chance!

bjhiltbrand, 2023-10-31:
> Please shared the esp32 version when you have a chance!

It's in PR right now. Once that's approved, it will be rolled in to the project. I think the pins are correct, but I'm not positive. I left the traces in place between the ESP8266 and ESP32 boards, so in theory, you should be able to use the same board for whichever D1Mini you choose. If the pin assignments need to be different, we will need to change that in the schematic.

Kaldek, 2023-11-01:
Closing as these now exist and can be improved/developed as needed.

closed 2023-11-01 by Kaldek