Kaldek/rat-ratgdo #12: I have a ESP32-Cam MB module, can I use RATgdo on it?

LazaroFilm, 2023-10-28:
I have this spare ESP32-CAM from an abandoned project. Can it be used for this? I wonder if I could even get the camera to work at the same time. That would be great to. Heck on the door when closing.

Kaldek, 2023-10-28:
Mate as long as you can push the code to it from the [ESPHome Web Installer](https://ratgdo.github.io/esphome-ratgdo/) you should be fine.  They're all the same microcontroller at the back end.

Kaldek, 2023-10-28:
Confirmed that ESP32-CAM will work if ESPHome will flash it.

rlowens, 2023-10-28:
Since [the ESP32-Cam board](https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/) doesn't have the default GPIOs exposed that the ratgdo firmware expects (GPIOs 16, 21, 23), you will need to customize the firmware.

Most ESP32 GPIOs are usable for all functions, but `[you should refer to this page](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/)` to confirm what each GPIO can do.

You need to install the ESPHome Addon, then start with this yaml: https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v2board_esp32_d1_mini.yaml

Then you will want to change the board and pins. Comparing to this page https://devices.esphome.io/devices/ESP32-CAM I think you will want to use:
board: esp32dev

Then you need to change the pins used to valid ones for your board. I don't know if you can use the pins that are connected to the camera, so I'd skip those. If you aren't going to be using the SD card slot, the article says you can use those pins for your own use, so I think this might work:

  uart_tx_pin: GPIO12
  uart_rx_pin: GPIO13
  input_obst_pin: GPIO15
  status_door_pin: GPIO14
  status_obstruction_pin: GPIO2
  dry_contact_open_pin: GPIO4
  dry_contact_close_pin: GPIO16
  dry_contact_light_pin: GPIO3

As the article says, you need to use GPIO3 and GPIO1 for flashing the firmware with FTDI adapter since it doesn't have a USB chip on-board. You might want to upload a bare-bones ESPHome firmware first, then you can work on getting it perfect after you have it connected to WiFi and can upload firmware OTA. This is the one I start with on ESP32 boards https://github.com/rlowens/esphome/blob/main/zzstub-esp32.yaml

Kaldek, 2023-10-29:
Thanks @rlowens I didn't go into enough detail on differences with the ESP32-Cam module.  :-/

Kaldek, 2023-10-29:
Will close this issue in a couple of days if no further updates or concerns.

closed 2023-10-31 by Kaldek