Kaldek/rat-ratgdo #14: Using red wire for power.md

aasorokighb, 2023-10-28:
"The input capacitor of the voltage regulator module (usually rated 100uF) will be replaced with a 1000uF Low-ESR capacitor." Why would you put buffer cap on high voltage side? It is shorted by bus driver anyway. Should you put it on 5V side and use 5v to 3.3v on board regulator for driving board? Sorry, I gust ordered my stuff. It will take a few weeks till it arrives.

Kaldek, 2023-10-28:
It doesn't really matter now anyway as the voltage regulator and ESP can't be powered by the GDO's red wire.  I'm parking this for now.

Kaldek, 2023-10-28:
Issue closed as prototype of power supply using red wire current parked due to current draw too high.

closed 2023-10-28 by Kaldek

CrazyCoder, 2023-10-28:
@Kaldek Do you think it may work for Genie using the obstruction sensor wire that has ~12V?

Kaldek, 2023-10-28:
> @Kaldek Do you think it may work for Genie using the obstruction sensor wire that has ~12V?

If you can find a voltage source that can take the current, absolutely.

CrazyCoder, 2023-10-28:
I mean the existing sensors that are already installed and wired. They have 12V on them, but I'm not sure what extra current their power source can provide. I'm afraid to damage the Genie by trying to pull too much current for ESP32 via the buck converter.

CrazyCoder, 2023-10-28:
I gave it a try and it didn't work. Even 0.3A current is too much for the obstruction sensor, and its light goes off when I plug my test board.

The worst part is that even a voltage divider needed to read the pulses from the sensor with ESP32 seems to affect the sensor (its indicator LED becomes dimmer).

Maybe an optocoupler will work better?

Kaldek, 2023-10-28:
> I gave it a try and it didn't work. Even 0.3A current is too much for the obstruction sensor, and its light goes off when I plug my test board.

Yeah it's a bit of a bummer, but I suppose there's no real issue with using external power via USB.  It's mainly going to be an issue for development of a raw board using an ESP-12F for example.  Arguably any design which removes the USB interface is a bit of a pain though because reflashing the chip becomes annoying unless the reflash can be done OTA (which it generally can), and the initial flash before building using an ESP flashing cradle is also more work than just connecting a D1 Mini board to USB.

rlowens, 2023-10-29:
I plan on opening up my GDO and tapping 3.3v from the board.

alexyao2015, 2024-09-15:
Has anyone tried putting a large capacitor on the input to see if smoothing the power consumption allows the esp to be powered?

Kaldek, 2024-09-16:
> Has anyone tried putting a large capacitor on the input to see if smoothing the power consumption allows the esp to be powered?

It's just too much current.  A capacitor won't help unfortunately.

alexyao2015, 2024-09-16:
That's really surprising considering we are stepping down from 12v. How is it using that much current?

Kaldek, 2024-09-16:
The ESP8266 pulls between 0.11 and 0.2 amps of current.  I suspect the WiFi is what pulls the most, but the point is basically moot; as soon as the ESP8266 tries to power up the voltage collapses.  The ESP8266 won't boot and the wall controller shuts off.