Kaldek/rat-ratgdo #16: Help with setup?

ripnet, 2023-10-29:
While waiting for some MOSFETs to be shipped, I set this up on a breadboard with some BJTs and installed [esphome-ratgdo](https://github.com/ratgdo/esphome-ratgdo) on a leftover [ESP8266](https://www.amazon.com/gp/product/B07HF44GBT).

I can read from the "bus" just fine, it shows the motors in use, the position of the door, etc. But writing isn't working. I checked with an oscilloscope and it appears to be all normal, pulling down the 12V line to 0 when the TX pin is active—nice, clean edges. But the GDO isn't responding to my writes.

This leaves me to believe that there's either an issue with timing (probably not since reading works) or syncing the rolling code.  I'm fully prepared to wait until my FETs arrive and try again, but just thought I'd tinker with this for now.

Can someone explain to me how to set the "Client ID" and the rolling code? What happens when they get out of sync? Any help is appreciated.

P.S. Thank you for putting this project together. I'm also disappointed that in order to remove a dependency on a cloud service, we had a dependency on another person, who ironically has an open-source project.

Kaldek, 2023-10-29:
Hi mate, sorry to do the good old handball here but this is getting into code and only Paul Wieland can help at this time.  I mean, in future we're probably going to start picking away at his code but it's suuuuuper early days.

The downside to **that** is that Paul would likely tell you to buy one his boards rather than assist you with getting a BJT working.  

It's possible that somebody might like to try and get the setup working using BJTs.  But, again, it may require alterations to timing in the code.

Kaldek, 2023-10-29:
Re-opening to let the community reply in case someone has the resources to help.

Kaldek, 2023-10-31:
No response, closing issue.

closed 2023-10-31 by Kaldek