Kaldek/rat-ratgdo #20: Gerber files do not match KiCad files

dpeart, 2023-10-31:
When I look at the provided Gerber files they are not the same as the KiCad files.

The Gerber files have a large capacitor and only have a two pin terminals.

Gerber
[image]

KiCad
[image]

Kaldek, 2023-10-31:
@bjhiltbrand do we need another pull request?

bjhiltbrand, 2023-10-31:
Weird. I am regenerating the gerber files as I make changes. Let me look to see why they don't appear to be updating.

bjhiltbrand, 2023-10-31:
I would assume that the old files didn't get cleared out through the renaming of the project that I did recently, so both sets of gerber files existed for the older project name and the new project. The older files were then the ones chosen in a gerber viewer. I forcibly deleted the old gerber files and recreated the gerber file set. I have submitted a new PR.

Kaldek, 2023-10-31:
@bjhiltbrand no stress here mate.  It is the problem we all face once we submit work to the world.  If you'd prefer not to get too involved just let me know.

closed 2023-10-31 by Kaldek

bjhiltbrand, 2023-10-31:
I'm good, no worries. :)

dpeart, 2023-11-01:
I put an order in for 10 boards.  Built and assembled they are $3.5 each, add a$2.5 for the D1 Mini and I'm done.  All I have to do is solder in the D1 Mini.

Thanks

bjhiltbrand, 2023-11-01:
>  I put an order in for 10 boards.  Built and assembled they are $3.5 each, add a$2.5 for the D1 Mini and I'm done.  All I have to do is solder in the D1 Mini.
> 
> Thanks

Here's hoping they work. I haven't had a chance to order mine yet. :)

pleasanthill17, 2023-11-04:
> I put an order in for 10 boards. Built and assembled they are $3.5 each, add a$2.5 for the D1 Mini and I'm done. All I have to do is solder in the D1 Mini.
> 
> Thanks

I am struggling with Kicad to gerber conversion. Is there anyway you write a few lines what you did with the files to get the gerber files ordered or share your files/project on oshw. Also where did you get it assembled for $3.5. 

Thanks !

dpeart, 2023-11-04:
In KiCad you use file->fabrication outputs.  Output each file option.  Then zip up the directory you saved them in.

I use JLCPCB to make my boards. I've used them before and they are great.

On JLPCB you upload the zip file and it will extract everything about the board.  You can then just order the boards, but I also chose to have them assembled.  There you select that option, and you have to upload the Bill Of Materials and placement files.  This is where I had to make changes.  They give you an example of each, so it is easy to create the file formats they need based on the files KiCad export.

Then verify placement and make your order.  I had to manually move the screw terminal as KiCad calculated the placement incorrectly but it is easy to do on their website.  Everyone gets a $10 of coupon so use that.

dolfelt, 2023-11-09:
@dpeart How did you find the correct terminals in JLC? It doesn't find them from the CPL and when I search none of them are showing up for the 3 terminals. Thoughts?

dpeart, 2023-11-09:
Here is the BOM I used:
`Comment	Designator	Footprint
10k	R1,R2,R3	R_0805_2012Metric
WeMos_D1_mini	U1	WEMOS_D1_mini_light
AO3400A	Q2	SOT-23 AO3400A
2n7002	Q1	SOT-23 2n7002
Screw_Terminal_01x03	J1	TerminalBlock_Phoenix_MDKS-`

bjhiltbrand, 2023-11-09:
Also, you only need to choose 1 option out of the 3 on the board to have populated, if you want. You can also choose any terminal you want with a 5mm, 3.5mm, or 2.54mm pitch. I put them all there to allow options for the various pitches. Here are the different types of connectors I had in mind:

[image]

dolfelt, 2023-11-09:
Thanks for the help! Do the terminals get in the way of the USB that powers the ESP32?

Anyone have an example of a fully assembled PCB?

bjhiltbrand, 2023-11-10:
Ha! I never really thought about it. I just measured 4 USB cables (1 USB-C and 3 Micro USB), and 3 of them were about 18-20mm, with one being ~21mm. The kicad design measures 20mm between the D1 mini and J1. Also, if your USB is on top of your D1 mini, I measure it at 11mm off the board (assuming you are using sockets soldered to the rat-ratgdo and pins soldered to the D1 mini. The blue terminal pictured above (this is the 5mm pitch terminal that would fit in J1) is the one that might have clearance issues from the USB. It measures 10mm off the board. So, it might be a bit tight, but should fit fine.

I don't think anyone has an example yet, as we're all waiting for the slow boat from China right now. :)

bjhiltbrand, 2023-11-10:
While not the same project, this is the board that informed much of my design, and where I have extra parts for building mine when my boards get here. The D1 mini is designed to float over the top of the terminals, as long as the USB port is on the top of the D1 mini (I have seen some that have the port on the bottom).

[image]
[image]

dolfelt, 2023-11-10:
@bjhiltbrand or @dpeart, any idea how to get pin sockets added by JLC/PCBWay? I've tried adding a new footprint in kicad, but that feels like the wrong approach since it updates the pads. If it's not possible I can just solder them myself, but trying to save some time since it's only a few cents more for more parts added.

I've also updated the design to guarantee no conflicts with the USB by rotating the board sideways. Would also allow additional terminals on the other side if it's ever necessary. 
[image]

bjhiltbrand, 2023-11-10:
I'm not sure on how best to only have the sockets populated by a board house. I've never had one build a board for me before, as I prefer to do the soldering myself. I wonder if you can just put that in notes somewhere, that you would like to use 4- 1x10 sockets on the board? I like the redesign. You should commit that as another option in the kicad files folder. I noticed on my design just barely, that the footprint for the ESP32 didn't actually keep out the copper on the ground plane for the antenna. I have fixed that and submitted a new PR for it. Please check your redesign for the same issue before making your board.

dolfelt, 2023-11-10:
Whoops! Definitely already submitted it for build.

bjhiltbrand, 2023-11-10:
I don't know that it will make a difference, now that I think about it. If you have the D1 mini installed using sockets and pins, it will be above the board, and separated enough to have a negligible effect on wifi signal anyway.

dpeart, 2023-11-10:
You have to put the pins into the schematic and set their footprint.  You can always solder them on when you get the board back.  this what I'm going to do.  They came with my D1 mini.

dolfelt, 2023-11-10:
Thanks for all your help! Now I wait for the board and work on HomeKit integration. Doing this all because of the blocked MyQ API. And local is better anyways.

bjhiltbrand, 2023-11-10:
> Thanks for all your help! Now I wait for the board and work on HomeKit integration. Doing this all because of the blocked MyQ API. And local is better anyways.

So, I'm building at least 3 of these - 2 for me and 1 for my dad. I use Home Assistant, so that should be pretty straightforward to integrate. My dad, however, uses HomeKit/HomeBridge. I have been having trouble figuring out if there is an integration ready yet for HomeBridge. What route are you planning on going to integrate yours with HomeKit?

dpeart, 2023-11-10:
I should get my 10 boards on the 22nd.  I have 6 to install and my brother 4.  Hopefully they work well :)

Kaldek, 2023-11-11:
> @bjhiltbrand or @dpeart, any idea how to get pin sockets added by JLC/PCBWay? I've tried adding a new footprint in kicad, but that feels like the wrong approach since it updates the pads. If it's [...]

Do you mind if I use this image on our main README?  It's so pretty.

dolfelt, 2023-11-11:
Not at all @Kaldek!

dolfelt, 2023-11-11:
> What route are you planning on going to integrate yours with HomeKit?

@bjhiltbrand My initial thought (without doing much research) is to fork the [esphome-ratgdo](https://github.com/ratgdo/esphome-ratgdo) repo and see if I can't use the official [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) to add the necessary discovery and functionality.

I saw the [homekit-ratgdo](https://github.com/ratgdo/homekit-ratgdo) repo, but it seems that it's not functional yet, and plans to target the ESP8266. Overall I'd rather stick with ESP32 (and ESPHome) for speed, performance, and flexibility reasons. If you have any ideas, let me know.

dolfelt, 2023-11-17:
@Kaldek @bjhiltbrand Got my PCBs from China yesterday. Exciting!

[image]

Got it all hooked up, only to realize that my jackshaft opener (RJO20) falls into the Security+ 1.0 category and nothing is working. Guess I wait and see if the Sec+ 1.0 update to the ESPHome fork will work. 

Do we know if this board with work with 1.0? Does it match the v2.5 specs like it says in the Readme or just the 2.0 board? Excited to test it out once something is released.

dpeart, 2023-11-17:
Got my boards today as well.

bjhiltbrand, 2023-11-17:
Nice! My components should be here today and my boards should be here tomorrow. I can't wait!

bjhiltbrand, 2023-11-17:
> Also, @bjhiltbrand I'm pretty sure the Black and White labels for the terminals are reversed. 😬

Oh S***. I had that fixed at one point! I must have had it revert. Sorry about that. It looks like it's still good on the D1 mini 8266 version, but the other designs all have them switched. I'll fix that right now.

bjhiltbrand, 2023-11-17:
> Does it match the v2.5 specs like it says in the Readme or just the 2.0 board?

It should match the 2.5 specs.

bjhiltbrand, 2023-11-17:
@dolfelt, can you double check pin D7 on your ESP32-wide board? it should have continuity, but when I was fixing the silkscreen issues, I ran it through design rules checker and there were a bunch of errors. Once of which was that KiCAD didn't show a trace to follow from the schematic to the board. I updated the board from the schematic, and KiCAD brought in a new ESP32 footprint. I deleted and re-created the trace from the terminal, through the resistors, and to the pin, and I don't think it made a difference in the board you had made, but I figured I would let you know. Where I didn't design that variant, I figured I'd ask you for verification, since you did. ;)

dpeart, 2023-11-17:
Mine is with the D1 Mini

rlowens, 2023-11-17:
> > Does it match the v2.5 specs like it says in the Readme or just the 2.0 board?
> 
> It should match the 2.5 specs.

It matches the gpio pins used by the ratgdo v2.5, but the v2.5 does have a few more resistors which we just found out about. https://github.com/Kaldek/rat-ratgdo/issues/44

dolfelt, 2023-11-17:
@rlowens I guess it's back to prototyping! 😄

dolfelt, 2023-11-17:
> @dolfelt, can you double check pin D7 on your ESP32-wide board? it should have continuity, but when I was fixing the silkscreen issues, I ran it through design rules checker and there were a bunch of errors. Once of which was that KiCAD didn't show a trace to follow from the schematic to the board. I updated the board from the schematic, and KiCAD brought in a new ESP32 footprint. I deleted and re-created the trace from the terminal, through the resistors, and to the pin, and I don't think it made a difference in the board you had made, but I figured I would let you know. Where I didn't design that variant, I figured I'd ask you for verification, since you did. ;)
> 
> Also, I gotta say... that GitHub action to build the gerbers is slick!

I’ll take a look. Do you know if there’s a way to do multiple boards with a single schematics? Probably worth trying to keep a single schematic per board vs multiple.

bjhiltbrand, 2023-11-17:
I don't know. I'm not overly well-versed in KiCAD. I googled that question before (since the board layout would be the same for all D1 mini (ESP32 and 8266) modules, but I didn't spend a lot of time on searching and didn't find anything real quick.

dolfelt, 2023-11-18:
I’m not either @bjhiltbrand. Software engineer by trade, but just barely an electronics tinkerer. 😅

bjhiltbrand, 2023-11-18:
> I’m not either @bjhiltbrand. Software engineer by trade, but just barely an electronics tinkerer. 😅

Haha. Same here. QC/DevOps Engineer by trade, but really like the whole open hardware concept.

bjhiltbrand, 2023-11-21:
Got my boards and built it today. Works great powered off the battery backup.
[image]

Kaldek, 2023-11-21:
> I’m not either @bjhiltbrand. Software engineer by trade, but just barely an electronics tinkerer. 😅

Same same.  Well, InfoSec architect but same on the tinkerer level.

dpeart, 2023-11-21:
That is a clean looking board.