Kaldek/rat-ratgdo #22: Can you run Ratgdo MQTT code on this hardware?

davidcoulson, 2023-10-31:
I know the simplest way is just to try it, but wasn't sure if anyone had already had success.

Specifically I am needing to support both Security2.0+ and Security+ 1.0 openers. AFAIK the esphome code doesn't support Security+ 1.0 yet (at least I didn't see an option to pick it when I flast a esp8266 with it).

rlowens, 2023-10-31:
Yes, the ratgdo MQTT firmware works on this schematic.

BUT, I don't know if we've had any confirmation that the Security+ 1.0 software successfully works on this schematic.

Note that this schematic is built off of the ratgdo v2.0 hardware. We haven't seen the ratgdo v2.5 hardware up-close to tell exactly what differences there are.

According to Paul Wieland at https://paulwieland.github.io/ratgdo/09_faq.html the ratdgo 2.5 hardware is required for Security+ 1.0 or Dry Contact control methods:

> Security + 2.0 door openers require ratgdo v2.0 control board or later
> Security + 1.0 & Dry Contact door openers require v2.5 control board or later

Kaldek, 2023-10-31:
> I know the simplest way is just to try it, but wasn't sure if anyone had already had success.
>
> Specifically I am needing to support both Security2.0+ and Security+ 1.0 openers. AFAIK the esphome [...]

Essentially this repo wasn't created to provide a complete trace-accurate schematic, although more than one person has volunteered some.  The best way to look at this repo is a means of getting ratgdo running if you can't wait for shipment of an 'official' board from Paul Wieland.  The best approach is to find out from the official ratgdo repo on what is required for Security+ 1.0 support, and then use this repo to help build a board which has the right circuitry.

closed 2023-10-31 by Kaldek