Kaldek/rat-ratgdo #23: Schematic for Bare ESP8266 Module (ESP-12F) - Review Needed

bjhiltbrand, 2023-11-01:
Hello, I am working on building a schematic for a bare ESP8266 chip, and would like a review of what I have so far. I am trying to figure out if I have the correct resistors in the right places for pull up or pull down usage where needed, as well as the button correctly wired for  shorting to ground in order to program the ESP module when needed. Please take a look and let me know what you think:

[image]

rlowens, 2023-11-01:
I am having trouble finding exactly what bootstrap resistors are actually required. The NodeMCU dev board schematic https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF has pull up resistors on RST, EN, GPIO 0, GPIO 2; and pull down resistor on GPIO 15.
[image]

The [ESP8266 Dev Kit schematic from Esspresif](https://www.espressif.com/sites/default/files/documentation/ESP8266-DevKitS-V1.0_reference_design_0.zip) also has those resistors:
[image]

Note they also have capacitors to keep the RST/EN high.

So, I think some of those resistors could be omitted and rely on the internal resistors, but I think all the dev boards just include the external resistors to be sure.

Here's the power portion from NodeMCU:
[image]

And from Esspresif:
[image]

Here's the switches from NodeMCU:
[image]

And the switches from Esspressif:
[image]

bjhiltbrand, 2023-11-01:
Thank you so much! This is immensely helpful. I think most people use external resistors because the internal ones are a bit "weak" and it's incredibly cheap to add that extra layer of protection. I was originally thinking of using an AMS1117 for power, but I was looking at the spec sheet, and I don't think it will drop from 12v to 3.3v. Looks like most people use another converter to go from 12v to 5v, and then the AMS1117 to go from 5v to 3.3v. I was also thinking of using an XL1509 regulator to do a direct conversion, but that involves a ton of components. I have all those components (from a recent build of a number of OpenSprinklers), but I don't know if that's the best choice for others to use...

Kaldek, 2023-11-02:
> Thank you so much! This is immensely helpful. I think most people use external resistors because the internal ones are a bit "weak" and it's incredibly cheap to add that extra layer of protection. I [...]

On my other project I have which uses an Arduino Nano Every, if I switch to using raw ATmega microcontrollers I'll be designing a PCB which uses an [MPM3610](https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MPM3610GQV-Z/document_id/2090) single-chip voltage regulator.  Basically, there are parts out there we can use which consolidate everything into a nice tiny package, and are designed to work in noisy environments.

dpeart, 2023-11-04:
I'm glad you are doing this.  I was going to, as it would have been cheaper to get the entire thing built.  I was impatient and just bought the D1 Minis.  With this, it will be very easy for anyone to just order whatever number of boards they want.

bjhiltbrand, 2023-11-07:
I have taken a stab at this and just submitted a PR for the bare 8266 version. I opted to just shove a buck converter on the board for voltage regulation, as it already had the circuit laid out, and was cheaper than the MPM3610. I am going to get this board sent to JLCPCB and give it a shot in a couple weeks. :)

dpeart, 2023-11-07:
I should have my boards from JLXPCB with the D1 mini on the 24th.

Kaldek, 2023-11-08:
> I have taken a stab at this and just submitted a PR for the bare 8266 version. I opted to just shove a buck converter on the board for voltage regulation, as it already had the circuit laid out, and [...]

So, funny story....PCBWay want to sponsor some free boards for mentioning them in the project.

bjhiltbrand, 2023-11-08:
> So, funny story....PCBWay want to sponsor some free boards for mentioning them in the project.

Nice! I went with JLCPCB because it was 4 dollars shipped for 5 boards, and PCBWay would have been 10.59 shipped for 5 boards. That being said, there are usually 5 dollar off coupons for PCBWay, but I didn't want to deal with the hassle this time. I've used both, and both are great board houses. Of course, I now fixed an issue with my bare ESP8266 boards (missing silk screen and I'm an idiot and tied the positive to ground on the power converter section) that will require me to make a few "modifications" on the boards that I just ordered when they get here. I might order a second round of boards from them once I verify that these first board will work with the modifications.

bjhiltbrand, 2023-11-08:
Also, as I put out a schematic for the bare ESP8266, I will close this issue.

closed 2023-11-08 by bjhiltbrand

dpeart, 2023-11-09:
Curious why you didn't go with a usb port?  that would have made it super easy to flash with ESP-Home.

bjhiltbrand, 2023-11-09:
> Curious why you didn't go with a usb port? that would have made it super easy to flash with ESP-Home.

I don't know... I have a USB to TTL adaptor, so I figured I would put the header on and be done with it. I'm very amateur in my ability to design PCB boards, so that probably added to the decision. ;) A USB port would be easy to add to the design, though.

bjhiltbrand, 2023-11-09:
Plus, wouldn't I need to add a USB driver chip (like the CH340) to the design?

dpeart, 2023-11-09:
yes it would