Kaldek/rat-ratgdo #25: IRLB8721 is **NOT** suitable

Kaldek, 2023-11-01:
We have discovered that the IRLB8721 does not switch at all due to it needing current for switching.  

All schematics must be updated to use either the 2n7000 or the RFP30N06LE.   

For the attention of all who have submitted content:
@bjhiltbrand 
@poldim 
@rlowens

For the attention of known people with issues:
@rohinrohin
@vaidehi-joshi
@sanjay900
@ccutrer

Massive apologies to all who have been bitten by this problem.

sanjay900, 2023-11-01:
Thankfully i wasn't even able to get my hands on the `IRLB8721` so im already using a `2n7000`

bjhiltbrand, 2023-11-02:
So far, for the KiCAD schematics, I opted for SMD components, so I was using AO3400A mosfets instead of IRLB8721 mosfets. You think the schematics are still good with the AO3400A mosfets? I would assume we are still good, as the On-Resistance vs. Gate Voltage mentioned in the link you posted shows the mosfet not responding until it hits 4-5v, whereas the same graph for the AO3400A shows it responding at 2-3v.

ccutrer, 2023-11-02:
Switching from IRLB8721 to another 2n7000 fixed it. I now have full control. What would the difference be with a RFP30N06LE (I have two more to build)?

knissdesign, 2023-11-02:
I just built this project today, with the now-unsuitable IRLB8721. Can you define what symptoms arise from this? I’ve got the project successfully running but it seems most times I have to send commands twice for the garage door to respond, sometimes with delay. Curious if this IRLB8721 could be my issue? Logs in the web interface specify:

[component:214] | Component preferences took a long time for an operation (0.06 s).
[component:215] | Components should block for at most 20-30ms.

Thank you!

ccutrer, 2023-11-02:
My symptom was I had complete status, but no commands would ever work.

Kaldek, 2023-11-02:
> Switching from IRLB8721 to another 2n7000 fixed it. I now have full control. What would the difference be with a RFP30N06LE (I have two more to build)?

2n7000 requires 3v to fully switch on whereas the RFPO30N06LE only needs 2 volts.  GPIO voltage is 3.3v which is borderline if there is any voltage sag.  So, reliable data transmissions might be an issue using 2n7000.

Kaldek, 2023-11-02:
> I just built this project today, with the now-unsuitable IRLB8721. Can you define what symptoms arise from this? I’ve got the project successfully running but it seems most times I have to send [...]

Yes, it could.  The issue with the IRLB8721 will be that it either does not switch the signal at all or will do it unreliably.  I don't know if your log entries are related but you could check the C++ code for those errors to see what the triggering conditions are.

Kaldek, 2023-11-02:
> So far, for the KiCAD schematics, I opted for SMD components, so I was using AO3400A mosfets instead of IRLB8721 mosfets. You think the schematics are still good with the AO3400A mosfets? I would [...]

They will be fine if you use the AO3400A as that's what the original design from Paul Wieland uses.

CrazyCoder, 2023-11-02:
RFP30n06le is considered obsolete. Maybe find a better replacement option?

Kaldek, 2023-11-02:
> RFP30n06le is considered obsolete. Maybe find a better replacement option?

Hmm, maybe we just stick with the 2n7000?  I've had zero issues with mine.

sanjay900, 2023-11-02:
> I just built this project today, with the now-unsuitable IRLB8721. Can you define what symptoms arise from this? I’ve got the project successfully running but it seems most times I have to send [...]

I get that log as well, i don't actually think it's a real problem, i've seen it on a bunch of ESPhome projects

brettcp, 2023-11-02:
Thank you for this!  I'm fairly new to following schematics but I was fairly confident I had everything right.  I could see the correct status of everything, but I wasn't able to get any response when sending commands.  After swapping the IRLB8721 for a 2N7000, commands are now working as expected.

Ordered some RFP30N06LE already and will move this from a breadboard over to a proper ESP8266 shield.

sanjay900, 2023-11-02:
> > RFP30n06le is considered obsolete. Maybe find a better replacement option?
> 
> Hmm, maybe we just stick with the 2n7000? I've had zero issues with mine.

I also second this, i have no issues with mine either.

knissdesign, 2023-11-02:
> Yes, it could. The issue with the IRLB8721 will be that it either does not switch the signal at all or will do it unreliably. I don't know if your log entries are related but you could check the C++ [...]

Update for anyone else troubleshooting! I replaced the IRLB8721 with a 2n7000 and everything is now working as expected. The IRLB was indeed the cause for my symptoms.

The 2n7000 currently seems to be successful, to tag on to others saying the same.

Kaldek, 2023-11-02:
Well folks, what do we think?  @rlowens shall we just recommend a 2n7000 for through-hole and be done with it?  I haven't had a single issue.

sanjay900, 2023-11-02:
Has anybody here had a problem with the 2n7000?

CrazyCoder, 2023-11-02:
> Has anybody here had a problem with the 2n7000?

I'm not sure how well it will work for the dry contact mode with no security. In my testing, it was not dropping voltage to zero like RFP30N06LE did: https://github.com/Kaldek/rat-ratgdo/issues/5#issuecomment-1786383189.

Switching serial line is one thing, but maybe it will have issues with an old-school control panel and LED when it needs to properly short it to the ground?

I didn't test it on the real device, so it might work just fine.

rlowens, 2023-11-02:
I think we should list several tested and possible parts and what problems they have/might have.

Browsing around digikey for MOSFETS similar to AO3400A (which the ratgdo v2.0 uses) found this:
[FDU6512A](https://www.digikey.com/en/products/detail/rochester-electronics-llc/FDU6512A/11536494) (obsolete but in stock, $0.78/each)
[IRLU3802PBF](https://www.digikey.com/en/products/detail/international-rectifier/IRLU3802PBF/12616676) (active status, $0.48/each)
or [these search results](https://www.digikey.com/en/products/filter/single-fets-mosfets/278?s=N4IgjCBcoGwJxVAYygMwIYBsDOBTANCAPZQDaIALGGABxwDsIAuoQA4AuUIAyuwE4BLAHYBzEAF9CAVnoAGRCBSQMOAsTIgAzPTkUELEBy69BoiYRiyYCpSryESkctrpx9bTpB79hYyeDg6GzQse3UnEHkmfzA3TWDlULVHcgAmWVTAhEIweDBUihAczVTNTWzwEs0aQuLtMHlimE0KVKKQdMypQoMjLxNfCX9raBABABMuAFoGiA8udvYAT1ZcLnRsFHFxIA)
Do any of those look good? I'm not an expert here.

Kaldek, 2023-11-02:
> I think we should list several tested and possible parts and what problems they have/might have.
> Do any of those look good? I'm not an expert here.

Yeah unfortunately neither am I and that's how I got into the IRLB8721 mess.  

I think I would be happy - for through-hole designs anyway - to recommend the 2n7000 as long as I can perform a test where I pull down the 12v line and measure the current flowing through the 2n7000 drain to source.  If it's less than 300 milliamps we should be fine.

brettcp, 2023-11-03:
> Yeah unfortunately neither am I and that's how I got into the IRLB8721 mess.
>
> I think I would be happy - for through-hole designs anyway - to recommend the 2n7000 as long as I can perform a test [...]

I'm interested to hear what the consensus is with the 2n7000.. I built a prototype today using the RFP30N06LE which I had to place on the opposite side of the PCB than the rest of the components due to its larger size (I have limited experience with this stuff so its the best I could do).. I would like to be able to just use another 2n7000 if possible as I'm using a small ESP8266 shield that's not much larger than the ESP itself.  I'm only using the 3 pins to connect to the opener itself (black/red/white).

This has been a fun project so far, definitely learning a lot.  Looking forward to Homekit / Homebridge support (which I saw is in development now).

jsb5151, 2023-11-07:
Came here to mention RFP30N06LE was obsolete as per DigiKey.  Same question as rlowens: any recommended through-hole mount MOSFETs that would work other than the 2N7000?

ccutrer, 2023-11-07:
I can confirm that RFP30N06LE works. And more reliably than my door with a 2N7000. The latter I often have to send commands twice, but with the former it works first time every time.

Kaldek, 2023-11-08:
> I can confirm that RFP30N06LE works. And more reliably than my door with a 2N7000. The latter I often have to send commands twice, but with the former it works first time every time.

Way to complicate my life as I was just about to commit to saying "use a 2n7000" 🤣

jsb5151, 2023-11-08:
Well, I don't know where to ask, so I figured I'd ask here.  Couldn't we use optocouplers for this?  Aren't we simply replicating UART at a higher voltage level, on a single wire?

Mind you, I haven't fully studied the logic on the bus, but I'm doing this exact thing with a Honeywell alarm system at 12V logic level on the return line (has RX on the yellow line, but has both RX/TX on the green line).

Kaldek, 2023-11-08:
> Well, I don't know where to ask, so I figured I'd ask here. Couldn't we use optocouplers for this? Aren't we simply replicating UART at a higher voltage level, on a single wire?
>
> Mind you, I haven't [...]

Well now, that's heading into ratgdo design territory which is a question more for Paul Wieland (if he'll answer it; I've been banned from commenting in his repo due to my efforts here).

As long as you can detect when the red wire is pulled to GND for a means of serial RX, and you can pull the red wire to GND as a means of serial TX, then you can choose to use whatever component you want!

I personally have used Optocouplers to mimic button presses in some personal projects, but I've never tried an optocoupler for switching at 9600bps.  I did ask good ol' ChatGPT 4 about this and the response was "you can try but be aware of issues related to capacitance, bandwidth, the length of the isolation barrier", etc.   

Maybe a better way of putting this is that some of my older projects used optocouplers because I was worried about things that weren't actually a problem and could easily just have used a MOSFET.

ccutrer, 2023-11-08:
https://www.mouser.com/datasheet/2/196/Infineon_IRLB8743_DataSheet_v01_01_EN-3363390.pdf specs look better then the IRLB8721. I don't plan on switching mine out, so someone else wants to try this one.

Kaldek, 2023-11-08:
> https://www.mouser.com/datasheet/2/196/Infineon_IRLB8743_DataSheet_v01_01_EN-3363390.pdf specs look better then the IRLB8721. I don't plan on switching mine out, so does someone else want to try this one?

That's worse!  It's input capacitance is 5 times that of the IRLB8721 and that's what we believe is causing the IRLB8721 to not switch when driven by the ESP8266.

ccutrer, 2023-11-08:
Oh well, like I said, I only barely know what I'm talking about. I was looking at the R(DS) vs. V graph of both. The 8721 resistance doesn't drop significantly until nearly 5V.

Kaldek, 2023-11-08:
The issue here is that the most suitable MOSFETs are the AO3400A as used in the original ratgdo PCB or a BSS138 (very similar).  Both are SMD, not through-hole.  

For TO-92 it's really limited and so far it looks like the 2n7000 but it might depend on the batch you get as to how happily it switches at 3.3 volts.

CrazyCoder, 2023-11-08:
One option for through-hole/breadboard design would be to get the original AO3400A and a [SOT23-3 to DIP adapter](https://www.amazon.com/Chironal-Double-Side-SOT23-3-Adapter-Converter/dp/B07MQBF2DD/).

Kaldek, 2023-11-08:
> One option for through-hole/breadboard design would be to get the original AO3400A and a [SOT23-3 to DIP adapter](https://www.amazon.com/Chironal-Double-Side-SOT23-3-Adapter-Converter/dp/B07MQBF2DD/).

Bugger me, if it's that easy we should just recommend this.

jsb5151, 2023-11-09:
Welp I just got my shipment of RFP30N06LE!  However, I’m wondering if that model can be used for both MOSFETs?  I don’t have any 2N7000s handy.

Kaldek, 2023-11-09:
> Welp I just got my shipment of RFP30N06LE! However, I’m wondering if that model can be used for both MOSFETs? I don’t have any 2N7000s handy.

Can't hurt to try; on RX you're using a 12v trigger to drive the MOSFET on/off and therefore pull 3.3v low or not on the RX pin of the ESP.  If you can see data on the bus, it's working fine.

rlowens, 2023-11-09:
> Can't hurt to try;

Well, can't it? The "Absolute Maximum Rating, Gate to Source Voltage" is +10V for[ the RFP30N06LE ](https://www.mouser.com/datasheet/2/308/rfp30n06le-1195941.pdf) so I think it might not survive 12V.

Kaldek, 2023-11-09:
> > Can't hurt to try;
> 
> Well, can't it? The "Absolute Maximum Rating, Gate to Source Voltage" is +10V for[ the RFP30N06LE ](https://www.mouser.com/datasheet/2/308/rfp30n06le-1195941.pdf) so I think it might not survive 12V.

Ah I missed that.  Too many stats, too little time.  😬

jshatch, 2023-11-09:
> > I can confirm that RFP30N06LE works. And more reliably than my door with a 2N7000. The latter I often have to send commands twice, but with the former it works first time every time.
> 
> Way to complicate my life as I was just about to commit to saying "use a 2n7000" 🤣

Mine using 2x 2N7000's seems to work the first time every time, haven't had to resend any commands.

Kaldek, 2023-11-09:
> Mine using 2x 2N7000's seems to work the first time every time, haven't had to resend any commands.

Yeah same here.  We're going to pivot to recommending the AO3400A mounted to a SOT23 to DIP converter board.  It just simplifies everything as the "recommended option".   Even though the 2n7000 works for many.

bdistin, 2023-11-10:
> Mine using 2x 2N7000's seems to work the first time every time, haven't had to resend any commands.

Just built my first one tonight. Worked first time every time with 2x 2N7000s. Didn't know about this issue, but noticed the pics showed 2x 2N7000s, and yoloed it even though I also bought the other transistor.
[image]

Kaldek, 2023-11-10:
Yeah it's weird.  My ESP8266 is putting out 3.2 volts and seems to switch the 2n7000 reliably.

sanjay900, 2023-11-10:
> Yeah it's weird. My ESP8266 is putting out 3.2 volts and seems to switch the 2n7000 reliably.

Mine is also reliable

jshatch, 2023-11-10:
Well I built my second one and used a RFP30N06LE this time, works about the same except for I get errors on that one that say:
```
00:24:21 | [W] | [component:214] | Component preferences took a long time for an operation (0.07 s).
00:24:21 | [W] | [component:215] | Components should block for at most 20-30ms.
```
I suppressed the errors because otherwise it works perfectly but I wonder if it has something to do with that MOSFET.  Maybe it's a little slower?

ccutrer, 2023-11-10:
> Well I built my second one and used a RFP30N06LE this time, works about the same except for I get errors on that one that say:
>
> ```
> 00:24:21 | [W] | [component:214] | Component preferences took a [...]

I saw that with my first one I built with a 2n7000, but not the RFP30N06LE ones. Go figure. But yeah, I wouldn't worry about it.

jshatch, 2023-11-10:
It's probably internal differences between the two openers I have then. I would check by swapping the ratgdo's but I don't want to... 😁

Kaldek, 2023-11-12:
Recommending sn7000 for through-hole and closing this issue.  SMD recommendations will stick with what Paul Wieland uses.

closed 2023-11-12 by Kaldek

tronikos, 2023-11-16:
FAQ.md still mentions IRLB8721