Kaldek/rat-ratgdo #28: Will this work with the LA412 gate opener?

jaredhobbs, 2023-11-03:
Got this working with my garage door opener, works great! I also have a liftmaster swing gate opener (LA412, manual here: https://support.partner.liftmaster.com/s/article/LA412UL-Installation-Manual) which has Security 2.0. Does anyone know how to use this project to control the gate opener?

rlowens, 2023-11-03:
Reading over https://github.com/PaulWieland/ratgdo/issues/90 it looks like the gate openers don't have a Security+ 2.0 wired button to tap for the info. Reading over your manual I don't see one either. Just wireless.

You could wire an esp to the Single Button Control (page 35) and use that to tell the gate to open/close, but you wouldn't get any feedback on gate status.

jaredhobbs, 2023-11-03:
I'll poke around with a multimeter and see if I can find a security+ 2.0 wire and if that fails, I guess I'll wire up the SBC

closed 2023-11-04 by Kaldek