Kaldek/rat-ratgdo #31: Can see logs and statuses but cannot control the garage door

phutchins, 2023-11-07:
I've built the circuit with an ESP8266 Mini D1 on a breadboard based on the schematic and breadboard picture. I've flashed it with the Esphome ratgdo (https://ratgdo.github.io/esphome-ratgdo/). I can access the web UI and see logs going through if I open/close the garage door with MyQ. The statuses update, like the motion, obstruction and open/closed status.

When I click the toggle door button, light, or door up/down buttons, it doesn't work. In the logs I see 'Toggle door' Pressed. Then the 'Rolling code counter', then saving preferences to flash. A little bit later I'll see a Status: door=CLOSED light=OFF lock=UNLOCKED showing that the door has not moved or changed.

Any ideas on where to start looking? 

It might be hard to see but I've attached a picture of the board. Also attaching a screenshot of some logs showing the end of where I closed the door with the MyQ app, then an attempt at opening it through HomeAssistant.

Garage Door: Chamberlain B4613T
Board: ACEIRMC ESP8266 [[link](https://a.co/d/fldpaNH)]

[image]

[image]

Kaldek, 2023-11-07:
This means your TX line isn't working.   Which board did you click when you flashed it?
[image]

phutchins, 2023-11-07:
Ah, yeah, I clicked the top left one because it said ESP8266. I just re-flashed it and set it up again in HA. After flashing it, I moved it back to the bread board and booted it up. It was seeing updates from the garage door again but still didn't open/close it. I rebooted it and now it is really slow to load the web interface. Might the wiring or some lack of configuration be causing the extreme slowness? I.E. it still hasn't loaded the full UI and its been 5 or so minutes.

phutchins, 2023-11-07:
I added the ESPHome addon to HomeAssistant and validated the node which pulled the config. I verified that I have the wires connected to the correct pins on the board itself (like tx, rx, and obst). Its loading faster now so maybe it was just doing something but still no luck with opening/closing.

Is there something I need to do with getting the current rolling code and syncing? I can't seem to find any helpful info on how to do that but I might be missing something.

phutchins, 2023-11-07:
Could it be an issue with the second transistor? I have the RFP30N06LE MOSFET but switched to using both of the small ones as that was what was in the picture to see if I could get it working.

Kaldek, 2023-11-07:
> Is there something I need to do with getting the current rolling code and syncing? I can't seem to find any helpful info on how to do that but I might be missing something.

Worst-case you need to ask it to Sync using the "Sync" button on the web interface.   

Also make sure your device is a Security+ 2.0 model by checking over on the original ratgdo project repo. 

You can also try swapping your MOSFETs around.  If you get "Collision detected" messages or just can't receive (but can transmit now) then that MOSFET is bad.

phutchins, 2023-11-07:
I've tried the sync button a few times with no luck either. I looked up the garage door opener and confirmed that it is a Security+ 2.0 model. I'll try swapping the MOSFETs. 

Here is the first chunk of config just in case that helps at all...

`INFO Detected timezone 'America/New_York'
substitutions:
  id_prefix: ratgdov25
  friendly_name: ratgdov2.5 ff7723
  uart_tx_pin: D1
  uart_rx_pin: D2
  input_obst_pin: D7
  status_door_pin: D0
  status_obstruction_pin: D8
  dry_contact_open_pin: D5
  dry_contact_close_pin: D6
  dry_contact_light_pin: D3
  name: ratgdov25-ff7723
esphome:
  name: ratgdov25-ff7723
  friendly_name: ratgdov2.5 ff7723
  name_add_mac_suffix: false
  project:
    name: ratgdo.esphome
    version: '2.5'
  build_path: build/ratgdov25-ff7723
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2023.10.6
esp8266:
  board: d1_mini_lite
  restore_from_flash: true
  framework:
    version: 3.0.2
    source: ~3.30002.0
    platform_version: platformio/espressif8266@3.2.0
  early_pin_init: true
  board_flash_mode: dout
preferences:
  flash_write_interval: 5s
external_components:
- source:
    url: https://github.com/ratgdo/esphome-ratgdo
    type: git
  refresh: 1s
  components: all
ratgdo:
- id: ratgdov25
  input_gdo_pin:
    number: 4
    mode:
      input: true
      analog: false
      output: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false
  output_gdo_pin:
    number: 5
    mode:
      output: true
      analog: false
      input: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false
  input_obst_pin:
    number: 13
    mode:
      input: true
      analog: false
      output: false
      open_drain: false
      pullup: false
      pulldown: false
    inverted: false`

phutchins, 2023-11-08:
Swapping the MOSFETs didn't make any difference. Anything I can do to test the outputs/inputs of the circuit to tell if its doing the right things?

phutchins, 2023-11-08:
The red wire going to the garage door opener does both the status updates for things like motion as well as commands from the ESP to open/close the door and such, correct? If thats true, then maybe the transistors aren't doing their job and setting up the line for communication?

Kaldek, 2023-11-08:
> The red wire going to the garage door opener does both the status updates for things like motion as well as commands from the ESP to open/close the door and such, correct? If thats true, then maybe the transistors aren't doing their job and setting up the line for communication?

There's no real setup going on.  The red wire from the GDO is +12v all the time unless something is transmitting (whether that is the GDO itself, the wall controller, or the ratgdo).  Any time something wants to transmit, it does this by pulling the +12v (red wire) to ground in pulses at 9600 bits per second (9600 pulses per second).  The RX MOSFET sees this as switching the MOSFET **off** for each pulse, which the D2 pin on the ESP module sees as brief pulses (at 9600 bps) of +3.3v.

As for the TX side of the ESP, pin D1 is not doing anything unless it wishes to transmit, at which point it will send +3.3v pulses (at 9600 BPS) to the gate of the TX MOSFET.  Each pulse will cause the +12v line (red wire) to get pulled to ground, which all other devices on the wire can detect as transmission data.

If you have an oscilloscope, what you need to do is monitor both pin D1 directly and look for +3.3v pulses when you try to transmit (e.g., open the door, query status, etc), and if that's working you need to monitor the red wire and look for inverted pulses from +12v to ground when you try to transmit.  That will tell you where the problem lies.

Also, try connecting the 3V3 pin of the ESP module to pin D1 and hold it there.  If the MOSFET is working the +12v line (red wire) will be pulled to Ground and your wall controller should switch off.

Kaldek, 2023-11-08:
I'm also starting to wonder if some of the weirdness people are seeing when using the 2n7000 is due to their particular variant of the 2n7000 not switching reliably at 3.3v.  Again, an oscilloscope can validate this.

sanjay900, 2023-11-08:
> I'm also starting to wonder if some of the weirdness people are seeing when using the 2n7000 is due to their particular variant of the 2n7000 not switching reliably at 3.3v. Again, an oscilloscope can validate this.

I even found i had problems when using it on a breadboard, but not when i built a PCB, though i very possibly swapped transistors when i moved to the PCB so it could be that as well.

phutchins, 2023-11-08:
First thing I tried was connecting the 3V3 pin of the ESP to pin D1 which in fact did NOT switch the wall controller off. I assume that this means there is an issue with the setup or MOSFET. 

I'm working through your explanation to see if I can troubleshoot with the oscilloscope. I'll get back to you with my findings there soon! I greatly appreciate the help. I feel like I'm close...

Kaldek, 2023-11-08:
> First thing I tried was connecting the 3V3 pin of the ESP to pin D1 which in fact did NOT switch the wall controller off. I assume that this means there is an issue with the setup or MOSFET.

Yeah sounds like your TX circuit is not pulling the red wire to ground at all.

Kaldek, 2023-11-08:
I made a video on how to test this:
https://www.youtube.com/watch?v=AlZNbD49sbE

phutchins, 2023-11-09:
@Kaldek , thanks so much for this video. I'm not sure what was wrong with my initial setup but I went through it all again, replaced the MOSFETs, tested based on your very helpful description above, and now it works! On my bench I was able to get continuity between the two sides of the tx MOSFET so I hooked it back up and it works!

I'm starting to put it on the pcb so I can have a permanent solution. I'll share pics when it's done!

Kaldek, 2023-11-09:
Awesome news mate!

Kaldek, 2023-11-09:
@phutchins I'll close thise but you can share your pics here when you're done.

closed 2023-11-09 by Kaldek