Kaldek/rat-ratgdo #35: 8500C and 2x 2n7000 MOSFETs doesn't seem to be working.

dorkage, 2023-11-10:
I have a jackshaft 8500C with a Yellow Learn button and 888LM wall panel.

I built a protoboard today following the latest schematics, I had gotten the IRLB8721 before it was know they were not suitable, so I went with 2x 2n7000.

When flashing to native RATGDO opener for Board V2.5 and going with Security+ 1.0, nothing seems to work. If I switch to Security+ 2.0 I see more things in the logs, but no commands work.

With Security+ 1.0
`Connecting to:
Homework...
Using static IP address
Set WiFi output power to: 20.50
.
 WIFI CONNECTED
               IP Address: 10.0.0.173
nb of attempts: 1
IMPROVhttp://10.0.0.173X

WiFi connected
Local IP: 10.0.0.173
SoftAP IP: (IP unset)
Server started
Webserver ready. Open http://GarageDoor.local/ in your browser
                                                              Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: GarageDoor/command/door
lightCommandTopic: GarageDoor/command/light
lockCommandTopic: GarageDoor/command/lock
Using security+ 1.0
Setup Complete
 _____ _____ _____ _____ ____  _____
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.51
IMPROVhttp://10.0.0.173X
                        ### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params:
willTopic: GarageDoor/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [GarageDoor/command/#]
QUEUE MSG SENT [homeassistant/cover/GarageDoor/config]
{
  "~": "GarageDoor",
  "name": "Door",
  "unique_id": "GarageDoor_4C:75:25:38:8E:B1",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "GarageDoor",
    "identifiers": "GarageDoor_4C:75:25:38:8E:B1",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.51",
    "configuration_url": "http://10.0.0.173/"
  }
[... 214 lines cut ...]
rolling code for 62129 10|door1 : 550100663B7DFCDF7FB7F860061A9EF4A96DBE
rolling code for 62129 10|door2 : 550100663B7DFCDF7FB7FA60061A9EF4A97DBE
rolling write successful
QUEUE MSG ARRIVED [GarageDoor/command/door]
open
MQTT: open the door
rolling code for 62129 11|door1 : 5501006529B6914DA4DA6A5132CA52178D3483
rolling code for 62129 11|door2 : 5501006529B6914DA4DA685132CA52178D1483
rolling write successful
QUEUE MSG ARRIVED [GarageDoor/command/door]
stop
MQTT: stop the door
The door is not moving.QUEUE MSG ARRIVED [GarageDoor/command/light]
off
MQTT: turn the light off
rolling code for 62129 12|light : 550100442FFFDDFDF7FF7F4016519F72737DBA
rolling write successful
QUEUE MSG ARRIVED [GarageDoor/command/light]
on
MQTT: turn the light on
rolling code for 62129 13|light : 550100420964B04B2C925921022A5F9189E4B1
rolling write successful
`

Rolling codes seem to do more, but I don't have any action from the opener for any functions. I used RAT-RATGDO V6 for D1 Mini and the native RATGDO firmware. Perhaps I need to change to the ESP Home firmware, but I was under the impression my opener was Security+ 1.0. What should I try next?

mikekay1, 2023-11-10:
I believe the esphome firmware does have security 1.0+

Yellow button ahould mean security 2.0+ double checked the gpio, resistor types and mosfet to make sure everything is correct?

mikekay1, 2023-11-10:
Have you tried syncing the rolling codes? I think those logs mean it communicating.

dorkage, 2023-11-10:
Thanks for the quick reply. 

I double checked it visually a few times, but I'll bring home a Fluke87 this weekend and make sure. 

As far as the version of security plus, the jackshaft openers usually fall under 1.0. Mine garage door will open if I short the remote wires.

Kaldek, 2023-11-10:
Can't really speak for v1.0 at the moment as everybody building a board for a GDO which is Security+ 1.0 doesn't seem to have much luck.

For MOSFET testing, you can follow this video I made.  I would guess in your case if your MOSFET circuit is working, this test will toggle your garage door.

https://youtu.be/AlZNbD49sbE

mikekay1, 2023-11-10:
> Thanks for the quick reply.
> 
> I double checked it visually a few times, but I'll bring home a Fluke87 this weekend and make sure.
> 
> As far as the version of security plus, the jackshaft openers usually fall under 1.0. Mine garage door will open if I short the remote wires.

Ah okay yea, shorting the wires is a def security 1.0 then.

This is all my google fu was able to find.

dorkage, 2023-11-10:
> Can't really speak for v1.0 at the moment as everybody building a board for a GDO which is Security+ 1.0 doesn't seem to have much luck.
>
> For MOSFET testing, you can follow this video I made. I [...]

I'll try this this weekend. I suspect in my case this would cause the door to open.

> Have you tried syncing the rolling codes? I think those logs mean it communicating.

I am not having luck finding out how to sync the rolling codes on either here or the official RATGDO page, any links?

rlowens, 2023-11-10:
> I am not having luck finding out how to sync the rolling codes on either here or the official RATGDO page, any links?

This is from a comment I read:
> Changing the client_id is like starting fresh from the GDO's point of view (it doesn't matter what the rolling code is when you change the client_id, but once you send the first command, the rolling code can only increase from that point on).

> When you change the client_id is normal for the first command to be ignored, but the following ones should start working. A simple test is change the client_id and toggle the light from Home Assistant a couple times and it should toggle the GDO light. If that doesn't work, you have something wrong with the wiring.

So, just change the client_id and give it a try I think.

dorkage, 2023-11-10:
> This is from a comment I read:
>
>
>
> So, just change the client_id and give it a try I think.

Again, I am not finding any information on how to do this.

Kaldek, 2023-11-11:
> Again, I am not finding any information on how to do this.

Apologies mate, but most of us are as much in the dark as you are.  You could try asking on the official ratgdo repo but as soon as Paul finds out you're using a self-made one you may even be banned from posting there.  I've been banned already, but I mean, he's probably pissed that I created an open-source schematic. 😬

Kaldek, 2023-11-14:
Closing due to lack of updates from OP.

closed 2023-11-14 by Kaldek