Kaldek/rat-ratgdo #37: 8266 nodemcu help

nodiaque, 2023-11-10:
Hello everyone,

I got some nodemcu 8266 laying around and I wanted to try this project. While I see it's for the D1 mini, I still flashed both 2.0 and 2.5 version of the 8266 and the flashing work, I also got the gui working. Now, the log are flooded with collision which I guess is because it's not using the right gpio. Right now, nothing is connected to it, it's just flashed and that's it.

My nodemcu have a green board and while flashing both 2.0 and 2.5, what I notice is with 2.0 I have 2 led one on the board while with 2.5 I only have one.

Outside of that, per basic, what GPIO are needed for what (outside of nodemcu I just mean generic)? And if I want to reassign to be useable on my nodemcu, how do I go about changing that?

Thank you

rlowens, 2023-11-10:
> I see it's for the D1 mini

Any ESP8266 or ESP32 will work, this is just the example.

> the log are flooded with collision

Expected behavior if you don't have it connected to a Garage Door Opener.

> with 2.0 I have 2 led one on the board while with 2.5 I only have one.

The ratgdo v2.0 uses pin D4/GPIO2 for TX and the ratgdo v2.5 switched to using pin D1/GPIO5 for TX. The blue LED on the ESP8266 module is on GPIO2. We recommend matching the ratgdo v2.5 pins, but many pins would work if you customize the firmware.

> what GPIO are needed for what

For connecting to the GDO:
D1/GPIO5 uart_tx_pin
D2/GPIO4 uart_rx_pin
D7/GPIO13 input_obst_pin

If you want to use optional inputs for toggle switches:
D5/GPIO15 dry_contact_open_pin
D6/GPIO12 dry_contact_close_pin
D3/GPIO0 dry_contact_light_pin
And the optional binary outputs are:
D0/GPIO16 status_door_pin
D8/GPIO15 status_obstruction_pin

> And if I want to reassign to be useable on my nodemcu, how do I go about changing that?

You load the ESPHome firmware, adopt it into the ESPHome Addon, then edit the device .yaml there to change the settings you want. Since it loads the defaults from a package, you can overwrite those settings by defining them in the device .yaml.

It will load the defaults from https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25board_esp8266_d1_mini_lite.yaml

And you will start with a device .yaml looking something like this:
```
substitutions:
  name: ratgdov2-11999c
  friendly_name: ratgdo
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: iWx8HgETC
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Which you might want to change to something like:
```
substitutions:
  name: ratgdov2-11999c
  friendly_name: Garage Door Opener
  uart_tx_pin: GPIO5
  uart_rx_pin: GPIO4
  input_obst_pin: GPIO13
  status_door_pin: GPIO16
  status_obstruction_pin: GPIO15
  dry_contact_open_pin: GPIO15
  dry_contact_close_pin: GPIO12
  dry_contact_light_pin: GPIO0
esp8266:
  board: nodemcuv2
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: iWx8HgETC
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
web_server:
  auth:
    username: !secret web_username
    password: !secret web_password
```

nodiaque, 2023-11-10:
I guess esphome addon is home assistant right? Cause I'm in openhab, I don't have HA. Right now I used the only flasher from esphome ratgdo page. How would I load the yaml without HA?

Thanks a lot for your help.

rlowens, 2023-11-10:
Yes, the ESPHome Addon is in Home Assistant. I'm not familiar with OpenHAB, but I assume you will need to be using MQTT to connect between OpenHAB and ratgdo. The official mqtt-ratgdo firmware is at https://github.com/ratgdo/mqtt-ratgdo and if you don't need to change any of the pins you could just use that directly as you have already put on your nodemcu. It is open-source so if you needed to change stuff you could get those files and recompile it (but I have no experience with that).

Or if you wanted to learn ESPHome, it doesn't have to be used with Home Assistant and you could install it stand-alone and compile in ESPHome's MQTT support and use that.

Either could work. But the least amount of work is just to use the pins that the mqtt-ratgdo v2.5 firmware expect and use that stock.

nodiaque, 2023-11-10:
Yeah there's not direct integration yet in openhab for esphome but from the community forum, someone is using this exact project since to control his garagedoor with the esphome want. He said the esphome have more feature than the mqtt version.

I'm more wondering how I would put the yaml file on the nodemcu. I see I could spin a docker of esphome so I guess from there?

rlowens, 2023-11-10:
Correct. Set up the esphome docker and use that to load the ESPHome UI page.

nodiaque, 2023-11-10:
AH, ok, and that's from where I can send him the famous .yaml file to configure it. Gotcha. Thank you! It's been about 10 years since I played with my nodemcu so I'm a bit rusted

nodiaque, 2023-11-11:
Last question, where do I wire all of the optional pin to? Like the door status and obstruction status pin? I don't see on the ratgdo website where they go

rlowens, 2023-11-11:
This is a the layout of ratgdo v2.0 we traced:
[image]

The optional inputs all have protection diode like this:
[image]

The optional outputs have AO3400A MOTFETs like the TX line.

But all of that depends on what you are connecting those optional lines to, if anything. The diodes on the inputs are to protect the ESP from high voltages if the driving device has them (like connecting to limit switches that are also connected to an older GDO at the same time). The MOSFETs are also to protect the ESP. But if you are just driving an LED to show when the door is open or obstructed, you can just use a simple resistor for example.

Are you planning on using any of the optional pins?

KWolfe81, 2023-11-12:
The above is helpful. Based on the schematics in the repo, I could not get this working with an 8266 or esp32.  Both installed ESPhome fine, lots of collisions as well.  There are multiple versions of the schematics that are all pretty much the same thing.  It would be helpful if all were removed or at a minimum renamed.  The FET pinouts on the schematics are also incorrect.  And lastly, the schematics in the kicad projects don't match the committed schematic images.  *shrug*.

I love the project and drive behind it all, but a bit of cleanup could really go a long way.

rlowens, 2023-11-12:
> There are multiple versions of the schematics that are all pretty much the same thing.

Yeah, I didn't notice that @Kaldek still had old versions in the schematics folder. I don't know why he is keeping them around. Quite confusing.

> The FET pinouts on the schematics are also incorrect

Which ones are incorrect how? I think we specifically have the Source/Gate/Drain labeled to try to avoid this problem. Note that any but the newest of the schematics should be ignored.

Kaldek, 2023-11-12:
> Yeah, I didn't notice that @Kaldek still had old versions in the schematics folder. I don't know why he is keeping them around. Quite confusing.
>
>
> Which ones are incorrect how? I think we [...]

Oddly enough I'm cleaning these up right now.

KWolfe81, 2023-11-12:
Hey its all good!  I'm thrilled you guys tackled this and am happy to work through it.   Please double check me on all this but....

Referring to: schematics\ratgdo open source ESP-32_schem_v5.png

For the RFP30N06LE: the Drain and Source are backwards
Schematic shows:
[image]

But for TO-220 packages, I'm seeing the following pinout:
[image]

KWolfe81, 2023-11-12:
For the 2N7000: The source and the Gate are switched.

Schematic shows:
[image]

But I see (for TO-92 package):
[image]

KWolfe81, 2023-11-12:
Correction: I think your pin numbers are correct for SOT-23 packages.

rlowens, 2023-11-12:
> Referring to: schematics\ratgdo open source ESP-32_schem_v5.png

Which is why that file was superseded by https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png

Kaldek, 2023-11-12:
> For the 2N7000: The source and the Gate are switched.
> 
> Schematic shows: [image]
> 
> But I see (for TO-92 package): [image]

Don't get me started on my "adventures" with using Fritzing for these diagrams.  I have uploaded the newer versions with erased pin numbers.  Fritzing was assuming TO-220 package format and wouldn't let me edit it.

When I have more time, everything will use the KiCad schematics.

KWolfe81, 2023-11-12:
This is certainly an adventure.  Using a transistor tester, turns out I was sent the wrong part.  I can now toggle the garage door open/closed but status isn't working.  Anyways, thats it for tonight, thanks again.

nodiaque, 2023-11-12:
Wow I wasn't expecting to start a hole discussion, that's nice seeing the project active.

I just come back from the part store and they didn't had the bs170. But he gave me a 2sk583 saying it's the exact equivalent.

I'm gonna try today to put everything on my board and test.

As for why the optional pin. I see a pin to get the status of the door and another of the obstruction sensor. These would be nice to have the information. The other one I was looking at is the one for the light. I have seperate light in my garage that I always wished came on with the one in the opener that is motion detected.

nodiaque, 2023-11-12:
hmmm the schematic I look at doesn't show it like that. Are you looking at v9?

[image]

KWolfe81, 2023-11-12:
Question on the RFP30N06LE: your schematic shows the gate tied directly to the GPIO pin with a 10k pulldown to ground.  In the traced image above, the gate is connected to the gpio via a 10k resistor with no pulldown.  I assume you want the pulldown and your schematic is right, but just calling out a difference.

[image]

[image]

rlowens, 2023-11-12:
> Question on the RFP30N06LE: your schematic shows the gate tied directly to the GPIO pin with a 10k pulldown to ground. In the traced image above, the gate is connected to the gpio via a 10k resistor [...]

Yes, that is a known difference. We don't understand why Paul Wieland wired it like that since a pulldown seems to make more sense.

nodiaque, 2023-11-12:
I've just soldered everything but now I'm a bit loss. By what I understand, there's 3 wire to connect. black to obstruction sensor, red to control and white to ground. But how do I wire the obstruction sensor? I have 2 white wire on my liftmaster. Looking at original 2.5 board connection, there should be wire for the obstruction pin
https://user-images.githubusercontent.com/4663918/276749741-fe82ea10-e8f4-41d6-872f-55eec88d2aab.png

He also seem to connect the ratgdo between the sensor and the panel. I though I was conneting everything directly in the liftmaster?

KWolfe81, 2023-11-12:
@nodiaque I'm using an ESP32 and am looking at "ratgdo open source ESP-32_schem_v6.png"

@rlowens @Kaldek I also had a bad transistor 🤦‍♂️.  Anyways, got it replaced (went with two 2N7000's) and it's all working perfectly now!  Thank you soo much!

nodiaque, 2023-11-12:
> @nodiaque I'm using an ESP32 and am looking at "ratgdo open source ESP-32_schem_v6.png"
> 
> @rlowens @Kaldek I also had a bad transistor 🤦‍♂️. Anyways, got it replaced (went with two 2N7000's) and it's all working perfectly now! Thank you soo much!

ah my bad, since it was a nodemcu thread I assume ;)

rlowens, 2023-11-12:
> He also seem to connect the ratgdo between the sensor and the panel. I though I was connecting everything directly in the liftmaster?

Note the white lines on the ratgdo v2.5 connecting the 2 "Red Ctrl" and 2 "Blk Obst" terminals, indicating they are directly connected. He just added more terminals so you don't have to put 2 wires into the same terminal as much. The ratgdo is wired in parallel with the sensors and control panel, not in series between them and the GDO.

nodiaque, 2023-11-12:
ok. So what I did (I have liftmaster 8550w) is:

red to red, black to black and I took 2 white wire into each of the white and put it into the white on my ratgo.

I fire it and it work, I have the status and it update. But, I cannot open/close. I'm looking to see if it's my MQTT that isn't working (I flashed with mqtt cause openhab doesn't play well yet with esphome).

nodiaque, 2023-11-12:
it seems it doesn't listen to the command, I do see the command in mqtt explorer. I think maybe because I didn't put any prefix

nodiaque, 2023-11-12:
well, just tried putting all the transistor with dupon connection so I can switch between them and I can't get anything to work anymore. I don't even get why the reading of the status isn't working anymore since I didn't touch that.

Kaldek, 2023-11-28:
Closing due to lack of updates.

closed 2023-11-28 by Kaldek