Kaldek/rat-ratgdo #38: RFP30N06LE - discontinued, here's what I was suggested

nodiaque, 2023-11-10:
Hello everyone,

Looking for a RFP30N06LE, I found out by digikey that it's discontinued
https://www.digikey.ca/en/products/detail/onsemi/RFP30N06LE/458769

Digikey offer me these as a substitues

HUF76423P3
STP55NF06
STP36NF06L (back order)

Is any of these a good substitute?

Thank you

Kaldek, 2023-11-11:
> Hello everyone,
> 
> Looking for a RFP30N06LE, I found out by digikey that it's discontinued https://www.digikey.ca/en/products/detail/onsemi/RFP30N06LE/458769
> 
> Digikey offer me these as a substitues
> 
> HUF76423P3 STP55NF06 STP36NF06L (back order)
> 
> Is any of these a good substitute?

You're honestly better off trying a 2n7000 in TO-92 package and seeing how you go.  It's what I'm about to recommend.

nodiaque, 2023-11-11:
ok, I was reading the doc as last resort scenario to use that

Kaldek, 2023-11-11:
> ok, I was reading the doc as last resort scenario to use that

Yeah we're about to wrap this up and change all the recommendations on the main page.  Apologies, it's a really active project right now.

nodiaque, 2023-11-11:
oh, so since I'm starting building very soon, can you "spoil" me the change so I get to the new version?

whattheschnell, 2023-11-11:
any of the BS170 in an equivalent package should work.  Its a better choice than RFP30N06LE to begin with for this application.  Much lower CISS, similar Vgs(th) to the 2n7000, and even cheaper.

Kaldek, 2023-11-12:
> oh, so since I'm starting building very soon, can you "spoil" me the change so I get to the new version?

Main page updated.  I'm just recommending 2x 2n7000 for through-hole designs.  SMD obviouly will stick with Paul Wieland's chosen components (2n7002 for RX and AO3400A for TX).

closed 2023-11-12 by Kaldek