Kaldek/rat-ratgdo #41: Issue with 2N7000

ripnet, 2023-11-13:
I wanted to provide feedback on three GDOs that I've installed.

The first has a [050DCTWF](https://www.chamberlain.com/receiver-logic-board-dc-wifi/p/050DCTWF) logic board, and this works well two 2n7000's.

The other two have [050DCTBLK](https://www.liftmaster.com/logic-board-bluetooth-enabled-lock/p/050DCTBLK) logic boards, and will not work when using a 2n7000 on the transmitter. I ended up using [IRLB8721PbF](https://www.digikey.com/en/products/detail/infineon-technologies/IRLB8721PBF/2127670) FETs on those, and they are working.

I'm not entirely sure why the logic board would make the difference with how my ESPs would switch transistors.

I know this isn't an issue per se, but I thought the community might benefit from my experience. Apologies if this isn't the forum for this.

Kaldek, 2023-11-14:
Weird.  As far as we know, the IRLB8721's weren't working at all for others.

This might well be why Paul Wieland uses an AO3400A as it might bridge the gap between the two extremes.

TaylorTWBrown, 2023-11-14:
Very interesting. Thanks for the insight - I'm having trouble getting my transmission to work as well, and I have the same chamberlain. 

However, I'm also only using 1k resistors while I wait for higher values to arrive.

Can you comment on the factors that make some npn transistors more suitable than others for this application?

Kaldek, 2023-11-14:
> Can you comment on the factors that make some npn transistors more suitable than others for this application? Essentially, we've built a half-duplex logic voltage shifter, right?

The ESP8266 needs to switch the red wire to ground via a 3.3v trigger to the gate (for TX).  Therefore a MOSFET that can switch based on that voltage and can handle the +12v Drain to Source is needed.

For RX, a MOSFET is needed that can switch off when the +12v gets pulled to ground.  So unless there is data being transmitted, the RX MOSFET is always on, which pulls the RX pin of the ESP8266 to ground (Drain to Source).   

Those are the requirements.

It is a half duplex voltage shifter but it's inverted both for RX and TX.

closed 2023-11-15 by Kaldek