Kaldek/rat-ratgdo #43: Sec+1 and two 2N7000 not working

tronikos, 2023-11-15:
See `https://github.com/PaulWieland/ratgdo/issues/243` for my issue including equipment and logs.

I have checked the circuit several times and it seems correct.

Steps I have done to troubleshoot:

Multimeter at continuity mode between red and white wires: D of Q2 (or G of Q1) and S of Q2 (or S of Q1) no sound while ratgdo is booting. After a few unsuccessful attempts of “Looking for security+ 1.0 wall panel...” it resulted in “No wall panel detected. Switching to emulation mode.” and then there was some rhythmic chirping sound on the multimeter. This indicates to me there was data transmitted over the line. I think the logs (see linked issue above) indicate that too but the status seems to always be reported as open. Applying 3.3V to D1 (G of Q2) there was a continuous sound on the multimeter meaning the 2N7000 Q2 used for TX was successful in pulling its D to the ground and the door opened.

Multimeter at continuity mode between D2 (D of Q1) and GND (S of Q1): high frequency chirping sound while ratgdo firmware was running. Applying 3.3V to red wire (G of Q1) continuous sound. This suggests 2N7000 Q1 is also working? I tried this on my bench. Is it worth trying it again connected to the GDO which has 16V instead of 3.3V?

Multimeter at continuity mode between D1 (G of Q2) and GND (S of Q2): continuous sound while booting and then low frequency chirping sound after “No wall panel detected. Switching to emulation mode.”.

Do you have any suggestions how to troubleshoot this further? Do my tests confirm wiring is correct and the two 2N7000 are working as expected? Is there any point trying a different transistor?

Kaldek, 2023-11-15:
Hi mate, we're lacking someone who has stated they have a working open source board using a Security+ 1.0 system.  Without that, and given the code is maintained by Paul Wieland, we're unable to help.

Your best bet is to buy a board from Paul and wait for delivery.  If there are issues after that which are related to the code, he will likely assist.

Kaldek, 2023-11-28:
Closing due to lack of updates.

closed 2023-11-28 by Kaldek