Kaldek/rat-ratgdo #44: Now have a v2.5 in hand for references

foreverimagining, 2023-11-16:
It finally arrived!

Alright, this may be a bit of information overload, but I'd rather provide more than enough starting out and therefore, limit the need for further questions. I broke out my DSLR to get some decent macro photos of the various components, traces, and vias because it really is surprisingly small in person. I have compiled them with smaller references marking approximately where on the board front and back I think the pictures are in order to make review a little easier. I can provide more, if necessary; I just need to know what parts need focusing on.

Let me know if you want any testing done, and I am also tempted/willing to desolder the terminals to get a look at what's underneath them. I'll be doing some soldering, anyways, for this project. I figure I'll leave off installing it for at least a few more days, partially because I don't know if I'm even tall enough to plug the little bugger in.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Kaldek, 2023-11-16:
Well that's a lot to take in.   I imagine it will mostly help the folks struggling with 1.0+ door openers and a board they made themselves.

Reall then the question is what's the exact circuit tracing for RX and TX.  Let me see if I can give you some questions of tests to run for validation.

Kaldek, 2023-11-16:
@rlowens this is super interesting. 

What do you think about the TX logic.  Is that a 10k pulldown on the Drain of the TX MOSFET you think?

Kaldek, 2023-11-16:
@foreverimagining the best thing to do here is to determine the path between the Red wire pin and the D1 pin, and then the red wire pin and the D2 pin.

Our current **understanding** of the logic is that the wiring goes:

RX:
- Red wire --> Gate of "7002" MOSFET, with no resistor in series
- Drain of 7002 MOSFET --> Pin D2, with no resistor in series
- Source of 7002 MOSFET --> GND, with no resistor in series

TX:
- Red wire --> Drain of X0YL MOSFET (no resistor in series)
- Gate of X0YL MOSFET --> Pin D1  (no resistor in series)
- Pin D1 --> GND (10k resistor in series)
- Source of X0YL MOSFET --> GND (no resistor in series)

If you can measure those paths with a multimeter and confirm or correct this assessment it will help a lot.  We're interested in whether there is a resistance measurement where we expect one (and what the value is, in ohms), and whether there is a resistance measuremenr where we **don't** expect one (and what the value is, in ohms).

foreverimagining, 2023-11-16:
Sorry it took so long to get back. I had a nightmare of a time taking readings in the scant space between the mosfets and the connectors, so I ended up desoldering the connectors on that side, which was it's own adventure. Anyways, I have some answers, and because I don't entirely know what I'm doing and am using this as an opportunity to learn, we're going to play show your work. You can just skip to the bottom unless the results look weird.

Firstly, based on my ESP 8266 D1 mini boards, Pins D1 and D2 are the third and fourth down as marked in this picture
[image]
so those are the pins I measured with. Next, I had my multimeter set on 200Ω for all the measurements. Some of them gave readings that seem awfully low (not that I know what they should be), and some of them gave no reading and remained at OL. I made sure to try multiple times for each measurement to make sure I wasn't just missing a pad. I also had a couple of readings that seemed consistent at up to 0.8Ω, but as a last check before posting, I tried again and they came down to 0.6Ω. 

The results are as follows with a diagram to ensure I was orienting everything correctly and measuring on the right mosfets:

  **RX:**
	  • _Red wire --> Gate of "7002" MOSFET, with no resistor in series_    : OL/No reading
	  • _Drain of 7002 MOSFET --> Pin D2 (4R), with no resistor in series_  : 0.6Ω
	  • _Source of 7002 MOSFET --> GND, with no resistor in series_          : 0.6Ω

  **TX:**
	  • _Red wire --> Drain of X0YL MOSFET (no resistor in series)_             : 0.6Ω
	  • _Gate of X0YL MOSFET --> Pin D1 (3R) (no resistor in series)_          : OL/No reading
	  • _Pin D1 --> GND (10k resistor in series)_                                           : OL/No reading
	  • _Source of X0YL MOSFET --> GND (no resistor in series)_                : 0.6 Ω
[image]

Let me know of anything else, or if I did something wrong.

tronikos, 2023-11-16:
If your meter has continuity mode it will be much easier, see https://www.youtube.com/watch?v=5G622WDZaHg for a random video that explains how to use it

For the RX side we were expecting G to go to the red wire. With the multimeter in continuity mode start touching different points on the board until it beeps, keep a note what component it is, measure its resistance and continue on the other end of that component until you trace the red wire.

Similarly for the TX side try to trace what the circuit is between D1 and Gate of X0YL.

What's the resistance between source and gate of X0YL? We expect 10kΩ. It might be worth measuring resistance between each pair of the SGD pins for both MOSFETs.

foreverimagining, 2023-11-16:
Ok, I definitely made some rookie mistakes because some of them seemed to work at the lower setting and I didn't know what I was looking for.

My multimeter does have a continuity mode. I think part of what tripped me up is that Red Ctrl does not go directly to Gate or Source, but instead shows continuity with multiple resistors. I had poked around a little because I wasn't getting much of anything. Having turned up my resistance setting, Red Ctrl reads 10kΩ at both Gate and Source, but following continuity, I think it actually goes to Source on the 7002 mosfet. I'll do some charting for clarity.

The resistance between Source and Gate of the X0YL mosfet is actually 11kΩ. The three resistors that are labeled OIC are 10kΩ, but the one labeled OIB is 1kΩ.

I'll go through it a bit more and return with more values.

foreverimagining, 2023-11-16:
Ok, I went through and I think I traced all the continuity on this side of the board, but I'm starting to drag, so I may have missed something. Here's a markup of that.
[image]

As for resistance values, here's the whole lot:

**7002 Mosfet**
Red Ctrl to Gate: 10kΩ
Drain to Pin D2: 0.6Ω
Source to GND: 0.6Ω
Red Ctrl to Source: 10kΩ
Red Ctrl to Drain: OL
Gate to Source: 20kΩ
Pin D1 to Source: 10kΩ

**X0YL Mosfet**
Red Ctrl to Drain: 0.6Ω
Gate to Pin D1: 1kΩ
Pin D1 to GND: 10kΩ
Red Ctrl to Source: 10kΩ
Red Ctrl to Gate: 21kΩ
Source to GND: 0.7Ω
Gate to Drain: 21kΩ
Source to Drain: 10kΩ
Gate to Source: 1kΩ

tronikos, 2023-11-16:
This is very helpful! Thanks for all the measurements and the annotated pictures. So the schematics in the repo are missing 3 resistors:
- 10 kΩ between Red Ctrl and GND
- 10 kΩ between Red Ctrl and G of Q1
- 1 kΩ between G of Q2 and pin D1

I'll see if I can find an 1 kΩ resistor to try this out on a Sec+1 GDO.

Kaldek, 2023-11-16:
> This is very helpful! Thanks for all the measurements and the annotated pictures. So the schematics in the repo are missing 3 resistors:
>
> * 10 kΩ between Red Ctrl and GND
> * 10 kΩ between Red Ctrl [...]

So it looks like we have:

- 10k pulldown on Red wire (aka "10k between Red Ctrl and GND")
- 10k in-series resistor between Red Wire and RX MOSFET gate (aka "10k between Red Ctrl and G of Q1")
- 1k in-series resistor for current limiting of switching the TX MOSFET (aka "1k between G of Q2 and pin D1")

As the circuit works **without** these, they are there for removal of floating voltage and to provide further guarantees that the ESP8266 GPIO drivers aren't damaged.  In other words, "best practice".   I'd need to ask one of my EE friends just *how* much they would be recommended.

tronikos, 2023-11-16:
I added the missing resistors on my board but I can still not control my Sec+1 GDO. The only difference is I'm using two 2N7000. Controls do not work but neither does receiving the correct status. In the logs I see it enters emulation mode which polls GDO for current status. The reply is always that the door is open. The fact that I get some communication initiated by ratgdo suggests to me changing the TX MOSFET will likely not help.

@foreverimagining I'm confused by your last measurement
X0YL Mosfet
Gate to Source: 1kΩ
This doesn't make sense based on the other measurements and your picture. Can you measure again? It should be 11kΩ, not 1kΩ.

foreverimagining, 2023-11-16:
>  Can you measure again? It should be 11kΩ, not 1kΩ.
You're right. I measured again, so either I skipped a key when I typed it in or hit the wrong pads when I measured. I'll go back and fix the first comment so as not to confuse anyone who doesn't read the whole conversation.

tronikos, 2023-11-16:
@foreverimagining looking at your pictures again I think you might have shorted red ctrl and blk obst, maybe by accident while you desoldered the connectors on that side? Can you check if there is continuity between them? There shouldn't be. All the components you took measurements from are on the side you connect the red/black/white wires. The circuit for the obstruction sensor should have two 10kΩ resistors. Pin D7 (ESP8266) to Wht GND should be 10kΩ and Pin D7 to Blk Obst should also be 10kΩ.
If that's true then the schematics in this repo are only missing the 1kΩ resistor.

foreverimagining, 2023-11-16:
> I think you might have shorted red ctrl and blk obst, maybe by accident while you desoldered the connectors on that side? Can you check if there is continuity between them? 

Doesn't look like it. I didn't get a beep, and I tried all combinations. For confirmation, I did test red to red and black to black, which beeped as expected.

I actually didn't think to follow the Blk Obst anywhere since the questions had been about Red Ctrl, the pins, and GND. I'll work on that for you now.

foreverimagining, 2023-11-16:
> The circuit for the obstruction sensor should have two 10kΩ resistors. Pin D7 (ESP8266) to Wht GND should be 10kΩ and Pin D7 to Blk Obst should also be 10kΩ.

Before I move on to following the continuity, Pin D7 to GND and Pin D7 to Blk Obst do both measure as 10kΩ.

thayerfox, 2023-11-16:
I've not fully checked, but here is my schematic with the updated from foreverimaging's measurements.  I also have the dry contact and status components from the v2.0 board schematic included.  I seem to have the same number and values of components as the v2.5 ratgdo has (15 total).  

(I'll check it over a little more later, but figured this might help to visualize what was being communicated.)
[image]

tronikos, 2023-11-16:
@thayerfox I think R3 should be connected to the left of R8.

thayerfox, 2023-11-16:
> @thayerfox I think R3 should be connected to the left of R8.

Yep, good catch.

foreverimagining, 2023-11-16:
If it helps any, this is what I've followed so far. I'll have to take off the connectors on that side to get much more.
[image]

tronikos, 2023-11-16:
Thanks. So the obstruction sensor is using 2 10kΩ resistors matching the schematics in this repo. They are on the opposite side which is strange but I guess that's where there was space. And the schematics in this repo are indeed missing the 3 resistors mentioned earlier which unfortunately didn't make a difference controlling a Sec+1 GDO.

Kaldek, 2023-11-17:
> I added the missing resistors on my board but I can still not control my Sec+1 GDO. 

I think for us pioneers of building our own boards, you need to read the source code on how the ratgdo is trying to control Sec+ 1.0 systems.  Really only then is it possible to perform a proper diagnosis and measurement.  As an owner of a Sec+ 2.0 system there's little I can personally do to help diagnose.

tronikos, 2023-11-17:
I've read the code and I've also figured out the very simple circuit of my wall panel that controls my GDO. They do completely different things so it seems my GDO isn't compatible. I've ordered a reed sensor and I'll use either a spare relay I have or the 2N7000 MOSFET to pull the red wire to the ground.

foreverimagining, 2023-11-17:
Here's another information drop. I think I got everything on this side of the board (we'll act like D3 is on this side), but let me know if I missed anything. I decided to number identical components starting from the top of the picture for easier matching, which may be different from peoples' schematics.

[image]

**Resistance Measurements:**
Trigger Light to Cathode 1: 0.6Ω
Anode 1 to Pin D3: 0.6Ω

Trigger Open to Cathode 2: 0.6Ω
Anode 2 to Pin D5: 0.6Ω

Trigger Close to Cathode 3: 0.6Ω
Anode 3 to Pin D6: 0.6Ω

Pin D0 to Mosfet 1 Gate: 10kΩ
Status Door to Mosfet 1 Drain: 0.6Ω
Mosfet 1 Source to GND: 0.6Ω
Mosfet 1 Gate to Drain: OL
Mosfet 1 Gate to Source: OL
Mosfet 1 Source to Drain: OL

Pin D8 to Mosfet 2 Gate: 10kΩ
Status Obst to Mosfet 2 Drain: 0.6Ω
Mosfet 2 Source to GND: 0.6Ω
Mosfet 2 Gate to Drain: OL
Mosfet 2 Gate to Source: OL
Mosfet 2 Source to Drain: OL

thayerfox, 2023-11-17:
Interesting, thanks for that.  So that confirms the following schematic then: 
[image]
Its interesting that a 10K series resistance was used for the gate on the two status outputs.  The esp8266 gpio doesn't need that level of current limiting.  100-1K would have made more sense.  They used a 1K current limiting resistor for the gate of Q2, though they used a 10K for the gate of Q1 in the v2.5.  Perhaps this was to counteract the 12V red control line being unable to draw much current from the GDO.  I can't see them wanting to slow down the switching speed.

I'm assuming they are relying on the internal pullups of the esp for Q3 and Q4.  I probably would have put a 10K pulldown on those if it had been me, as you can't control the gpio state until the uC boots up.

As for Kaldek's question about how much the 1K R8 is needed.... Connecting the ESP 12mA GPIO directly to drive the AO3400A gate isn't something to lose sleep over, its not going to burn anything and I would have likely omitted it from the design.

kaechele, 2023-11-17:
Great timing, as I was just working on this today.
The schematic as posted by @thayerfox works for me. The schematic in the repository as of this morning didn't work for me, I did not see any packets towards my ESP32. However, opening the garage door worked, but controlling the light didn't.
With the schematic as posted by @thayerfox everything works as expected.

dolfelt, 2023-11-17:
@thayerfox do you have a PR or branch with the updated kicad schematics? Looking to update my board designs. 

I’ll test this new schematic with my Sec+ 1.0 Jackshaft once the ESPHome 1.0 support comes out.

thayerfox, 2023-11-18:
> @thayerfox do you have a PR or branch with the updated kicad schematics? Looking to update my board designs.
> 
> I’ll test this new schematic with my Sec+ 1.0 Jackshaft once the ESPHome 1.0 support comes out.

I do, but my board is a little too modified to be suitable for a PR.  (I've added a 6-30V DC-DC to power off the 12V battery in my opener.)  https://github.com/thayerfox/rat-ratgdo/tree/full_featured_ratgdo2.5/kicad_files/D1%20Mini%20-%20ESP8266%20-%20Tubez

(Don't laugh at the tubez part, its an inside joke from long ago at university about a vacuum tube caricature 😆)

dolfelt, 2023-11-18:
Ok, thanks @thayerfox! I can take a stab at creating a more simple version using your schematic. Thoughts on the accuracy of what we're seeing vs the schematics @Kaldek and @bjhiltbrand? Sounds like you verified this working @kaechele. Are you using the Sec+1.0 for your garage opener?

kaechele, 2023-11-18:
My opener is a Security+ 2.0 model.

bjhiltbrand, 2023-11-18:
@thayerfox, why am I here making all these janky board designs, when you are clearly much more gifted in the ways of KiCAD? ;) Mind if I use your KiCAD files to re-work all the board designs/add full-featured variants to this repo?

thayerfox, 2023-11-18:
Sure, use what you want, modify to suit your needs/rename as needed.  I will have more time early next week if you want some extra help as well.

As for the K7805, its what I had laying around and was in stock a jlcpcb.  You could certainly use the 3v3 version and power the esp8266 direct, or via the 3v3 pin on the d1mini.  I was planning to add the 500mA fuse in line with the protection diode as well.  (You bypass the fuse on the d1 mini when powering through the breakout pins)

dolfelt, 2023-11-18:
I have updated just the basic (red, white, black) circuits (ESP32) with the additional resistors and updated my board. It's no longer wide and just mini. I'll get a PR going soon.

[image]

foreverimagining, 2023-11-23:
If everyone is in agreement that @thayerfox's schematic looks good and everything make sense, and unless anyone has any further tests for me to try, I'm thinking I'm going to solder connectors back on my board and try installing it tomorrow. It's been awfully quite in the repository in the last few days, so I'm assuming things are going well, or people are just busy with the holiday.

Kaldek, 2023-11-23:
> It's been awfully quite in the repository in the last few days, so I'm assuming things are going well, or people are just busy with the holiday.

I don't even celebrate Thanksgiving since I'm over here in Australia.  I'm just busy in general!

kaechele, 2023-11-23:
No holiday in Canada either. I can just confirm that my installation is still working great.
Thanks for sharing and the effort you put in.

foreverimagining, 2023-11-23:
I always forget that Thanksgiving isn't particularly internationally pervasive since it's such a big get-together over here. Most holidays are rather optional until we start getting to this part of the year. I'd never remember when Black Friday is without it, though.

> Thanks for sharing and the effort you put in.

Happy to help. It was a good learning experience for me, too.

thayerfox, 2023-11-23:
> If everyone is in agreement that @thayerfox's schematic looks good and everything make sense, and unless anyone has any further tests for me to try, I'm thinking I'm going to solder connectors back [...]

Thanks for your help in taking those measurements.  Solder your connectors back on and go enjoy your local control!

Sudo-Rob, 2023-11-25:
I really don't understand the resistor network at the gate of Q1.  For one, R6 adds a load (however light) to the wireline bus.  (FYI, the wall panel gets its power from the data line.)  Additionally, R7 slows down the speed of the mosfet saturation (due to gate capacitance), which admittedly could be purposeful.  Moreover, there are times when the bus doesn't get pulled fully to ground per [https://github.com/argilo/secplus/issues/5](https://github.com/argilo/secplus/issues/5).  Wouldn't the resistor network aggravate the task of finding a suitable choice of a mosfet when a low state can be above ground?  With a 45k pullup resistor at the input to the ESP module, that's only about 70 nA of drain current at saturation.

It seems to me a voltage divider is needed (similar to the obstruction signal) or perhaps the schematic is wrong.  If there is a voltage divider, the value ratio between the two resistors should 2.5, or thereabouts, depending upon the mosfet.  The question remaining is what values to select.  Too low and it's a load on the wireline bus, and too high it slows down the mosfet.

thayerfox, 2023-11-25:
You can't think of R6/R7 as a voltage divider.  When the mosfet is at steady state, there's no current flowing through the gate. 
 Applying voltage to the gate is effectively just charging a capacitor until the mosfet reaches steady state.  Thus those resistors are not a voltage divider as was described in your link (monitoring the line state with a raspi gpio).  R6 is a pulldown and R7 limits switching current of the fet.  Yes it does delay the gate charge rate, or delay the fet switching time, but I doubt this is why 10k was used.  Its likely just a current limiting resistor to be compliant with the alleged super weak control line.

Sudo-Rob, 2023-11-25:
That's the point.  It isn't a voltage divider, but maybe it should be.

The concern is Q1 might not reach cutoff because, per the data sheet, the minimum threshold voltage for a 2N7002 is 1 volt at a drain current of 250 uA.  At saturation, the voltage for sinking 70 uA can be below 1 volt, which is why I linked the capture of the data bus.  Moreover, performance below 250 uA isn't even provided in the data sheets, just the max leakage at 0 Vgs (5 uA).

Simply put, unless the bus reaches the cutoff range, you'll never pass data.  Use a voltage divider and you can gain some operational margin.  Other options include finding a mosfet having a higher minimum threshold (some have said >2V) or use some other way of interfacing with the data bus.

Sudo-Rob, 2023-11-25:
Here's a visualization of actual data.  This plot of data should be familiar to you:

[image]

Here's an  excerpt of the data:

[image]

Note the voltage of the last data point.  It's 1.44 volts. Its above the minimum point of cutoff for 250 uA of current, let alone 70 uA, which needs to be some amount lower than 1 V (plus some operational margin).  There's a good chance you won't pass the data.

Also look at the sags in the level of the high value.  This is because each successive message is depleting charge from the supercapacitors in the wall panel.  It's the reason you should try to minimize the parasitic load on the data bus.  Once data activity stops, the voltage recovers.

Kaldek, 2023-11-28:
> Here's a visualization of actual data. This plot of data should be familiar to you:

@Sudo-Rob can you clarify if what you're saying is that you don't believe the resistors should be in place between the control wire and the RX MOSFET?

All I can say is that I sure don't have any resistors in use on the control wire side.  To date my prototype board I built has been rock solid.  I can also say this repo wasn't set up to clone Paul Wieland's work; it was set up to provide the necessary functionality for people who cannot or choose not to, buy his board.  Ergo, I will defer to expertise rather than "what Paul did".  

If this means I need to update the schematics to remove components which are causing debate, I'll do that.

Sudo-Rob, 2023-11-28:
I'm a little late to the party, but I'm going down a very different path.  

I am not saying to take out the resistors and make a direct connection.  While that can work, I'm not sure it'll work for all systems out in the field.  I would chose a voltage divider where 2 volts would be divided to produce about 0.75 volts.  That would ensure Q1 achieves cutoff.  Not knowing more about the design of the data bus prevents me from making a guess of the resistor values, but I could model the circuit to see what happens with rise and fall times.  I could start with 82k and 47k. It still would represent a burden on the data line, but a much lower one as compared to 10k. 

FYI, I have built and tested a much different interface that largely avoids what I perceive as issues with Q1.  I consider it still in beta.  It is more complicated as it uses 11 resistors, a small integrated circuit, and a mosfet (for Tx data). That's just for the data and obstruction sensor.

I am not trying to disparage a design that is working well for many people.  It's difficult to argue with success.

thayerfox, 2023-11-28:
You are correct, if we are trying to handle a 1.44V low signal, the 2n7002 will be partially conducting.  Since we have no external pullup on the esp8266 IO line except for the 40-60k internal pullups, its not going to turn off.

There's perhaps a few ways to work around this, add a 1k pullup on the esp side, though this is sensitive, as we are playing with partial conductance (in sim a 1.7V Vgate is trash with a 1k pullup, but a 1.6V Vgate is reasonable... too sensitive to be practical.)

Your idea of a voltage divider is a good one, you'd move R6 to the other side of R7, but it would require some scope work to see what we can get away with for loading  (what is causing the 1.44v offset on low?  is there effectively a 70k pullup on the 12V line?).  Keeping a load resistance of 10K, we could maybe do a 3k/7k voltage divider to get ~0.45Vgate or 5k/5k for 0.75Vgate.

Alternatively, we could go find a different FET with a higher Vgate threshold.  Something like FDN86246 might be suitable.  Its got a Vgs threshold of 2V min.  Its also got a cool name, the POWERTRENCH.  (its a newer technology UMOS/trench fet which is pretty cool, but its expensive.)
[image]

I do notice from those plots that it appears to only be the wall panels which are unable to pull to 0V.  The GDO seems to have no trouble.  Are we even looking at data from the wall panels, or only the data the GDO puts on the bus?  Perhaps that is why the ratgdo 2.5 works today?

Kaldek, 2023-11-28:
Can I ask if you folks (@Sudo-Rob and @thayerfox) are Digital Logic guys?

The reason I ask is that in digital logic, you don't generally need to be fully high or fully low (although you **should** design circuits that do this consistently where possible), and your signal high/low cutoff point is usually somewhere in the middle (e.g. ~2.5 volts for what we commonly refer to as TTL/5v logic).

The point I'm making here is whether this is effort to fix a problem which doesn't exist.  However, that is my assessment of my interpretation of what you have written, nothing more.

Kaldek, 2023-11-29:
> The 2N7002 will likely partially and fully conduct during the red circled parts of the graph, which in turn will pull the GPIO line on the esp below the digital low threshold. The ESP will not see this data and it will appear as if there is no activity on the bus. The 2N7002 will switch during the green data, as it dips well below the 1V Vgs threshold.

Empirically I can say that the 2n7000 I use can see the wall controller data, because it's the wall controller which advertises motion detection.  In ESPHome Web GUI it's presented like so:
[image]

It wouldn't be a huge stretch to write one's own code to just monitor the control wire, either using an oscilloscope on the Gate and Drain side of the RX MOSFET, and couple that with a log of what the GPIO pin on the ESP8266 is seeing.

thayerfox, 2023-11-29:
Do we know for sure if the wall panel motion is being read directly by the ratgdo?  Could the gdo be replying with an acknowledge that it got the motion detection message from the wall plate, and that's actually what the ratgdo is using?  Ive not dug too deep into the ratgdo protocol decoding.

I mean we know these devices are working... If that scope plot above is correct it's just a bit of a mystery for the wall panel signal.  It's possible too that the scope plot had some other issue occuring too.

Sudo-Rob, 2023-11-29:
The FDN86246 is a good find and it's not horribly expensive ($1.29 in unit quantity).

nik-taylor, 2023-12-02:
Can confirm the schematic from @thayerfox worked for me for a liftmaster 8500 (which is one of the strange openers labeled as security + 2.0 and have a yellow learn button, but they use the older wireline protocol from the Security + 1.0 openers).  It needs 2.5 software so I flashed the original MQTT 2.51 software.

mmotley999, 2023-12-04:
> Can confirm the schematic from @thayerfox worked for me for a liftmaster 8500 (which is one of the strange openers labeled as security + 2.0 and have a yellow learn button, but they use the older [...]

I'm almost certain I have the same model (8500), says Sec 2.0+ and has a yellow button and the sticker says it was made in 2016.

In my case the original schematic, with the 2n7000 MOSFETs and only 3 10k resistors (pulldown on TX and voltage divider on the obstruction side), works perfectly for the past 2-3 weeks using ESPHome 2.5 code.   It works consistently for all features including status for motion off the panel, obstruction sensors blocked, light on/off, remote lockout on/off, open/close/status of the door.

thayerfox, 2023-12-13:
My PCBs finally arrived and I was able to test them out.  Working well so far on my Liftmaster 8550-267 opener.  I am using 12V off the battery backup to power the unit.  

I also tested out the status and dry contact IO using a basic esphome config to toggle the outputs etc.  They appear to function as expected.  

[image]

blademan8, 2023-12-21:
> My PCBs finally arrived and I was able to test them out. Working well so far on my Liftmaster 8550-267 opener. I am using 12V off the battery backup to power the unit.
> 
> I also tested out the status and dry contact IO using a basic esphome config to toggle the outputs etc. They appear to function as expected.
> 
> <img alt="image" width="458" src="https://private-user-images.githubusercontent.com/3465633/290056759-f16c0307-77e2-4a1d-aa0f-60591b4e202e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMjIxNTYsIm5iZiI6MTcwMzEyMTg1NiwicGF0aCI6Ii8zNDY1NjMzLzI5MDA1Njc1OS1mMTZjMDMwNy03N2UyLTRhMWQtYWEwZi02MDU5MWI0ZTIwMmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjFUMDEyNDE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDBkNWY1ODNkZjkyMjE3ZDNlNzVlNzJhMzZmOTI2ZWFiOWYzZGY3YTNiYWY1NWYyNWUwZjg2OGViYTIyMzMwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.d1UCCs34dstE9j3cihcaNuG9nwUnGumaIUHS4cpZtbc">

Oh nice! Any chance you can share the kicad for that?

thayerfox, 2023-12-21:
> Oh nice! Any chance you can share the kicad for that?

Sure, https://github.com/thayerfox/rat-ratgdo/tree/full_featured/kicad_files/D1%20Mini%20-%20ESP8266%20-%20Tubez

Kaldek, 2023-12-26:
Closing as this is more accurately going to lead to additional Pull Requests rather than being an Issue.

closed 2023-12-26 by Kaldek

conradopoole, 2024-01-08:
> Interesting, thanks for that. So that confirms the following schematic then: <img alt="image" width="764" src="https://private-user-images.githubusercontent.com/3465633/283920363-bf829a0b-775d-4086-9c3a-6eb77baaf8f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2NzY5OTEsIm5iZiI6MTcwNDY3NjY5MSwicGF0aCI6Ii8zNDY1NjMzLzI4MzkyMDM2My1iZjgyOWEwYi03NzVkLTQwODYtOWMzYS02ZWI3N2JhYWY4ZjAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDhUMDExODExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmY1MGE3ZmU2Yzk4OGZjOTk0YjY2NTk5YWE0YTYyYzkxOTFhNGJhNzVjZWI3ZDg5MWYwYzE5MGNjYTQ0ODJkYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Xs4e7NbO5SMi9WNfv8AhJWbVkH8Qs7GgpznSxL8f2sc"> Its interesting that a 10K series resistance was used for the gate on the two status outputs. The esp8266 gpio doesn't really need that level of current limiting. 100-1K would have made more sense. They used a 1K current limiting resistor for the gate of Q2, though they used a 10K for the gate of Q1 in the v2.5. Perhaps this was to counteract the 12V red control line being unable to draw much current from the GDO. I can't see them wanting to slow down the switching speed.
> 
> I'm assuming they are relying on the internal pullups of the esp for Q3 and Q4. I probably would have put a 10K pulldown/up (depending on desired state) on those if it had been me, as the gpio state is indeterminate until the uC boots up and the onboard pullup/downs are enabled.
> 
> As for Kaldek's question about how much the 1K R8 is needed.... Connecting the ESP 12mA GPIO directly to drive the AO3400A gate isn't something to lose sleep over, its not going to burn anything and I would have likely omitted it from the design.

@thayerfox Did you verify the orientation of the Dry contact diodes D1, D2, and D3? From the MQTT code seems like some of those pins are defined as inputs but all diodes in your schematic are pointing agains the ESP.

[image]
[image]

thayerfox, 2024-01-08:
Yes, going by the ratgdo wiring diagram, those "inputs" pull to ground to actuate.
[image]