Kaldek/rat-ratgdo #46: IMPORTANT!!! - PCB boards/gerber files for all boards except D1 mini - 8266 prior to 11/17/23 have White and Black silkscreen swapped!

bjhiltbrand, 2023-11-17:
I was made aware of, and fixed an issue where the silkscreen that I had placed on most boards in the kicad files folder (the D1 mini - ESP32, ESP32-Wide, and Bare 8266 boards) had the terminal for the white wire (ground) swapped with the terminal for the black wire (obstruction sensors). Please note that if you have had boards manufactured prior to today, that the silkscreen is incorrect, and you will need to make note of this when connecting your wires. I apologize for any inconvenience this causes.

@Kaldek, please review and approve this pull request as soon as you can. Thank you.

https://github.com/Kaldek/rat-ratgdo/pull/45

dolfelt, 2023-11-18:
Thanks for the fix @bjhiltbrand!

Kaldek, 2023-11-20:
Approved and merged but maybe @rlowens did it as my weekend memory is fuzzy.

closed 2023-11-20 by Kaldek