Kaldek/rat-ratgdo #47: I can read the door status, but not control it

dpeart, 2023-11-17:
So I built one of my boards with the D1 mini.
I hooked the red, white and black wires to the GDO.

I can read the state in HA, motor running, obstruction clear or not and it knows if the door is open or closed.

I cannot control the door at all.  No open, close, light on/off.

ideas?

bjhiltbrand, 2023-11-17:
I went through the schematics again and I think I have them right. If not, I sincerely apologize. Have you taken a look at this issue?

https://github.com/Kaldek/rat-ratgdo/issues/31

@Kaldek also has a video that he did linked at the bottom of that issue to do a little troubleshooting. Also, make sure that you flash the ESPHome firmware for the 2.5 board so the pins match up the way they should.

When I get my boards tomorrow, I'll solder it up and have a shot as well.

dpeart, 2023-11-18:
the gnd pin on the D1 Mini is not connected to ground.  It is in my PCB in Ki-CAD, checking with JLCPCB now to see what happened.

dpeart, 2023-11-18:
Sure enough.  When JLCPCB read it in it got messed up.

[image]

dpeart, 2023-11-18:
KiCad
[image]

dpeart, 2023-11-18:
I'll blue wire it and see if that fixes the issue.  I can see that it would be difficult to ground the red wire if your grounds are different

bjhiltbrand, 2023-11-18:
Interesting. Well, now that they messed up, can you get an updated board with a good silkscreen for free? ;) I don't know if anyone here has a solid grasp of KiCAD (you or @dolfelt probably do for sure), but if they do, it might be a good idea to review all the designs to make sure I didn't screw stuff up.

dpeart, 2023-11-18:
KiCad is fine.  Just need to figure out what happened

dpeart, 2023-11-18:
Yep, grounding pin 10 and everything works.  Grounding is as the entire back plane is ground

closed 2023-11-18 by dpeart

maskedpirate, 2023-11-29:
@dpeart - I'm having the same issue of only being able to read status but not control it with my esp32 variant. In my case the obstruction sensor is never updated either; only door, light, rolling code, etc.

I used the kicad PCB files from this repo and exported all the relevant files with the JLCPCB plugin and had them assemble the components. When you say you grounded pin 10 (GND), do you mean you soldered a wire from the GND pin to the GDO WHT GND? If it matters, I'm powering the ESP32 via USB.

Thanks!

dpeart, 2023-11-29:
My pin 1- (GND) was left floating, so I scraped a bit of the green coating away to expose the ground plane.  I then put a drop of solder to connect the pin to the ground plane.