Kaldek/rat-ratgdo #48: What is the current status of this work?

antamy, 2023-11-21:
I just came across all of this the other day, so trying to figure out the current status.    I see that there is discussion re: 2.5 with thayerfox.  Is the intent that that will be folded in to this repo?  I was going to pull the trigger on getting bare ESP8266 boards (since I have a bunch of those processors), but not sure whether a) the schematics in this repo work and b) whether they're all going to be updated to the 2.5 board.

I'm happy to take a crack at creating a new schematic etc for the bare board, just not sure where I should do that.  Please advise!

Kaldek, 2023-11-21:
Hi mate, it's a fairly open project and folks who want to merge new stuff in just ask and generally I approve.  I've been a bit busy to check and make sure there's no status confusion over the last week.

If that lack of time continues I'll change the structure of the repo so folks can ask directly about specific schematics.  The idea was based on the very first high level schematic and a "roll your own" approach.  It's been complicated somewhat by the addition of PCB schematics which can be printed.  I'm totally cool with storing them and working with the providers of those schematics to keep them up to date etc, but it still needs to be viewed as an experts-only repo.

Kaldek, 2023-11-21:
My other advice is to use Github desktop and clone the repo so you can open the schematics in KiCad and confirm if the existing bare ESP8266 PCB schematic works for you.  It's the easiest way to check.

closed 2023-11-21 by Kaldek