Kaldek/rat-ratgdo #51: using a raspberry pi pico w

godmar, 2023-11-27:
This is more of a discussion question, so feel free to close this issue.

It seems like the hardware schematics would be compatible with a RPi Pico W, wouldn't it?  Has anyone tried that? Would a Pico be more likely to work with the existing 12V power if it drew less current? 

I'm also a bit confused about what's open source. Apparently, this project reverse-engineered the ratgdo board and put different software on it.  I understand that the GDO uses some kind of serial protocol that the microcontroller reads - what protocol is this and where is the code for it?  At first glance, I didn't see it in either this repo or in the ratgdo repo - but I may have missed it?  

Thank you.

Kaldek, 2023-11-27:
> I'm also a bit confused about what's open source

Well, the original ratgdo project is itself open source, hence the ESPHome fork.  However the board design was closed source, so this repo is really just an open source design either for using modules (like the Wemos D1 Mini or the ESP 32 WROOM), or using a bare ESP8266.  It gets you the necessary circuitry to interface with the GDO.

rlowens, 2023-11-27:
Yes, this could work on a rp2040 mcu (the mcu in the RPi Pico W), and the esphome firmware might even work without changing much at all since esphome supports rp2040 already.

As far as running off of the 12V power, I don't know.

Re: open source: The software for ratgdo is at https://github.com/ratgdo:
https://github.com/ratgdo/mqtt-ratgdo is the (original) mqtt firmware written in c++ for Arduino I think? it uses https://github.com/ratgdo/secplus (written in c) and https://github.com/ratgdo/espsoftwareserial

https://github.com/ratgdo/esphome-ratgdo is the esphome firmware which uses the above

and the work-in-progress https://github.com/ratgdo/homekit-ratgdo (homekit firmware, not working yet)

This project open-sourced a hardware schematic to interface mcu's running the ratgdo firmware to a GDO without buying the ratgdo hardware sold by Paul Weiland.

closed 2023-11-28 by Kaldek