Kaldek/rat-ratgdo #52: (2)2n7000 with wemos d1 mini no RX on 8500W sec2.0

hobbyist2, 2023-11-27:
Hello.

I have build the circuit with a breadboard and soldering 2n7000.  I also have tried two different wemos minis d1's with ratdgo 2.0 (switched d1 to d4)  and 2.5 .  On the 8500w I connect the control wire from the ratgdo to the red wire going to the wall control, the ground wire to the black going to the wall control and leave the obstruction wire disconnected.  With the web application for the ratdgo I am able to toggle the door, though I dont get the correct door status nor does any other function works like turning on lights.  Am I running into a limitation with the 2n7000?  What troubleshooting steps should I take next?

Kaldek, 2023-11-27:
Your 8500W is the US version of my 3800.  If the only thing you can do is toggle the door, it sounds like you have connected to the red terminal on the GDO rather than the green terminal.

Take a look at the FAQ and make sure you're wired up correctly:
https://github.com/Kaldek/rat-ratgdo/blob/main/FAQ.md

hobbyist2, 2023-11-27:
> Your 8500W is the US version of my 3800. If the only thing you can do is toggle the door, it sounds like you have connected to the red terminal on the GDO rather than the green terminal.
>
> Take a [...]

The red terminal on the 8500w goes to the control panel. The green terminal on the 8500w is for the electronic lock

rlowens, 2023-11-27:
Does your GDO open/close if you short the red and ground wires? If so, it might be using Security+ 1.0 for the wired connection and won't work with the ESPHome firmware (which doesn't support Securit+ 1.0 or Dry Contact methods yet).

hobbyist2, 2023-11-27:
> Does your GDO open/close if you short the red and ground wires? If so, it might be using Security+ 1.0 for the wired connection and won't work with the ESPHome firmware (which doesn't support Securit+ 1.0 or Dry Contact methods yet).

This opener I installed recently. As my old opener had a solution that shorted the red and black, i tried it. It didnt work. I learned this opener is sec 2.0

Kaldek, 2023-11-28:
> Does your GDO open/close if you short the red and ground wires? If so, it might be using Security+ 1.0 for the wired connection and won't work with the ESPHome firmware (which doesn't support Securit+ 1.0 or Dry Contact methods yet).

I had a look at his model online; it's practically identical to mine and definitely Sec+ 2.0.

kaechele, 2023-11-28:
I had similar issues when I first put together my prototype. I could control the doors but not the lights or read any of the information. I replaced the 2N7000s with some others from the set I ordered and it started working. Seems like there may be variances, ESD mishaps or overheating issues while soldering. Not sure.

Kaldek, 2023-11-28:
> > I had similar issues when I first put together my prototype. I could control the doors but not the lights or read any of the data values. I replaced the 2N7000s with some others from the set I ordered and it started working. Seems like there may be variances, ESD mishaps or overheating issues while soldering. Not sure.
> 
> The prototype board and the one I soldered had different 2N7000s from the same batch. So there would have to be four that are/went bad.

Unfortunately as I've said to others, when it gets to this point it's time to break out the oscilloscope.  Quoting from one of my earlier responses to similar issues:

*If you have an oscilloscope, what you need to do is monitor pin D1 directly and look for +3.3v pulses when you try to transmit (e.g., open the door, query status, etc), and if that's working you need to monitor the red wire and look for inverted pulses from +12v to ground when you try to transmit. That will tell you where the problem lies.*

*Also, try connecting the 3V3 pin of the ESP module to pin D1 and hold it there. If the MOSFET is working the +12v line (red wire) will be pulled to Ground and your wall controller should switch off. This will prove the MOSFET circuitry is working.*

mikekay1, 2023-12-06:
I ordered the old **non** 2.5i kit since the new one is integrated with esp 12f and Im speculating it has some order licencing shit integrated now.

rlowens, 2023-12-06:
> speculating it has some order licencing shit integrated now.

What do you mean by that?

I do see the new "v2.52i" version on the purchase page now says it is "C 2023 ratCloud llc"
[image]
Odd name for a company for a device specifically to NOT be in the cloud. But as long as it runs the same firmware already released as open source that doesn't have any cloud connection, it should be fine.

kaechele, 2023-12-06:
> > speculating it has some order licencing shit integrated now.
> 
> What do you mean by that?

I'm interested to know as well. Seems like a wild speculation to make.

> I do see the new "v2.52i" version on the purchase page now says it is "C 2023 ratCloud llc"

I commented regarding this some other place, but I believe the LLC was set up to achieve exactly that: limit Paul's personal liability in this endeavour.
Smart Move. Chamberlain can't be super happy about this and has been known to be litigious on these kind of things in the past.

> Odd name for a company for a device specifically to NOT be in the cloud.

The "rat" in both ratgdo and ratCloud, as well as rat-ratgdo (this repository) stands for "**r**age **a**gainst **t**he". A very fitting name for a device and company with this mission.

> But as long as it runs the same firmware already released as open source that doesn't have any cloud connection, it should be fine.

The board shown above has no component on it that would indicate it contains any DRM protections against just running the open source software on it.

I deem the speculation baseless and until further evidence in support of it can be found, people should not assume that that is what's going on.

bill7531, 2023-12-08:
The issue with 8500w was addressed in the original ratgdo page. It need version 2.5 to work properly.

https://github.com/PaulWieland/ratgdo/wiki

mikekay1, 2023-12-12:
True opensource should just work on the device, I was getting ahead of myself. Was just weird he closed support and is asking for order numbers. Ill let you know when mine comes in, maybe I should have gone with the 25i.,.

Kaldek, 2023-12-22:
> True opensource should just work on the device, I was getting ahead of myself. Was just weird he closed support and is asking for order numbers.

I honestly suspect he's really pissed that this repo exists and had perhaps had people building their own boards asking him for support.  Rightly, he can refuse to offer that support.

closed 2023-12-26 by Kaldek