Kaldek/rat-ratgdo #54: Wemos D1 mini, circuit checked hundreds of times, still wont connect

mikekay1, 2023-11-30:
Hey guys me again resigned the whole thing geound up Chamberlain B970C, with batter backup. Something is going on here an what was a weekend project has developed into an insomnia. Security 2.0+ for sure.

Everything is fine, I have only 4 terminals:
R - 12V here for sure measured about 11.44
W - GND?
W - Obstruction White
B - Obstruction Black

Please someone tell me what I am doing wrong my circuit is perfect juat like working breadboard in the code. 

Loaded this firmware:
esphome-ratgdo/v2board_esp8266_d1_mini_lite.yaml@main

Im like ready to pay and just wait for the backorder except knowing me that shit wont work either when I get it.

Am I missing a fourth wire? 

Please help, please...

foreverimagining, 2023-11-30:
Have you tried the firmware for v2.5?

And if that doesn't work, how does your circuit compare to the more updated schematic here: https://github.com/Kaldek/rat-ratgdo/issues/44#issuecomment-1817084612?

mikekay1, 2023-11-30:
I have not tried v2.5, im trying to keep things simple. Honestly dont understanding why version 2.0 isnt working for me. Looking at the schematic now

mikekay1, 2023-11-30:
Im using 2x2n7000s, using gpio d1,d2 and d7 everything looks good.

mikekay1, 2023-11-30:
Hot damn, let me try that, I went around switching all the wires hope I didnt blow a MOSFET. Am I able to just switch the number in the firmware of esphome and reinstall?

esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main

"V25"

mikekay1, 2023-11-30:
Also side question can I just move GPIO back to D4 and be fine? Sounds like this is the only change?

foreverimagining, 2023-11-30:
> esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main

Looks like that is the correct firmware version.

> Also side question can I just move GPIO back to D4 and be fine? Sounds like this is the only change?

I would think this would be able to work, but they changed pins from D4 to D1 because D4 has to be pulled high at boot or else it will fail. Apparently, this caused some nuisances because it would pull the communication/power line to low, so it may be better not to use D4.

mikekay1, 2023-11-30:
Safe thanks for the reply man!

[image]
[image]
[image]

Still no go but at least no more collision detected errors?

everything in the circuitry is on point, all colors match respective output to GDO for easy tracing. Orange is GND to weemos.

still no status, I tried hitting Sync but nothing -- question about that, do I have to do anything special on the GDO or remote to sync current rolling code or something???

foreverimagining, 2023-11-30:
My experience from the legitimate v2.5 board (with yellow learn button) is that it's pretty plug-n-play. I flashed the firmware from the web installer, and it easily connected to my Home Assistant before I hooked it up to the GDO. Once I got the wiring done and plugged the D1 mini in again, everything showed up like expected, and it could control the door right off the bat. The only snag was a loose GND wire to one of the obstruction sensors that prevented me from closing the door until I switched it to another terminal on the lever connector.

Unless I'm not seeing something, I think you're missing these two resistors (besides the 1k, which may not be strictly required): 
[image]
I don't know how important they are, but it might be worth a try to add them.

If it is a rolling code issue, apparently you can just change the Client ID and try toggling the light a couple of times to see if it'll work. If it doesn't, it's supposed to be a wiring problem.

mikekay1, 2023-11-30:
I will look thank you

slhar, 2023-12-02:
I built 3 of these with 2x 2n7000. Only one (my second build) has worked, and on that one TX and the beam detection works, so I can only open/close the door and have safety features. 

I double and triple checked my translation of the schematic to the boards. I know why my first attempt failed. I am not sure why TX works, and RX does not when looking at the latest schematic the 7002 is still used on RX and TX uses the different mosfet.

slhar, 2023-12-03:
Second one was built on proto board. The third I breadboarded because, it must've been me. <shrug> not sure what the real difference is honestly.

mikekay1, 2023-12-05:
Welp! its been fun. I officially have determined that either I have some very weird edge case scenario or we are using components that fail for some but work for others. I have literally tried _every possible combination_ of this schematic including the new 2.5 version in hopes that maybe it was something to do with my GDO. everything is perfect, it will not detect any status no matter what I do. I think I will throw in my hat and buy one from paul, although with my luck that shit wont work either. Something tells me my battery backup on GDO is messing with me, I don't think its a bad ground. the short D1 to ground test works... oscilloscope showed everything in working order. I almost want to try this on a friends GDO but also dont have the time. This was supposed to make me look like a super smart guy, and instead I went into a rage trying to make this work. 

Deep down chamberlain sucks, smart things that are not local suck, its too often I am smart device TV for example and the app store or web-os is dead in 2 years, or the netflix/spotify(etc) app stops getting updates. I digress! Thank you all so much for the help @Kaldek you can close the ticket man, your work was very much appreciated! It was fun learning a few things from this project.

Cheers!

Kaldek, 2023-12-05:
I'm sorry it didn't work mate.  Please also buy one of Paul's boards and let us know if it works on the original.  It's useful information, and it may help someone else.

closed 2023-12-05 by Kaldek

mikekay1, 2023-12-05:
100 percent I will

slhar, 2023-12-05:
I ordered a ratgdo today as well. I have a lift master with a battery
backup too.

hobbyist2, 2023-12-06:
I have similar issues and made a post about it.  There is no resolution yet.  My 8500W also has a battery backup
https://github.com/Kaldek/rat-ratgdo/issues/52