Kaldek/rat-ratgdo #60: Wall panel occasionally resets

kaechele, 2023-12-07:
I am currently running a board conforming to this schematic:
[image]

Since attaching the board, occasionally, my wall panel will reset (i.e. the clock will have reset to 12:00 and is blinking on the display).

I have not been able to figure out when exactly it happens and what is happening on the bus at this time, but generally my garage door is not used a lot, maybe once a day if at all. I see a reset about once every 3-4 days.
Also, we generally control the door using wireless remotes or the wall panel.

My guess it that this is likely due to the parasitic load on the bus introduced by the rat-ratgdo. I remember seeing it mentioned in other issues (such as https://github.com/Kaldek/rat-ratgdo/issues/44#issuecomment-1826256679).

Opening this issue to track if this is happening to others as well or whether it's specific to my setup.

My setup:
- GDO: Liftmaster 8550-267
- Wall Panel: Liftmaster 880LMW
- M5Stack AtomS3 Lite with ESP32 as rat-ratgdo controller

[image]
<p float="left">
[image]
[image]
<p>

foreverimagining, 2023-12-07:
Well, Paul Wieland has removed the issues from his original Ratgdo repository, so I can't point you to the issue where people had noted this activity, nor do I remember if there was a solution or if it was another "that's just to be expected". Your wall panel looks to be about the right model, if memory serves, and I think the reasoning was something along the lines of the Ratgdo occasionally pulling the power line to ground long enough that the capacitors in the wall panel run out so it resets like any other power outage. 

I don't really remember or know enough, but I can tell you that it's almost certainly not because of your particular setup.

YBonline, 2023-12-07:
ratgdo v2.0 used the D4 pin which is pulled high on boot of the esp8266, causing the clock to reset whenever the esp8266 restarts. ratgdo v2.5 solves this issue by using D1 instead. I assume the pins that the esp32 pulls high on startup are different, but you need to make sure you use pins that are not on bootup. Even if your not rebooting manually, the board automatically reboots if it does not connect to Home Assistant or MQTT or something for a period of time

kaechele, 2023-12-08:
Good point. I'm using GPIO5 for TX, GPIO6 for RX and GPIO7 for obstruction sensor input. GPIO5 is a strapping pin on the ESP32, apparently. I can move it to GPIO8.

hacsjalano, 2023-12-09:
I've been struggling with this too.  With the ESPHome firmware, this would happen a *lot* - like almost every 3rd-5th time I opened or closed the garage door.  It *definitely* happened whenever I would restart HomeAssistant for any updates.  With the official firmware, it does not happen at all with routine opening/closing of the garage door, but with HomeAssistant reboots, it would happen maybe once every 2-3 times, and even then it was never a clock-reset.  It would just ask to press the wall button to re-sync the wall button the opener.

I have the official version 2.0 board from Paul Wieland.

At this point I'm just waiting to see if the ESPHome firmware will have the ability to read the battery charge level for the backup battery.  If that happens, I can remove the wall module completely because the only reason I have it is so I can see the battery level inside the garage door opener.

kaechele, 2023-12-09:
Actually, now that you say that, my wall panel is also the non-W 880LM. Fixed my initial post.
And to clarify: I'm, so far, not seeing any issues with having to reactivate the wall panel (i.e. the rolling codes going out of sync). It just loses it's time setting but comes back up fully functional.

hacsjalano, 2023-12-09:
I think we're seeing the exact same behavior.  To summarize, for anyone else reading later, I've tried both the ESPHome firmware and the official Ratgdo MQTT firmware using the V2.0 official Ratgdo hardware.

ESPHome behavior:

- 880LM loses clock time periodically, sometimes just with opening and closing the garage door a lot in a short period of time.  Definitely when rebooting Home Assistant instance.
- 880LM does *not* need to be re-sync'd to garage door, ever.  When 880LM loses clock, the garage door button opens and closes the door immediately with no need to press any additional buttons.

Official Ratgdo MQTT firmware:
- 880LM never loses clock time, regardless of Home Assistant rebooting or any number of times the garage door is opened or closed.
- 880LM does occasionally request that the I press the garage door open/close button as though it needs to re-sync with the garage door opener.  I only have to press this button once without any other actions needed on the garage door opener.  This happens very infrequently, but most often it happens when I reboot Home Assistant.

With both firmwares, Home Assistant and the Ratgdo appear to function perfectly completely unaware anything is wrong.  The sensors all work in Home Assistant and I can always open and close the door immediately from Home Assistant.  It's only the 880LM that experiences issues.

kaechele, 2023-12-12:
Related: https://github.com/ratgdo/esphome-ratgdo/issues/79

foreverimagining, 2023-12-12:
Must have missed it when I was looking, but that's the issue I had seen before. Looks like there's a combination of factors that are possibly contributing to the problem. Based on discussions in the Home Assistant forum, there are at least a few models of wall panels that aren't really compatible with Ratgdo, to varying extents.

hacsjalano, 2023-12-12:
That related issue was created by me, so apparently we're the only ones with this problem so far!  :)

closed 2023-12-26 by Kaldek