Kaldek/rat-ratgdo #63: Thanks! The latest KiCAD ESP8266 with additional resistors works!

russgmoore, 2023-12-19:
I have purchased two units from Paul but wanted to build one. I did 2-3 prototypes and breadboards on the 3 10k resistor configuration and two 2N7002 mosfet. I then tried this as I read that the voltages might need to be tuned down.
kicad_files/D1 Mini - ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf
Note I still used two 2N7002 mosfet plus 5 10k resistors and 1 1k resistor as per the diagram.

This build worked like a charm and has been running for a bit. 

So this isn't an issue but I think this schematic is probably better.  At least it was for me.

Ronan-cdn, 2023-12-20:
Thanks for posting that. I was able to use an old esp8266 nodemcu 'Lolin' and it works!
I followed the schematic you mentioned, 
https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf

The latest esphome firmware worked well, but I'm using this with homebridge, so prefer the mqtt apis... I loaded the latest ratgdo firmware on it and also works. Now I can control the garage door from an iphone or apple carplay.

rlowens, 2023-12-20:
> but I'm using this with homebridge, so prefer the mqtt apis

Note that the https://github.com/ratgdo/homekit-ratgdo firmware looks to be in a useable state now, in case you didn't see it.

closed 2023-12-26 by Kaldek