Kaldek/rat-ratgdo #64: ESPHome automated OTA - does it work on ESP32?

Kaldek, 2023-12-22:
Hi folks, for anyone running on an ESP32, can you tell me if the ESPHome-based OTA updates (from the ESPHome Add-On in Home Assistant) work?  OTA does not work on ESP8266 due to the amount of flash storage taken up by the combination of ESPHome Core and ratgdo.

gcormier, 2023-12-22:
I don't have any type of ratgdo but can tell you from experience yes, OTA will work from the HA addon for ESP32's. (Another reason nobody should be using the 8266 anymore with the cheap price of something like an ESP32-S3-MINI.)

Kaldek, 2023-12-23:
> I don't have any type of ratgdo but can tell you from experience yes, OTA will work from the HA addon for ESP32's. (Another reason nobody should be using the 8266 anymore with the cheap price of something like an ESP32-S3-MINI.)

OTA works on all of my other 32 ESPHome devices (30 of which are ESP8266, the other two are BK7231).  It's just the additional ratgdo code which pushes the ESP8266 over the edge.  I mostly agree that the ESP8266 is a bit "old hat" but I don't go so far as to say nobody should be using them!

Kaldek, 2023-12-24:
I have confirmed this works on ESP32!

closed 2023-12-24 by Kaldek