Kaldek/rat-ratgdo #66: Error: ESP does not have enough space to store OTA file.

williamcwallis, 2023-12-28:
I’m receiving the following error on my setup when trying to update the firmware.

[image]

Per the instructions on the ESPHome vs Native Ratgdo page, I added the d1_mini board code in my yaml as seen below. 

`esp8266:
  board: d1_mini
substitutions:
  name: ratgdov25-cda4c6
  friendly_name: ratgdov2.5 cda4c6
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 8Q3/b+egS7TntfSZieYFKZ23pWSlfXugj5P1sS6fhQE=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`

I am using the following board. https://a.co/d/6W7dEID

Apologies if there is something that I’m missing rather than this being an actual issue. Thanks.

CodeFinder2, 2023-12-28:
See https://github.com/Kaldek/rat-ratgdo/issues/64

You can use an ESP32.

Kaldek, 2023-12-28:
Hi mate, did you first reflash it using USB?  The first time you update the settings to "d1_mini" it still doesn't know about that capability, so you must first do a "manual" install via USB.  After that, it knows about the settings and space available.

Kaldek, 2023-12-28:
For @CodeFinder2 - I deleted your comment only because it missed the context of how I've provided instrructions for ESP8266 boards with 4MB flash.

williamcwallis, 2023-12-29:
I just flashed it manually via USB. That did that trick. No OTA problems now. Thanks for the help!

closed 2023-12-29 by williamcwallis