Kaldek/rat-ratgdo #68: Adding support for ESP32-H2

conradopoole, 2023-12-29:
This is an FYI on something I have been working on. I wanted to create a Matter Thread version of the ratGDO boards, so I decided to design one in KiCad. Granted we do not have a matter thread version of ratGDO but I plan on trying to port it using esp-matter and creating it as a LOCK + Light matter device.

Based on the ESP32-H2 chip and more precisely on the ESP32-H2-MINI-1 module. While working at it I also decided to add a buck converter so that it can be powered from either the USB-C port or an external power source between 12V and 24V, like the backup battery present in some GDOs.

Let me know what you think, I will be sending this revision to production soon, so hope to have the schematics and PCB validated soon.

Dimensions are around 48mm by 49mm. Screenshot of the 3D render below.

[image]

Kaldek, 2023-12-30:
Looks good mate.  I have no exposure to Matter so I can't really comment on that side of it but happy to host everything here!

rlowens, 2023-12-30:
My only feedback is the silkscreen naming of "RatGDO", since that might be trademarked by Paul.

linkedupbits, 2024-01-13:
I thought I heard there were stability issues with using wifi and thread/matter on the same chip at the same time - most espressif examples use the h2 as a network co-processor?
I have no empirical basis for this though…

conradopoole, 2024-01-14:
> I thought I heard there were stability issues with using wifi and thread/matter on the same chip at the same time - most espressif examples use the h2 as a network co-processor?
> 
> I have no empirical basis for this though…

Well, the ESP32-H2 doesn't have WiFi at all, but even if we used the ESP32-C6 that has both WiFi 6 and Thread, you would use one of the two radios not both at the same time.

Kaldek, 2024-01-22:
Please come back once this is tested and working so we can raise a Pull Request.

closed 2024-01-22 by Kaldek