Kaldek/rat-ratgdo #71: Permissions

CarlKnox, 2024-01-10:
I'd like to push a branch to be able to create a PR. However,
remote: Permission to Kaldek/rat-ratgdo.git denied to CarlKnox.

CarlKnox, 2024-01-10:
[image]
[image]
Just want to PR an alternate proto board layout.

CarlKnox, 2024-01-10:
Yellow is the 3v3 signals, Though I question using a voltage divider for D7.
I added a header pin for D8 since I'm using homekit-ratgdo. Not using it yet, but it's there when I need it.
```
#define STATUS_OBST_PIN         D8  // output for obstruction status
```

Kaldek, 2024-01-11:
Hi mate, anyone cane submit a Pull Request.  Did you fork the repository first?

CarlKnox, 2024-01-11:
Yep,
Tried to push before forking.
Tried PR after branching.
I can try again when I have more time, but thought you might take advantage of the photos in the meantime.

Kaldek, 2024-01-11:
Yeah, generally it's Fork, Edit, Push.  Noting that I myself are a Github n00b.

closed 2024-01-22 by Kaldek