Kaldek/rat-ratgdo #74: Support for other terminals for switches?

youaintnobody, 2024-01-15:
[image]

Hi, just wondering is there support for the other terminals as shown on the photo? Ability to add wall buttons for open/close/stop/light?

Kaldek, 2024-01-15:
There's a few schematics that have been provided that have them, yeah.  Check the schematics that have been uploaded.

youaintnobody, 2024-01-15:
I'll check again, probably my electronics n00bness causing me to miss the details... I'll keep looking.

youaintnobody, 2024-01-16:
I had a couple of [esp8266](https://www.amazon.com.au/DIGISHUO-ESP8266-NodeMCU-Development-Micropython/dp/B097D6Y12K) boards laying around and my old [freetronics experimenters kit](https://www.freetronics.com.au/products/experimenters-kit-for-arduino) so I attempted building my own rat-ratgdo board.

I only have the Red + White wire connected as I don't currently have an obstruction sensor installed. I have a [Merlin MT100EVO](https://p.widencdn.net/efatlq/MT100EVO-Tiltmaster-Manual-114D4626G).

It works but it's intermittent? I can control what's in Esphome/HA but eventual stops responding to commands. Door position will eventually show open/close status but not while it's moving.
1. Door
2. Lights
3. Lock remotes
4. Toggle

I can see Openings status but Motor status is always Not running.

Does anything look incorrect with my board?
[image]

I've also noticed when I have the board connected the GDO diag lights starts flashing. I'll reconnect it again to see what the diag codes are.
[image]

Lastly, I've ordered Wemos D1 mini's just to make sure I have an identical board and maybe new 2n7000's. Just want to make sure I've wired it up correctly before I solder to a board for permanent install.

foreverimagining, 2024-01-17:
Looks like you're missing some resistors. Ironically, the obstruction sensor part looks correct to me, and that's the part that probably isn't necessary if you're not going to connect it for a lack of obstruction sensors.

It looks like you're missing both 10k resistors on the RX line from D2, and there is a 1k resistor between D1 and the Gate on the mosfet on the TX line on the v2.5 boards.

I will be curious when you get this running better, though, because some people have had an issue with the RATGDO command to close the door not working when they don't have obstruction sensors. I've been suspecting that they are having the problem because it looks like they have the Blk/Obst wire connected to the GDO.

youaintnobody, 2024-01-17:
Thanks for taking a look and helping @foreverimagining.

I followed this originally: [ratgdo open source D1 Mini_schem_v6.png](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v6.png).

I've now revised it as per this one: [RatGDO-OpenSource-D1Mini-ESP8266.pdf](https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf).

Seems to still be the same. The Door control still seems to be flakey, but the Toggle door seems to work fine. Still no change in Motor status.

With the door control, am I meant to be able to open the door then stop it where ever and then press open or close? Currently if I use the slider or the open/close button if I stop it the door control stops responding. The only thing that seems to continue to work is Toggle door.

foreverimagining, 2024-01-17:
That second one is a more up-to-date schematic based on my v2.5 board, so that's a better one to go with. The first one does work for some, but the second one seems to work for more people when the earlier schematics fail.

> am I meant to be able to open the door then stop it where ever and then press open or close?

I just tested mine, and yes, in the ESPHome version of the firmware, you are supposed to be able to do all of that. 

It's a bit weird that you can do a command (toggle) but not the others. It at least means something is getting through on the TX line, but maybe your mosfet isn't matched well enough to be able to respond reliably. It would probably be a good idea to try different mosfets. On the SMD boards, the TX line has a AO3400A mosfet, though some people have gotten 2N7002s to work just fine. The 2N7002s seem to be a bit finicky, and some people have had to try several before they manage to find ones that work.

One other thing that might be affecting you depending on where you are, someone in Denver is reporting trouble with their original RATGDO in the HA forum that they think is due to the frigid cold temperatures. Mine seems to be fine on this comparatively warm day here in Illinois, but I also have a decently insulated, south-facing garage.

youaintnobody, 2024-01-18:
Just an update, I flashed using 2.5 instead of 2.5i. Traced all my wiring. The door control works for opening and stop, but not close. Toggle button always works. Still no motor status, does this work for people?

I have obstruction sensors... in the box, just need to find time to install it to see if that fixes my issue.

youaintnobody, 2024-01-18:
I've installed my obstruction sensor. I can now close via the Door control/cover.

Lastly before I migrate from breadboard to soldering to PCB, with the dry contact for open/close/light, does the D# pins just short to ground or do I do I need resistors between D# pins <-->button<-->10K ohm<--->ground?

foreverimagining, 2024-01-19:
Sounds like some good progress. And I believe that answers that on how well it works without obstruction sensors. It just seems weird that the RATGDO can't close the door without them. Must be something with the coded command.

> Still no motor status, does this work for people?

My motor status does update, though I'm personally only interested in the open/close statuses.

> with the dry contact for open/close/light, does the D# pins just short to ground or do I do I need resistors between D# pins <-->button<-->10K ohm<--->ground?

If you're not planning to use the Dry Contact side (and I don't know why you would need it if the Serial Line Control side works), you can just leave it off. If you do want to have it, I'm not entirely clear on your question because the open/close/light circuits are simply a diode between the microcontroller pins and the dry contact switch terminals. The dry contact statuses have a mosfet each and a 10k resistor between the microcontroller pins and Gates.

Maybe this'll help. 
[image]
This is from my almost-finished breakdown of the second schematic from above. It makes it a little easier to focus on the various particular parts. The labels on the left sides are going to the microcontroller pins and the labels on the right sides are all going to the corresponding terminals.

youaintnobody, 2024-01-19:
Thanks for your help mate!

I currently have a [Two Button Wireless Wall Button](https://www.gomerlin.com.au/product/two-channel-wireless-wall-button-2/) and would like to remove it and replace with a wired wall control/button. That's why I wanted to know how the Open/Close/Light switch/buttons would be wired up.

Am I correct in my understanding, that if I wire it up I'd have a wall switch to control open/close and light on/off?
Sorry mate, I'm a n00b here so just to validate:
Example: D# PIN---BUTTON---(1N414) |>|---BOARD GND?

Kaldek, 2024-01-19:
> Sounds like some good progress. And I believe that answers that on how well it works without obstruction sensors. It just seems weird that the RATGDO can't close the door without them. Must be something with the coded command.

Do we just want to fork the esphome-ratgdo repo and start adding our own code alterations for these edge cases?  I mean, I kinda already did it for 2MB ESP8266 modules.

rlowens, 2024-01-19:
> Am I correct in my understanding, that if I wire it up I'd have a wall switch to control open/close and light on/off?
> Sorry mate, I'm a n00b here so just to validate:
> Example: D# PIN---BUTTON---(1N414) |>|---BOARD GND?

The dry contact input pins on the actual ratgdo boards are wired like this:
D# PIN---(1N414) |>|---BUTTON---BOARD GND

The protection diodes are included in case you want to connect them for use with limit switches for sensing when the door is open/closed for dry-contact control method on older non-Security+ 1/2 GDOs (where the GDO would also be using those same limit switches and you don't want the ratgdo's D# PINs exposed to the GDO's higher voltage). Since that isn't what you are using them for, you don't need the diodes (or additional resistors as you initially asked). You can just do
D# PIN---BUTTON---BOARD GND

Note that you usually have to use 3 buttons (Open, Close, Light Toggle), but you can change your device .yaml in ESPHome to change the Open input pin to a toggle instead of Open Only, if you only want to use 2 buttons. You would connect the Dry Contact Close directly to ground, and use just the Dry Contact Open. And add this at the bottom of your device .yaml file:

```
#custom modifications using https://esphome.io/guides/configuration-types.html#extend
binary_sensor:
- id: !extend "${id_prefix}_dry_contact_open"
  on_press:
    - if:
        condition:
          binary_sensor.is_on: ${id_prefix}_dry_contact_close
        then:
          - cover.toggle: ${id_prefix}_garage_door

```

foreverimagining, 2024-01-19:
> Thanks for your help mate!
>
> I currently have a [Two Button Wireless Wall Button](https://www.gomerlin.com.au/product/two-channel-wireless-wall-button-2/) and would like to remove it and replace with [...]

I'm really not much less of a n00b than you; I've just been keeping an eye on the community for longer. XD

Ok, I think I get it now. You've got what amounts to another remote but want to replace it with what amounts to a simple, wired wall panel.

I think that would fall under the wiring in this diagram, but depending on your use case, may require some fiddling.
[image]

My understanding would be that it should be `D# pin <--> 1N414 Diode >| <--> terminal if you want easy connections <--> button <--> GND`. Where the fiddling might come in is how many buttons you want to deal with, 2 or 3, and then, possibly the kind of buttons.

foreverimagining, 2024-01-19:
> Do we just want to fork the esphome-ratgdo repo and start adding our own code alterations for these edge cases? I mean, I kinda already did it for 2MB ESP8266 modules.

I almost want to say yes, but I lean heavily no at this point for a few reasons.

1. This particular issue is actually already known in the ESPHome issue 146 since mid December, but they don't seem to be finding a simple solution. It sounds like the command used for closing the door won't trigger correctly without the obstruction sensors installed. Since the GDOs work without them using the remotes (and wireless wall buttons), there must be some different code to get around it, unless they're just using a command to toggle, which does still work. Which is now what I'm betting on.
2. This repository allows for a wide variety of use cases and could therefore require just as many forks to match them all. Since that is the case, it makes more sense for people to make their own forks as needed for different board styles.
3. There haven't been much in the way of software problems that I've noticed so far that are common and exclusive to the RAT-RATGDO community. This may change over time, though, and if the main repositories still aren't interested in making peace and working with us on support, then we could always revisit the decision.

That being said, your 2MB ESP8266 fork may be a good starting point for some of the cases in 2. to modify from, so maybe it will become useful/necessary to do so.

rlowens, 2024-01-20:
> Only thing I'm curious about is if I can add a DHT11 to it without causing issues.

I haven't added one myself but it shouldn't pose any problem. I don't the ratgdo firmware is so close to using all the RAM or CPU speed that adding more sensors would stop it.

closed 2024-01-22 by Kaldek

youaintnobody, 2024-01-29:
Anyone have any issues after rebooting the device and the door status is open when it's physically closed? Sometimes query status doesn't output the status.

CodeFinder2, 2024-03-23:
> Anyone have any issues after rebooting the device and the door status is open when it's physically closed? Sometimes query status doesn't output the status.

Yes, I am having synchronization problems after reboots too (often). Although I don't see *any* status after restarting (the ESP). Restarting the GDO (always?) fixed it. Still not completely happy with my current circuit / draft (and I am somehow blaming the used BS170 MOSFETs for this).

And I am also completely unable to control my GDO via open/close commands. Toggle is always working fine. Status works if it synced correctly once (see above). From what I've read in this thread, it seems to be REQUIRED to have an obstruction sensor? I don't have one (and haven't planned to install one up until now) but it seems I need one to make this work... 😐 Do you know whether [something like this](https://a.aliexpress.com/_EHyuaFF) (cheaper, from AliExpress) would work?

I am using a SHT3X temp/humidity and a VEML7700 ambient light sensor (both via I2C) on a ESP32 (via ESPHome). That's working flawlessly. (I can share a picture if you need one.) Thus, a DHT11 should work as well. 🙂