Kaldek/rat-ratgdo #77: Insteon/X10 Interference

powersra, 2024-01-17:
I have a love/hate relationship with electronics.  Thanks to the info on this repo (great job, btw) and a pack of transistors from Amazon I whipped up 5 of these bad boys for the garage doors on the house and shop.

Only 1 of 5 worked consistently... darn it.  Now, I'm no soldering expert but that had me shaking my head.  Was getting lots of collision warnings in the logs, dropped packets, very intermittent comms/status updates with ESPHome, etc.  First I thought the wiring to the gdos was suspect, so I painstakingly cleaned that all up, wire nuts and pigtails and nice and neat.  Same deal, 1 of 5.  Swapped a bunch of boards and their power supplies around, same result, 1 of 5.  Now the working board was still working but just on a on a different gdo.  By chance, I swapped just the power supply (old phone charger) from the working one to a different board wired a different gdo.  Turns out that was the trick, and success followed that one charger on different boards.

Hmm... _okay_ so maybe I have some crappy underpowered power supplies/chargers...but really, how many mA are needed for an ESP32?  Not much.  So tried about 8-10 others with higher mA output, just in case.  No dice.  Tried different cables.  No dice.  WTH?  

After a _lot_ more troubleshooting I happened to unplug my Insteon Powerline modem in the basement (yeah, who still uses that, right?  Well, the leak sensors and hard-wired in-wall switches are actually pretty nice...).  Anyway, magically ALL the boards are now rock solid.  Turns out the X10 signals were interfering with the ESP32 and the gdo comms.  

I assume the charger on the one working board has some noise suppression capability built in allowing the board to avoid the interference.  Other than ditching my Insteon setup, are there any mods that can be done to the rat-ratgdo circuit that would somehow filter out the noise?

[image]

Kaldek, 2024-01-22:
Hi mate, sorry for the lack of replies.

I'm personally nowhere near enough of an electrical engineer to provide any useful guidance on avoiding interference.  About the most I would be able to do if it was me was hook up an oscilloscope and monitore the signal wires with and without the modem enabled.  I mean, the modem could be **faulty** and an oscilloscope would show what is happening at least, so it could be compared to expected behaviour.

powersra, 2024-01-24:
No worries, I'll do some more testing to try and get it sorted.  Very strange problem!

closed 2024-01-24 by powersra