Kaldek/rat-ratgdo #78: One board is controlling both doors

Heinouser, 2024-01-17:
I have two garage doors with Liftmaster Security +2.0, and made two Ratgdos from D1 Minis with the wiring on shields. They are on the Ratgdo 0.9.0 HomeKit firmware. Trying to control one door controls both of them: open together, close together. The two wall remotes also control both doors, but the remote controls are fine.

At one point one Ratgdo was on the HomeKit firmware and the other one was on the MQTT firmware and was going through Homebridge.

I would of course first blame my amateur soldering and layout skills, but I would think that mistakes at that step would result in no operation at all. I got the same behavior when one was on a breadboard and the other was assembled on the shield.

The doors are not mechanically connected or wired together in any way, but the wires run parallel to each other when going to the wall remotes. Could the boards be introducing some sort of crosstalk somewhere?

Kaldek, 2024-01-17:
It's a pulldown to ground of a 12v wire so I would doubt crosstalk would be a factor here.

What I would do for testing is have them both running ESPHome and use the Web GUI of the ESP8266 to drive the door.  I say this mainly because I don't know what weird back-end linking might be going on (like, if the entities have somehow been grouped in HA?).

Heinouser, 2024-01-19:
I got logs for both doors, opening and closing. This is with one door on ESPHome and one on HomeKit. Thanks for looking at this, I think I definitely found an edge case.
[image]
[image]

Heinouser, 2024-01-20:
Got both doors on ESPHome and looked at the logs for both. Inconclusive: the other door is behaving as if its button was pressed, and the logs are exactly the same whether the instruction is sent to one door or the other. The only difference is here:

`[ratgdo:218]
Status: door=OPENING light=ON lock=UNLOCKED learn=INACTIVE`

This is Door 2 when the open instruction was sent to Door 1. On all the other logs, there is no learn=INACTIVE.

Related or not, it is possible to get the doors de-synced: one door will open while the other closes. That's all I can think of for now, hope that helps.

rlowens, 2024-01-21:
Are you absolutely sure the wires aren't connected together somehow (a staple holding them both is shorting them, etc.)?

If you disconnect the wall remotes, does it still happen then?

If so, with the wall remotes disconnected, can you watch the voltage on one pair of wires while shorting the other pair and vice--versa?

Kaldek, 2024-01-22:
> The two wall remotes also control both doors, but the remote controls are fine.

This is your starting point.  Since both your wall remotes are controlling both doors, you've got to resolve that issue.  I'm going to close this as it's *almost* guaranteed to be an issue unrelated to RatGDO.

closed 2024-01-22 by Kaldek

Heinouser, 2024-01-22:
Just to follow up, you were right. Disconnecting the remote wires at the wall button resulted in the same behavior. Disconnecting at the GDO fixed it. Then I find a sloppily repaired wire halfway between: installers got lazy (no surprise). New wires and it's perfect. The doors worked fine for years like this but it seems like the Ratgdo is more sensitive, which makes sense. Thanks for putting me on the right path!