Kaldek/rat-ratgdo #79: Just sharing a "retail" ratgdo product

youaintnobody, 2024-01-17:
Not sure if anyone has come across this yet, but it looks like Nate/Konnected are selling a product based esphome-ratgdo.

[Announcing two new Garage Door Opener products from Konnected!](https://community.konnected.io/t/announcing-two-new-garage-door-opener-products-from-konnected/35300)
[Smart Garage Door Opener White (v2)](https://konnected.io/products/smart-garage-door-opener)

I'm not affiliated, just thought I'd share in case someone wants to buy a similar product but a bit more polished/retail... but it does cost more.

rlowens, 2024-01-17:
Interesting. Expects to ship in March 2024, $89 (on sale at $74 currently with preorder) https://konnected.io/products/smart-garage-door-opener-blaq-myq-alternative

Notable differences I see:
1. "incorporates a pre-closing audible and visual warning with a built-in bright flashing LED and multi-tonal buzzer, making it safe and compliant with U.S. federal safety regulations"
2. "A power adapter is not required – the device draws power directly from the garage door opener" - The announcement page shows a 12V barrel jack for power which isn't shown on the preorder page:
[image]

[image]
I wonder how they are planning to power from GDO. Just using the Red Control wire (like the wall control panel does) hasn't been done with ratgdo yet, and powering from internal GDO 12V would require opening the case (unless you tap the backup battery which only some GDOs have).

They also claim that it is using esphome-ratgdo (GPL v2) and will have firmware free from https://github.com/konnected-io/konnected-esphome/blob/master/garage-door-esp32.yaml but I don't see any ratgdo code there yet (only dry-contact firmwares so far).

Kaldek, 2024-01-17:
Hah and Paul Wieland was worried about **THIS** repo!

youaintnobody, 2024-01-18:
Could they be using this? DC to DC from 30VDC to 5/3.3VDC?
[image]

Does Konnected share their schematic for their security board? I know when I was looking at it in the past you could flash your own nodemcu board and create your own, but I can't seem to find the info.

Kaldek, 2024-01-18:
> Could they be using this? DC to DC from 30VDC to 5/3.3VDC?

They absolutely could be.  It's not hard to to just use a voltage regulator which accepts a wider input range up to or exceeding 30 volts.  The LM2596 for example supports up to 40 volts DC input.

closed 2024-01-22 by Kaldek