Kaldek/rat-ratgdo #80: 2.5i in hand!

mikekay1, 2024-01-17:
Hey Kaldek,

Mike here, as promised came back with 2.5i.

Is there anything I can test for everyone?

rlowens, 2024-01-17:
First, can you verify what it says on each chip? Maybe post some nice close-ups of each?

Closest we have so far is the order page pic:
[image]

judeibe, 2024-01-17:
[image]
[image]
[image]
[image]
[image]

mikekay1, 2024-01-19:
Ah sorry guys, didn't get to it, looks like someone beat me to it! I'll post mine for comparison.
[image]
[image]
[image]
[image]
[image]