Kaldek/rat-ratgdo #81: Another product available: Gelidus Research AC Powered RATGDO Compatible Board

rlowens, 2024-01-21:
So this is interesting: https://www.gelidus.ca/product/gelidus-research-ac-powered-ratgdo-compatible-board/

Integrated 120V outlet passthru. And upcoming "315Mhz RF module headers (New software feature in progress)"

Reasonable price at $21 + Canada Post International Air 5-10 Days $9.00 shipping to US.

But without UL approval, I wouldn't use it. Prefer using an approved external USB power brick and only low-voltage inside my custom devices. Plus those programming headers and button right next to the exposed AC outlet is scary to me! (though for programming you would be powering it via the external FTDI adapter, not via the internal AC power supply).

[image]

[image]

[image]

[image]

[image]

Kaldek, 2024-01-21:
I'd just like to know if the creator based it off the reverse engineering **we** did or if they did it themselves.

Seems like decent enough clearance from the 120v lines to the 3.3v stuff.

closed 2024-01-22 by Kaldek

descipher, 2024-01-25:
> I'd just like to know if the creator based it off the reverse engineering **we** did or if they did it themselves. Only for curiousity! Ever since ratgdo went viral it was only a matter of time before other folks with manufacturing and shipping experience picked it up.
> 
> Seems like decent enough clearance from the 120v lines to the 3.3v stuff.

Not much to reverse engineer, I probed the GDO control signal and it clearly revealed a 0-12v serial signal on a single wire and the logic probe auto decoded it as 9600 baud, 8, 1, n. too easy. Since the remote button is powered from the line we can also deduce its normally high. I could use 8-10 circuit known interface methods to talk to it.  The real genius is Clayton Smith for breaking the encryption and allowing us to read the codes. Paul did a great job writing code for it and it's public domain. My intent is to get the code into ESPHome with some additional features. Problem is the code uses a an interrupt based Arduino Serial repo that's not in ESPHome. Timing is critical for the command control responses. By the way please use a3400's or a3402's not the 2n7000  The gate signal level threshold is better and will have a cleaner signal. 

As far as the AC danger, you are more at risk buying a fake UL listed USB chargers from AliExpress. Just think how often do we handle the module? Also CSA/UL listing costs 8-10K to get and even with that you can do something dumb and still get killed. My warnings on the info page are very clear, don't mess with it when powered by AC. It's safe to run it has CSA/UL design spec but no way am I paying that $ for the label.

Kaldek, 2024-01-25:
> The real genius is Clayton Smith for breaking the encryption and allowing us to read the codes. Paul did a great job writing code for it and it's public domain.

Not sure if you're involved in the ESPHome project.  But I'd like to briefly mention that Paul's actions have been pretty poor in this whole discussion.  Unrelated to me he silently closed any and all requests for open source schematics (a minor issue).  Then he banned me from his repo, and then he has silently deleted any issues I have raised in the ESPHome-ratgdo repo, particularly the one where I mentioned 2MB flash device support.  No discussion, just deleted it.

descipher, 2024-01-25:
Yes I am aware of that and I have contributed to ESPHome many times. complete LD2420 component, ATM90E32 features etc. 
The GRGDO happily runs a more capable ESP32 wink wink ..

descipher, 2024-01-25:
I would be interested in any issues when the code is refactored for the ESPHome community.

rlowens, 2024-01-25:
> By the way please use a3400's or a3402's not the 2n7000 The gate signal level threshold is better and will have a cleaner signal.

Haven't had anyone really knowlgeable chime in on transistor choice here. You suggest a3400/a3402 instead of the 2n7000 on the RX pin, but what about on the TX pin? There we have AO3400A if SMD, or someone suggested BS170 for thru-hole. What do you suggest?

descipher, 2024-01-25:
Both TX and RX can be the a3400 or a3402 it's just the 2n7000 that's not ideal. A BS270 is a better solution with a 1V threshold.

Kaldek, 2024-01-25:
Happy to update our through-hole recommendation to a BS270.  For anyone wanting to do through-hole for a self-made board would you accept the BS270 for both TX and RX?

foreverimagining, 2024-01-25:
If it helps any, I happen to have ordered some AO3402s as well as the components I need for my board. I could try that instead of the 2N7002 in a couple weeks when they all make it here to confirm its utility for SMD purposes---though I don't see why it wouldn't be more than sufficient. And not that I've had any noticeable issues from the 2N7002 on my original v2.5 board, but I also don't check it that often. Still, the problems seem to mostly be through-hole mosfets.

descipher, 2024-01-25:
The 2n7002 fits better than the 2n7000. The gate draw current and voltage thresholds of the 2n7002 are in spec for the application.

foreverimagining, 2024-01-25:
> The 2n7002 fits better than the 2n7000. The gate draw current and voltage thresholds of the 2n7002 are in spec for the application.

That would explain, at least in part, why the SMD boards never seem to have the same issues as the breadboard prototypes. And is pretty much why I decided to have some SMD-to-DIP boards made so that I can breadboard with the actual components I would likely be using.

And since I didn't mean to distract, is that a yes to Kaldek's question?

> For anyone wanting to do through-hole for a self-made board would you accept the BS270 for both TX and RX?

rct, 2024-01-25:
@rlowens - Good find, thanks for posting this!   Maybe I'm misguided, but I've been preferring an ESP32 solution.  This seems like a good choice. 

This issue is closed, makes sense-not an issue, but maybe discussions and/or the wiki should be enabled on this repo to capture some of the info and options that have been posted here. 

@descipher - First time I've seen your stuff. Where do you announce things?  I didn't see a mailing list or social info on your website. Can't wait to see what you are going to do with the 315mhz option. 

Also:

> My intent is to get the code into ESPHome with some additional features. Problem is the code uses a an interrupt based Arduino Serial repo that's not in ESPHome. Timing is critical for the command control responses.

I've just been very casually following things and waiting for things to somewhat gel. It's the first time I've seen a mention of the signal timing being critical. This of course makes sense to me as someone who is much more software oriented.

There's been a lot of discussion about the electrical properties, but I could see easily see latency as also a contributor to less than ideal repeatability.

descipher, 2024-01-26:
@rct I hang out on discord to discuss code or hardware design. I'm working the wireless feature in a private repo until its stable and usable. Its simply a Manchester encoded stream, the current RATGDO IRQ based code is a little temperamental. Paul must of had fun getting the timing just right. I will light up the code on my github page when its good. Really not big on general Social Media its very time consuming and tends to distract from this work. https://discord.gg/ZyRmjfka2f