Kaldek/rat-ratgdo #82: Overhead door compatability

mikedhanson, 2024-01-26:
I have a overhead door legacy 850 model 2029 and wondering if this would be compatible with that unit. I am not finding much online about it. 

Thanks

rlowens, 2024-01-26:
Skimming the manual at https://www.overheaddoor.com/garage-door-opener-legacy-850 doesn't mention "Security+" at all.

Does the wall panel work as a simple dry-contact? What happens if you disconnect the wall panel and just short the wires - does that operate the garage door?

What voltage is there between the wires to the wall panel? What voltage on the wires to the obstruction sensors (when clean and when obstructed)?

mikedhanson, 2024-01-26:
It is not dry contact. I am not sure what the voltages are but I sure check.

Kaldek, 2024-01-30:
I've done a fair bit of work recently on coding Arduinos for detecting pulse based serial data.  If yours has some kind of digital signal, especially if not encrypted, I think we can work on adding to the code base for ESPHome-ratgdo to support it.

mikedhanson, 2024-01-31:
This might be a long shot but is there something I can test here? I have no idea really what im doing when it comes to garage doors but im willing to give something a try and assist in anyway i can to get something working.

Kaldek, 2024-03-18:
Closing due to lack of updates.

closed 2024-03-18 by Kaldek