Kaldek/rat-ratgdo #83: Intermittent comms?

youaintnobody, 2024-01-30:
As per [this](https://github.com/Kaldek/rat-ratgdo/issues/74), I have built myself a 2.5 board currently on a breadboard and have been testing for 2 weeks now.

I've been noticing the following. Sometimes comms is intermittent.
- Pressing query status sometimes does not return a result.
- After a reboot (via ratgdo or physical power) the device reports the door as open even though the door is closed.
- The GDO light has switched off but not reflected in esphome/home assistant.

Has anyone else had these issues? I have added a DHT11 sensor but it makes no difference having this on or off. Open/Close/Stop/Toggle works most of the time. Toggling light on/off sometimes doesn't work or out of sync.

Parts used: 2 x 2n7000, 1 x 1K, 5 x 10K, NodeMCU ESP8266.

Kaldek, 2024-01-30:
Hi mate, there's been a bit of movement and for through-hole designs the recommendation is a BS270 MOSFET now.   I would recommend that as the first place to start, due to the issue with the 2n7000 being potentially marginal for reliable switching, depending on batch and manufacturer.

youaintnobody, 2024-01-30:
Thanks, I've ordered some BS270 to test out. I'll report back on my testing.

sanjay900, 2024-01-31:
I've just ordered some BS270's too, i was having problems with the 2n7000 as well. For whatever reason it seems my TX mosfet is working but none of the 8 mosfets i used for RX worked at all

Kaldek, 2024-02-03:
> I swapped out the 2n7000 with the BS270, I still seem to be having similar issues. It's still on a breadboard, wondering if soldering it onto a pcb would eliminate any connection issues.

It can definitely be an issue.  Some of my breadboards are utter trash and have terrible continuity.

sanjay900, 2024-02-05:
BS270 worked for me, though i think soldering them was killing them - worked fine on a breadboard but died when i soldered it together - i gave up and used some connector blocks and it worked first try

sanjay900, 2024-02-05:
> I swapped out the 2n7000 with the BS270, I still seem to be having similar issues. It's still on a breadboard, wondering if soldering it onto a pcb would eliminate any connection issues.

Had this same issue with a breadboard earlier but then i noticed the mosfets were in pretty loosely and moved it to another spot on the breadboard where it wasn't so loose and it worked

youaintnobody, 2024-02-07:
We’ve had a heatwave in the last week, my garage was reaching over 38°C and my device kept playing up. Device hasn’t been too bad this week.

Sometimes when the device reboots via software or from power off then on the status doesn’t update quick enough or it resumes at an open state.

lewisdiamond, 2024-02-10:
I'm using a esp32 dev board and used this schematic: https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png

Tried using 2n7000, bs270 and bs170. Nothing worked.

Kaldek, 2024-02-10:
> I'm using a esp32 dev board and used this schematic: https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png
> 
> Tried using 2n7000, bs270 and bs170. Nothing worked.

Sorry to say it but it's time to break out the oscilloscope.

sanjay900, 2024-02-10:
> I'm using a esp32 dev board and used this schematic: https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png
> 
> Tried using 2n7000, bs270 and bs170. Nothing worked.

Just to check, did you program it using the v2 or the v2.5 firmware? that schematic is assuming the v2 firmware, as the pins changed between the v2.5 and v2 hardware.

lewisdiamond, 2024-02-10:
> Sorry to say it but it's time to break out the oscilloscope.

I wish I had one.

> Just to check, did you program it using the v2 or the v2.5 firmware? that schematic is assuming the v2 firmware, as the TX pin changed between the v2.5 and v2 hardware, so if you used the 2.5 firmware, it would not let you control the GDO

I used the ESP Home version with the 2.0 and matched the pins with the schematics.

lewisdiamond, 2024-02-10:
This schematic: https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf with BS270s work fine. I can test with 2n7000s and BS170s so we can update the readme.

Kaldek, 2024-02-11:
More evidence that depending where your MOSFETs come from, the older recomendations are no bueno.

lewisdiamond, 2024-02-12:
> More evidence that depending where your MOSFETs come from, the older recomendations are no bueno.

The older schematic is very much subpar. I think it's worth removing it and keeping only the newer one. The device it's used on doesn't really matter as they're all 3.3v ESP. I'll provide some data on BS170 and 2n7000. A formal BOM would be useful.

closed 2024-02-13 by youaintnobody