Kaldek/rat-ratgdo #84: Update/Advice - Power for ESP board

jjarboe01, 2024-01-30:
So, I don't like having extra wires to outlets for USB power.  I took my Chamberlain opener (purple pairing button if it matters) apart and found this.  Unplug the unit from power.  The back pulls off with 3 screws, remove the wires for the button/sensors.  On the back panel at the top, you can disconnect the 2 plugs (they are pinned directionally and only fit their specific port).  Once you have the back cover down, you will see 2 boards.  One has relays, the other mounts to the top of that and is the actual logic board.  That board needs 24 volt DC to operate, and is fed by the transformer on the relay board.  The port pairing the two has 4 pins together, a space and another pin.  The lone pin is ground (you can follow the trace to the white connector for the sensors).  The total opposite pin is +24v.  I soldered wires to the relay board side to run outside the unit and connect to my ESP board with a buck converter that will handle up to 60vDC.  This way, the ESP is powered by the garage door opener on its own with no additional plugs required.  Just thought I'd throw this out here in case anyone is interested in a "cleaner" install.

ashishgururani, 2024-02-27:
The instructions make it sound like you can use the factory connector black/white for 12v power to a buck converter to get 5v or 3v. Is that not the case?

Kaldek, 2024-03-18:
Closing due to lack of updates.

closed 2024-03-18 by Kaldek