Kaldek/rat-ratgdo #85: Security+ 1.0 and 2.0 installer

jrsphoto, 2024-02-04:
I've got an older Chamberlin opener (purple button Security+ 1, and I've had some D1Mini ESP32 boards made.  In the docs it mentions:

"CRITICAL: You must install using the v2.0 install for ESP-32 boards as v2.5 is additional and unnecessary work for these boards at the current time."

However, when I go to [esphome-ratgdo](https://ratgdo.github.io/esphome-ratgdo/) to install, I select Security + 1.0 protocol, but when I try select the ratgdo v2.0 board, It says "ratgdo version 2.0 only works with security + 2.0". 

what am I missing? or did I just get the wrong board made?

jrsphoto, 2024-02-04:
figured it out

closed 2024-02-04 by jrsphoto

Darkblade48, 2026-04-05:
Long shot here (as I know this has been marked resolved from over 2 years ago), but what was the solution to this?