Kaldek/rat-ratgdo #88: Issues with a Chamberlain ML700EV GDO and breadboard draft

CodeFinder2, 2024-02-10:
Hi,

I tried to build a breadboard circuit based on a Wemos D1 mini pro which looks like this):
[image]
[image]
[image]

I can also try to draw this if needed. I (and also a friend) double-checked the wiring and it should be correct (?). Resistors also measured to be 9,99kOhm. I used 2n7000 MOSFETs (and also tried BS170, same problem).

This is how I wired it to the GDO:
[image]

I flashed it via the ESPHome website installer [here](https://ratgdo.github.io/esphome-ratgdo/), using "Security + 2.0; round yellow learn button" (first row, first pic) and "ratgdo v2.5x; v2.5, v2.5i, v2.52i, v2.53i (black PCB)" (second row, first pic) which seemed to work fine. (It is the correct software?)

My GDO should be Security 2+.

TL;DR: once my ESP is connected, my GDO does strange things (open/closing) and I don't see any status in ESPhome.

**The Problem**
When I connect it to my GDO and power it up, it immediately starts moving the garage door (GD). However, this just happens for about a second and then the motion is interrupted again. After about another second, it starts moving again. This repeats a few time until the GD finally closes completely and nothing more happens. In ESPhome's web UI, there seem to be no (valid) status reporting for the GD(O) and I am also unable to control the GD, light, etc.  After some "back and forth" of the GD, these spurious commands seem to stop. Note that I can control the GD using my normal (original) remote control (before, meanwhile and after these strange open/close commands). To my understanding, this seems to be an issue with the TX/RX circuit part but I do not see any issue.

Not sure if this is a helpful log but anyway:
```
[C][ratgdo.number:024]RATGDO Number 'Client ID'16:57:15
[C][ratgdo.number:026] Type: Client ID16:57:15
[C][switch.gpio:068]GPIO Switch 'Status door'16:57:15
[C][switch.gpio:090] Restore Mode: always OFF16:57:15
[C][switch.gpio:031] Pin: GPIO1616:57:15
[C][switch.gpio:068]GPIO Switch 'Status obstruction'16:57:15
[C][ratgdo:082]Setting up RATGDO...16:57:15
[C][ratgdo:083] Output GDO Pin: GPIO516:57:15
[C][ratgdo:084] Input GDO Pin: GPIO416:57:15
[C][ratgdo_secplus2:059] Rolling Code Counter: 185816:57:15
[C][ratgdo_secplus2:061] Protocol: SEC+ v216:57:15
[C][ratgdo.sensor:041]RATGDO Sensor 'Openings'16:57:15
[C][ratgdo.sensor:041] State Class: ''16:57:15
[C][ratgdo.sensor:041]RATGDO Sensor 'Paired Devices'16:57:15
[C][ratgdo.sensor:041] State Class: ''16:57:15
[C][ratgdo.sensor:045] Type: Paired Devices16:57:15
[C][ratgdo.lock:012]RATGDO Lock 'Lock remotes'16:57:15
[C][ratgdo.lock:013] Type: Lock16:57:15
[C][ratgdo.switch:068]RATGDO Switch 'Learn'16:57:15
[C][ratgdo.switch:070] Icon: 'mdi:plus-box'16:57:15
[C][ratgdo.switch:090] Restore Mode: always OFF16:57:15
[C][ratgdo.binary_sensor:036]RATGDO BinarySensor 'Motion'16:57:15
[C][ratgdo.binary_sensor:036] Device Class: 'motion'16:57:15
[C][ratgdo.binary_sensor:038] Type: Motion16:57:15
[C][ratgdo.binary_sensor:036]RATGDO BinarySensor 'Obstruction'16:57:15
[C][ratgdo.binary_sensor:036] Device Class: 'problem'16:57:15
[C][ratgdo.binary_sensor:040] Type: Obstruction16:57:15
[C][ratgdo.binary_sensor:036]RATGDO BinarySensor 'Button'16:57:15
[C][ratgdo.binary_sensor:044] Type: Button16:57:15
[C][ratgdo.binary_sensor:036]RATGDO BinarySensor 'Motor'16:57:15
[C][ratgdo.binary_sensor:036] Device Class: 'running'16:57:15
[C][ratgdo.binary_sensor:042] Type: Motor16:57:15
[C][ratgdo.cover:014]RATGDO Cover 'Door'16:57:15
[C][ratgdo.cover:014] Device Class: 'garage'16:57:15
[C][ratgdo.light:014]RATGDO Light16:57:15
[C][restart.button:017]Restart Button 'Restart'16:57:15
[C][safe_mode.button:022]Safe Mode Button 'Safe mode boot'16:57:15
[C][safe_mode.button:022] Icon: 'mdi:restart-alert'16:57:15
[C][homeassistant.time:010]Home Assistant Time:16:57:15
[C][homeassistant.time:011] Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'16:57:15
[C][web_server:168]Web Server:16:57:15
[C][web_server:169] Address: ratgdov25i-b33329.local:8016:57:15
[C][mdns:115]mDNS:16:57:15
[C][mdns:116] Hostname: ratgdov25i-b3332916:57:15
[C][ota:097]Over-The-Air Updates:16:57:15
[C][ota:098] Address: ratgdov25i-b33329.local:826616:57:15
[C][api:139]API Server:16:57:15
[C][api:140] Address: ratgdov25i-b33329.local:605316:57:15
[C][api:142] Using noise encryption: YES16:57:15
[C][improv_serial:032]Improv Serial:
```

I can also try to collect a log of the strange open/close attempts after powering it up (please let me know how I should do that ideally).

(Off-topic: as far as I understood, the green +/- battery terminals should have 30V but I measured 0V. It that correct? I have no backup battery installed but also don't know where to do that. Once this all works, I thought about powering the ESP via these terminals, if possible.)

**What I already tried**
I also tried with and without connecting the black OBSTACLE wire but noting changed (but AFAIK, this should't have an effect anyway). For try-and-error, I also switched D1 and D4 pins at my ESP8266 (because I read something about possible different PIN mappings) but that didn't help. In fact, after that, no spurious open/close commands where triggered anymore but also still no status avail (and still no control using ESPhome).

**Remarks**
Generally, my GD is working just fine. I also connected a wired wall switch (as part of my GDO system, using the same red and white terminals) and that worked fine as well.

Any help/hints/advice is highly appreciated. Also, please let me know if you need more logs (and how I should collect them).

Thank you very much! :-)

foreverimagining, 2024-02-10:
Yours sounds like it might be doing some extra funny things, so I'm not completely sure I'll be able to help, but I will proffer what knowledge I have.

Firstly, I'm assuming that you are using [this](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v6.png) schematic since I only see 3 resistors. This one has worked for a number of people, but others have only had luck with [this other](https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf) schematic, which is a little more up to date and based on my original v2.5 RATGDO. That's a good place to start troubleshooting.

Also:

> I used 2n7000 MOSFETs (and also tried BS170, same problem)

The 2n7000s have been notoriously finicky, so it's probably better to just stick with BS270s, which have recently been recommended as much more suitable and, therefore, reliable. I pulled up datasheets for the BS170 to compare to the BS270, and the BS170 looks like it has a wider range for the Gate Threshold Voltage. Actually, it appears to have the same range as the 2n7000, so it's probably also not as good of an option as the BS270, though, I am by no means well-educated on the matter.

A few other things I can remark on:

> I flashed it via the ESPHome website installer [here](https://ratgdo.github.io/esphome-ratgdo/), using "Security + 2.0; round yellow learn button" (first row, first pic) and "ratgdo v2.5x; v2.5, [...]

This looks like the correct firmware to me since you are using an ESP8266 not an ESP32, and yes, you appear to have a Security+2.0 GDO with a yellow learn button that is not a jackshaft opener. That being said, I don't know if there is a specific difference with the D1 mini _PRO_ that might affect the compatibility somehow. If all else fails, this could be the problem.

> I also switched D1 and D4 pins at my ESP8266 (because I read something about possible different PIN mappings)

Previous versions of the RATGDO used D4, which is pulled HIGH at boot, and it did cause some GDOs to open on their own occasionally when the ESP restarted. I have been under the impression that they fully opened, and then, didn't close on their own, so yours seems like it might be doing something different, unless my understanding is incorrect. The firmware for v2.0 should be using D4, but the firmware for v2.5+ should be using D1. That is, unless you change the yaml file in ESPHome to assign different pins, but it's easier to just use the original pins and move the wire on the breadboard if necessary.

> I also tried with and without connecting the black OBSTACLE wire but noting changed

Yeah, this probably won't benefit your issues because the obstruction sensors can totally cause their own issue, and you don't seem to be having it. Basically, the close command doesn't work if you don't have the obstruction sensors connected, though toggle still does. It's a weird thing.

> (Off-topic: as far as I understood, the green +/- battery terminals should have 30V but I measured 0V. It that correct? I have no backup battery installed but also don't know where to do that. Once this all works, I thought about powering the ESP via these terminals, if possible.)

I don't have a battery backup, so I won't be helpful here, but other people have made RAT-RATGDOs that run off their battery backups, so it should be possible. You can take a look at what other people have done, unless someone who knows a bit more about it wants to chime in.

And one last word of advice: loose connections are the absolute devil. We frequently have people having trouble with breadboards that seem to be unable to work out the problems no matter what they try, and it may ultimately come down to loose connections. There never seem to be problems with the PCBs that use SMD parts, and there don't seem to be many problems involving soldered perfboards, either (besides 2n7000s). If you try everything else and nothing works, you may just have to go all-in and solder something a bit more permanent and give that a go. There can also be loose connections at the GDO terminals, though, if they don't bite down well enough on the wires, so really, there are no guarantees.

youaintnobody, 2024-02-11:
Connect your red/ctrl wire to terminal 0 (E-serial).

On the photo you have it connected to terminal 1, 1 and 2 normally has a dry contact/button that toggles open/stop/close.

CodeFinder2, 2024-02-13:
Thank you very much for your answers / input @foreverimagining  and @youaintnobody, really appreciated! :-)

Took me some time and some more testing to try a few things but I seem to be getting closer.  According to your photo, I really seem to have wired it incorrectly to my GDO. The [official ratgdo docs](https://paulwieland.github.io/ratgdo/03_wiring.html) also state that "If your Security + 2.0 door opener has an eserial terminal (common outside of North America) connect the red/ctrl wire to eserial and not the red terminal." So, thank you for that pointer!

I already tried that and the spurious opening/closing issue disappeared (now, nothing happens when connected as it should). Unfortunately, I am still unable to control it and/or do not see any status. However, my up/down arrows at the back of my GDO now indicate an error code by blinking (1 Up 3 Down). According to [this](https://support.chamberlaingroup.com/s/article/Garage-Door-Opener-Diagnostic-Codes#ErrorCode1UP3DOWN), it means that the "Door control panel [aka my rat-ratgdo circuit] wires are shorted for more than 4 seconds."

I already tried disabling the RX part of my circuit (to just receive the status) but it still shows the blink code after powering up the GDO. I will now try different (non-flexible) wires. After that, I will probably re-do my entire circuit (probably on a larger breadboard or so) because the blink codes seem to be indicating some shorted wires.

@foreverimagining: The used schematic you have linked is correct (with 3 resistors). Also tested your other schematic but no success. By the way, not connecting the obstruction sensor terminal anymore during all of these tests (as I don't have one anyway). After that, I will try to get my hands on BS270 MOSFETs and test with them (but I feel it should work at least "somehow" with the ones I already have...).

Can anyone share a ESPhome screenshot of a working installation and log output? Or give me some hints on what I should pay attention? I am always wondering what should be shown in the logs or web UI once its (at least partially) working...

CodeFinder2, 2024-02-13:
Unbelievable...it seems to work (based on [this](https://raw.githubusercontent.com/Kaldek/rat-ratgdo/main/images/Breadboard_working.png) but it should really be equal to what I build before).
TL;DR: *it seems like a combination of 2 issues: wrong terminal at GDO (green "0" is correct, not red "1") and bad MOSFETs (working with BS170).*

Details:
I switched the 2N7000 and realized that the blinking codes where generated when I used 2 specific (broken?) 2n7000 MOSFETs. I picked 2 others of them and blinking was gone although still no control/status.

I then used 2 BS170 and got the same behavior (no blinking but also no status/control). I then tried my two remaining BS170 and voila...I now ended up with this log (might be helpful for others):
```
22:57:03	[D]	[ratgdo:210]	Lock state=UNLOCKED
22:57:03	[D]	[ratgdo:189]	Learn state=ACTIVE
22:57:07	[D]	[ratgdo_secplus2:213]	Query paired devices of type: ALL
22:57:07	[D]	[ratgdo_secplus2:213]	Query paired devices of type: REMOTE
22:57:07	[D]	[ratgdo_secplus2:213]	Query paired devices of type: KEYPAD
22:57:07	[D]	[ratgdo_secplus2:463]	Collision detected, waiting to send packet
22:57:07	[D]	[ratgdo_secplus2:213]	Query paired devices of type: WALL_CONTROL
22:57:07	[W]	[ratgdo_secplus2:430]	Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
22:57:07	[D]	[ratgdo_secplus2:466]	Collision detected, waiting to send packet
22:57:07	[D]	[ratgdo_secplus2:213]	Query paired devices of type: ACCESSORY
22:57:07	[W]	[ratgdo_secplus2:430]	Have untransmitted packet, ignoring command: GET_PAIRED_DEVICES
22:57:07	[W]	[ratgdo_secplus2:107]	Triggering sync failed actions.
22:57:07	[W]	[component:214]	Component ratgdo took a long time for an operation (0.15 s).
22:57:07	[W]	[component:215]	Components should block for at most 20-30ms.
22:57:07	[D]	[ratgdo:270]	Openings: 252
22:57:07	[D]	[number:012]	'Rolling code counter': Sending state 8370.000000
22:57:07	[D]	[sensor:093]	'Openings': Sending state 252.00000 openings with 0 decimals of accuracy
22:57:07	[D]	[ratgdo:278]	Paired device count, kind=KEYPAD count=0
22:57:12	[D]	[switch:016]	'Learn' Turning OFF.
22:57:12	[D]	[number:012]	'Rolling code counter': Sending state 8371.000000
22:57:13	[D]	[number:012]	'Rolling code counter': Sending state 8372.000000
22:57:13	[D]	[number:012]	'Rolling code counter': Sending state 8373.000000
22:57:13	[D]	[ratgdo:095]	Door state=OPEN
22:57:13	[D]	[ratgdo:204]	Light state=ON
22:57:13	[D]	[ratgdo:210]	Lock state=UNLOCKED
22:57:13	[D]	[ratgdo:189]	Learn state=INACTIVE
22:57:13	[D]	[switch:055]	'Learn': Sending state OFF
22:57:13	[D]	[ratgdo_secplus2:213]	Query paired devices of type: ALL
22:57:13	[D]	[number:012]	'Rolling code counter': Sending state 8374.000000
22:57:13	[D]	[ratgdo:278]	Paired device count, kind=ALL count=0
22:57:13	[D]	[sensor:093]	'Paired Devices': Sending state 0.00000  with 0 decimals of accuracy
22:57:13	[D]	[ratgdo_secplus2:213]	Query paired devices of type: REMOTE
22:57:14	[D]	[number:012]	'Rolling code counter': Sending state 8375.000000
22:57:14	[D]	[ratgdo:278]	Paired device count, kind=REMOTE count=0
22:57:14	[D]	[ratgdo_secplus2:213]	Query paired devices of type: KEYPAD
22:57:14	[D]	[number:012]	'Rolling code counter': Sending state 8376.000000
22:57:14	[D]	[ratgdo:278]	Paired device count, kind=KEYPAD count=0
22:57:14	[D]	[ratgdo_secplus2:213]	Query paired devices of type: WALL_CONTROL
22:57:14	[D]	[number:012]	'Rolling code counter': Sending state 8377.000000
22:57:14	[D]	[ratgdo_secplus2:213]	Query paired devices of type: ACCESSORY
22:57:14	[D]	[number:012]	'Rolling code counter': Sending state 8378.000000
22:57:14	[D]	[ratgdo:278]	Paired device count, kind=ACCESSORY count=0
22:57:14	[D]	[esp8266.preferences:238]	Saving preferences to flash...
22:57:16	[D]	[light:036]	'Light' Setting:
22:57:16	[D]	[light:047]	  State: OFF
22:57:16	[D]	[number:012]	'Rolling code counter': Sending state 8379.000000
22:57:26	[D]	[light:036]	'Light' Setting:
22:57:26	[D]	[light:047]	  State: ON
22:57:26	[D]	[number:012]	'Rolling code counter': Sending state 8380.000000
22:58:14	[D]	[esp8266.preferences:238]	Saving preferences to flash...
```

Really a mess that the whole thing is so picky in terms of MOSFETs.

closed 2024-02-13 by CodeFinder2

CodeFinder2, 2024-02-14:
One more thing: up until now it seems really unstable when the thing is (re-)started. Not sure if this is a MOSFET thing or if its related to my breadboard design (loose connections, as @foreverimagining suggested). Really annoying and definitely not close to something a can/will use for the final circuit (non-prototype).