Kaldek/rat-ratgdo #89: Sync Status

ur7x, 2024-02-13:
Not sure how/where to report this..  The Ratgdo page does not seem to have an issues tab. 

I'm running a "3 resistor" version on older Security 1+ door openers with a Red Learn buttons
My D1 Minis ares flashed with the MQTT Security 1+ version from the Ratgdo site.

By and large very happy with this system and this "hack"

**My Issue:**

From time to time the doors will report their status as "Syncing"   Which is **not** a normal state for a Cover in Home Assistant.

Even more interesting, Once the "Syncing" State is over.  The door will report as newly "Closed"  and MyQ will send me an alert that the door just closed... Which is interesting since it never opened. (I haven't unplugged my MyQ system until I was confident that this local solution was 100%)  

I have a MQTT sniffer running and it reports that the little gizmo got a close command right after the sync event...Not sure if this came from HA or from the RATGDO D1mini itself... HA history/logs report the door never opened and no close command was issued.  But HA logs also fail to report that the door was in a Syncing state since HA doesn't know what that is.  

It is also my experience that the light on/off seems to be a toggle that unaware of the state of the light.  Commanding a light on event will turn my light off if they were already on!  Not sure if that is what is going on or if it is a timing issue with MQTT. 

Anyone else experiencing this?

Has anyone taken a look at the RATGDO "firmware"?  Maybe take out the "syncing" status..

foreverimagining, 2024-02-13:
This may not be the best place for this issue if it's just a software problem, but you may also not be able to get much help from the main repositories using a RAT-RATGDO. That being said, here's some background that might get you going.

> The Ratgdo page does not seem to have an issues tab

Paul Weiland ported the code for the various versions of the RATGDO firmware into separate repositories and subsequently closed the Issues section in his original repository. You can look through the MQTT Issues [here](https://github.com/ratgdo/mqtt-ratgdo/issues) to see if there's anything like what you've been experiencing.

> From time to time the doors will report their status as "Syncing"

It's hard to say what this syncing is without having something a bit more verbose. It could be the device trying to sync the door status, as that is a function that can be commanded, but it looks like the code is saying that is only required for Security+2.0. Your device could also have gotten out of sync with the wall panel (some of which are not compatible), and the RAT-RATGDO and wall panel will have to do a "x31 31" dance trying to get back together.

> It is also my experience that the light on/off seems to be a toggle that unaware of the state of the light. Commanding a light on event will turn my light off if they were already on! Not sure if that is what is going on or if it is a timing issue with MQTT.

Looks like you're right, so if it doesn't have the correct status, it will just toggle regardless.
```
// Light functions
void lightOn(){
	if(lightStates[lightState] == "on"){
		Serial.println("already on");
	}else{
		toggleLight();
	}
}

void lightOff(){
	if(lightStates[lightState] == "off"){
		Serial.println("already off");
	}else{
		toggleLight();
	}
}

void toggleLight(){
	if(controlProtocol == "drycontact"){
		Serial.println("Light control not supported with dry contact control.");
	}else if(controlProtocol == "secplus1"){
		getStaticCode("light");
		transmit(txSP1StaticCode,4);
	}else{
		getRollingCode("light");
		transmit(txSP2RollingCode,SECPLUS2_CODE_LEN);
		writeCounterToFlash("rolling",rollingCodeCounter);
	}
}
```

Alternatively to the above, I'm pretty sure the [ESPHome version](https://github.com/ratgdo/esphome-ratgdo) of Security+1.0 is up and running now, not just in testing, so you could try that firmware to see if it has the same problems, unless you have some reason to prefer MQTT.

Alternatively to the alternative, there is this ["man-in-the-middle"](https://github.com/Kaldek/rat-ratgdo/issues/76) variation upcoming that so far doesn't seem to have the same problems for Security+1.0 as have been seen in the original RATGDO (random unprompted openings, strange status updates that are contrary to reality, wall panels not working, etc.). You might have to wait a bit, but I'm suspecting this will be the way to go for most people with Security+1.0.

ur7x, 2024-02-13:
> Paul Weiland ported the code for the various versions of the RATGDO firmware into separate repositories and subsequently closed the Issues section in his original repository. You can look through [...]

Thanks for that... After  a quick read.  Ya, lots of people with Security 1+ doors are experiencing exactly what I am experiencing.  Both ESPHome and MQTT have the same issue.  Seems that the 889LM wall panels have an internal battery that charges off of the opener once a day or so and during that event the system goes off line reports a "syncing" event 

Thanks for also figuring out the toggle issue too...  If I turn off the occupancy sensor on the wall panel the system is considerably more reliable/accurate.

foreverimagining, 2024-02-13:
Happy to help.

Security+1.0 GDOs are "supported", but they've never worked quite right. There's been lots of people with problems (I hear there's a lot of complaints on Reddit), and I've come to the conclusion that it's probably just not all fixable through software. The really funny thing about your situation is that 889LMs are _the_ recommended wall panel, but even they have been turning up issues. I'm excited for the "mad-in-the-middle" option because it seems like a fantastic solution, and it's just further proof that Paul should have just made the hardware open-source by now so other people can help innovate.

closed 2024-03-18 by Kaldek