Kaldek/rat-ratgdo #90: not working with the code

denveronly, 2024-02-15:
Hey there,
Thanks for great work, so i followed and made a pcb, soldered it and got flashed.

I have Security 2.0 GDO
[image]
The board i used is D1 8266. With the pins like in the project soldered.
I flashed this code.  And have it working partialy...
1) I can open and close
2) i can't stop or open\close partially (tried full close and full open several times)
3) i see obstruction 
4) i really can only open and close and see obstruction, no other functions are visible

[image]

`substitutions:
  name: ratgdov2-68ff2d
  friendly_name: ratgdov2 68ff2d
packages:
#  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_lite.yaml@main
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25board_esp8266_d1_mini_secplusv1.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: pqxWqX5iuWO0RWORzt0YbebgYkdYa4q9MnOYrU3RMsA=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password`

[image]
[image]

[image]
[image]
[image]
[image]

denveronly, 2024-02-15:
Maybe it is the way it should be?
I added manually seconds of Closing and Open Time (16sec)
And i have closed and open States, i can close and open partially

Or it should be automatic?

rlowens, 2024-02-15:
Several problems here.
First off, you board is wired wrong:
[image]

[image]

Obst pin (D7 by default but you used D0, which should work I think but you need to specify it in your device .yaml which I don't think you did) is supposed to be connected as a voltage divider between GDO Obst wire and GND, but you thought a cross-over in the schematic was a junction and connected it to part of the D1 tx line instead of GND.

Second, in the firmware: you have a security+ 2.0 GDO but are using security+ 1.0 firmware? I'm surprised this is working at all.
[image]

You want "v25board_esp8266_d1_mini.yaml@main"

I think if you cut that bad trace and connect as I indicated it should work. Also need to specify in your yaml 

```
substitutions:
  input_obst_pin: D0
  status_door_pin: none
```

ashishgururani, 2024-03-03:
@denveronly did you figure this out? I am having the same issue here https://github.com/Kaldek/rat-ratgdo/issues/91

Kaldek, 2024-03-18:
Closing due to lack of updates.

closed 2024-03-18 by Kaldek