Kaldek/rat-ratgdo #91: Made two, one works 100%, other only basic functions works

ashishgururani, 2024-03-01:
So I made two identical ones for my two garage doors by soldering myself (had the W1 mini's laying around from before). The first one works 100%. The second one, only door open close works so it is obviously connected to some level. but none of the other functions works to report status - like door status shows OPEN while it is closed, I lot of other stuff is in UNKNOWN or NA state. What could be possibly wrong with the second one? 

[image]

Sudo-Rob, 2024-03-02:
It would be useful to know your system configuration (e.g. V1.0 or V2.0), the type of mosfet you're using, and a link to the schematic you've used.

My guess is you're in emulation mode because you're not passing data.  I would try changing the mosfet associated with the received data path (Q1).  The gate threshold voltage could be too low.

ashishgururani, 2024-03-03:
This is the schematic I am using using 2n7000
[image]

so, I made a fresh third one since I had more spare parts thinking one of the mofset or resistor may be bad. Interestingly, it has the same exact issue. The first one works great but the second and third one has the same issue - can open and close but not able to read any data. I connected the second and third to the first garage door opener thinking may be my other garage door opener is not sending serial data or something but they don't work connected to the "good" garage door opener either. Hmm, I may have to live with just one for now so I am unsure how to troubleshoot any further.

Sudo-Rob, 2024-03-03:
Does the good unit work with the second GDO?  Does the obstruction circuit work?  Can I assume you're using through-hole parts?  If so, you should double check the orientation of the mosfets as there can be a different lead orientation if BS170 variants are somehow mixed in with the 2N7000s.  

https://www.vishay.com/docs/70226/70226.pdf

Unfortunately, 2N7000 mosfets can have a fairly broad variation of Vgs.  If you have  a 100k resistor, you might try wiring it between the gate of Q1 and ground.  That would indicate the data bus is not getting close enough to ground, or the mosfet isn't totally turned off (or not discharging fast enough) even if it is getting close enough to ground.  If it still doesn't work, try two 100k resistors in parallel.

What kind of test equipment do you have?  A simple logic probe and a multi-meter would be useful.  At least you might be able to test the parts to see if they're usable in the circuit.

https://www.amazon.com/Elenco-Electronics-LP-560-Logic-Probe/dp/B000Z9HAP4

ashishgururani, 2024-03-03:
Yeah, the good unit works with the second GDO. I didn’t connect the
obstruction wire with any of the units. Yes, thru hole components and
orientation of transistors is exactly the same for all the units.

I will try by using a 100k resistor instead of 10k.

I have a multimeter but not sure what to test with it. The ESP is obviously
working. The resistors read 10k on multimeter. ESP reads the GDO to some
extent to not say that no GDO is connected. It can send open close commands
but not read any of the other sensor data. Well I don’t need much of the
other sensor data but it keeps saying that the door is open when it’s
closed which is not good.

foreverimagining, 2024-03-03:
The mosfets are probably the problem since the 2N7000s are so finicky. The current recommendation is to use **BS270**s, but you may be able to try all of the 2N7000s you have around and get lucky finding another one that works. Since commands work, the problem should be on the RX line and if nothing else, [this](https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP8266/RatGDO-OpenSource-D1Mini-ESP8266.pdf) more up-to-date version of the schematic uses additional resistors, which is the only version that works for some people. I'll be curious if the 100k resistor works since that's a new suggestion.

> I didn’t connect the obstruction wire with any of the units.

This isn't necessarily a problem, especially if you don't have obstruction sensors attached to your GDO, but it should be noted that there is a bug where this sometimes causes the "close" command to be nonfunctional. If you find that to be the case for you, "toggle" can still be used to close the door but would obviously be more efficacious if you have open/close status.

Sudo-Rob, 2024-03-03:
@foreverimagining beet me to this, but here's my reply anyhoo.

> Yes, thru hole components and orientation of transistors is exactly the same for all the units.

Anything in a TO-92 package will look similar.  Check the markings on the transistor.  

>  I will try by using a 100k resistor instead of 10k. 

That's not the mosfet of concern.  I'm referring to Q1, which looks at the data on the red wire.   When the voltage on the gate crosses its threshold when going from high to low, it should change state.  That is, the voltage at D2 should "open" and momentarily go to 3.3 volts.  I'm suggesting you connect a pulldown resistor (or two) between the gate (a.k.a. the data bus) and ground.  This will help the transistor by marginally changing the gate voltage when the data line is low.   

By the way, this is what others have been using.  Look at R6 and R7.  Also note that Q1 is a 2N7002, which has a slightly higher minimum Vgs.  (FWIW, my opinion is the threshold should ideally be between 2 and 3 volts).  

https://github.com/Kaldek/rat-ratgdo/pull/49

> I have a multimeter but not sure what to test with it. 

You might try to put it on the AC setting and probe between ground and the mosfet drain connected to D2. If the transistor is not changing state, it should be a fairly stable reading of 0 volts.  You can also try the DC setting, which should jump around a bit if the transistor is working.

foreverimagining, 2024-03-03:
> @foreverimagining beat me to this

Lol, this seems to be a regular occurrence here. Still, you had good additional input that added value to the conversation. It can just take time to type everything up and a lot can happen in between. XD

Sudo-Rob, 2024-03-03:
@foreverimagining, it's a bit of a 'hail Mary.'

If you're interested, look here:  https://github.com/Kaldek/rat-ratgdo/pull/75

foreverimagining, 2024-03-03:
> @foreverimagining, it's a bit of a 'hail Mary.'
> 
> If you're interested, look here: #75

Ah, from the alternative one. I'll have to look into it more sometime. I'm always interested in learning new things, and I've pretty much jumped into electronics with 3 feet recently, so it ought to be something good to learn from.

closed 2024-03-18 by Kaldek