Kaldek/rat-ratgdo #98: Issue with new v2.5 Board and Security 1.0+

dolfelt, 2024-03-27:
I have a jackshaft opener and am having trouble getting anything to work. I'm using an updated schematic from @thayerfox and when I put the v2.5 ESP32 firmware on my board, there doesn't seem to be any status updates at all. The ESPHome interface loads, but no buttons do anything and the statuses never update.

Does anyone have any ideas on how to debug this? I used to be able to at least get the door to open and close, but with the new schematic I've got nothing. Any help is appreciated! Thank you.

thayerfox, 2024-03-27:
Are you using the ESP32 or the ESP8622 microcontroller?

rlowens, 2024-03-27:
Post the schematic you are using and picture of your wiring. Also post the logs. And what model of opener?

dolfelt, 2024-03-27:
I am using ESP32 and have a Chamberlain RJO20 with the purple button but still Security 1.0+ 

The main problem is that I'm not seeing any logs in the web UI. I can get into the interface and when I click buttons the log shows the action, but nothing else.

[image]
[image]

dolfelt, 2024-04-07:
@thayerfox @rlowens I can see the logs, but the only message that looks like an error is `Triggering sync failed actions`

From what I can tell all the GPIO pins are correct, and the schematic of my board matches what is in the repository. I might end up going back to the old board and see if that works, but it was unreliable and my garage door would open randomly, which isn't ideal. 😅 

```
[C][logger:166]: Logger:
[C][logger:167]:   Level: DEBUG
[C][logger:169]:   Log Baud Rate: 115200
[C][logger:170]:   Hardware UART: UART0
[C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[C][ratgdo.number:028]:   Type: Rolling Code Counter
[C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[C][ratgdo.number:024]:   Unit of Measurement: 's'
[C][ratgdo.number:030]:   Type: Opening Duration
[C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[C][ratgdo.number:024]:   Unit of Measurement: 's'
[C][ratgdo.number:032]:   Type: Closing Duration
[C][ratgdo.number:024]: RATGDO Number 'Client ID'
[C][ratgdo.number:026]:  Type: Client ID
[C][switch.gpio:068]: GPIO Switch 'Status door'
[C][switch.gpio:091]:   Restore Mode: always OFF
[C][switch.gpio:031]:   Pin: GPIO26
[C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[C][switch.gpio:091]:   Restore Mode: always OFF
[C][switch.gpio:031]:   Pin: GPIO5
[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[C][gpio.binary_sensor:016]:   Pin: GPIO18
[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[C][gpio.binary_sensor:016]:   Pin: GPIO19
[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[C][gpio.binary_sensor:016]:   Pin: GPIO17
[C][light:103]: Light 'Light'
[C][version.text_sensor:021]: Version Text Sensor 'Firmware Version'
[C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[C][ratgdo:082]: Setting up RATGDO...
[C][ratgdo:083]:   Output GDO Pin: GPIO22
[C][ratgdo:084]:   Input GDO Pin: GPIO21
[C][ratgdo:088]:   Input Obstruction Pin: GPIO23
[C][ratgdo_secplus1:045]:   Protocol: SEC+ v1
[C][ratgdo.lock:012]: RATGDO Lock 'Lock remotes'
[C][ratgdo.lock:013]:   Type: Lock
[C][ratgdo.binary_sensor:036]: RATGDO BinarySensor 'Motion'
[C][ratgdo.binary_sensor:036]:   Device Class: 'motion'
[C][ratgdo.binary_sensor:038]:   Type: Motion
[C][ratgdo.binary_sensor:036]: RATGDO BinarySensor 'Obstruction'
[C][ratgdo.binary_sensor:036]:   Device Class: 'problem'
[C][ratgdo.binary_sensor:040]:   Type: Obstruction
[C][ratgdo.binary_sensor:036]: RATGDO BinarySensor 'Button'
[C][ratgdo.binary_sensor:044]:   Type: Button
[C][ratgdo.cover:014]: RATGDO Cover 'Door'
[C][ratgdo.cover:014]:   Device Class: 'garage'
[C][ratgdo.light:014]: RATGDO Light
[C][restart.button:017]: Restart Button 'Restart'
[C][safe_mode.button:022]: Safe Mode Button 'Safe mode boot'
[C][safe_mode.button:022]:   Icon: 'mdi:restart-alert'
[C][homeassistant.time:010]: Home Assistant Time:
[C][homeassistant.time:011]:   Timezone: 'UTC0'
[C][web_server:169]: Web Server:
[C][web_server:170]:   Address: ratgdov25.local:80
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: ratgdov25-c8b2c0
[C][ota:096]: Over-The-Air Updates:
[C][ota:097]:   Address: ratgdov25.local:3232
[C][ota:103]:   OTA version: 2.
[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[C][api:139]: API Server:
[C][api:140]:   Address: ratgdov25.local:6053
[C][api:144]:   Using noise encryption: NO
[C][improv_serial:032]: Improv Serial:
[W][ratgdo_secplus1:059]: Triggering sync failed actions.
```

foreverimagining, 2024-04-07:
Have you tried disconnecting your wallpanel? It kind of sounds like it might be a `0x37` panel issue to me, which often prevents statuses from being read but usually still works with commands. Disconnecting your wallpanel will force it into emulation mode, and if your RAT-RATGDO starts to function perfectly, the wallpanel is probably incompatible with the current code.

> but it was unreliable and my garage door would open randomly

This is a fairly common issue with Security+1.0 GDOs. Security+1.0 is "supported" but very buggy. If it turns out to be your wallpanel, I'd recommend checking out the unfortunately-yet-to-be-released "man-in-the-middle" option in Discussion #95, which also seems to be able to fix the unintended openings. If we don't hear back from the OP one of these days, it might be a project that a few of us in the community should simply try to replicate.

dolfelt, 2024-04-07:
I just tried the old board design before https://github.com/Kaldek/rat-ratgdo/pull/50 and it seems to work. I am getting status and can open and close the door. Not sure why the new design with the additional two resistors would be so different.

dolfelt, 2024-04-12:
Any insights @thayerfox regarding the changes in #50?

trrevvorr, 2024-06-02:
I can confirm that ESP32 D1 Mini + ESPHome RATGDO v2.5 works for my Security + 1.0 garage door opener using the schematic pre-#50. I have not tested the post-#50 schematic with this setup.

100ideas, 2024-06-13:
Hey @nsayer, hey @mitchjs, ratgdo deleted some of your comments in issue number 132 in the homekit-ratgdo repo. I think because they included links to this repo (kaldek). just fyi.

nsayer, 2024-06-13:
I wound up just replacing the Liftmaster with a Genie. I would have been happy to help troubleshoot this, but this is our 2nd home and we aren’t here a lot and it needed to work *now*. Sorry.

dolfelt, 2024-06-13:
@thayerfox can you provide some more context on your #50 changes? I still haven’t been able to get them to work, and the pre-#50 still seems a bit flakey on one of my garage doors.

foreverimagining, 2024-06-13:
The difference between the [older](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png) schematic and the [newer](https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP32/RatGDO-OpenSource-D1Mini-ESP32.pdf) schematic is exactly 3 resistors, specifically R4, R5, and R6. They may or may not be absolutely necessary because a lot of people have gotten the older schematic to work, and many others have only been able to get the newer schematic to work. It's all a bit finicky and may come down to the specific components used (I'm looking at you mosfets :shakes fish:).

Are you using through-hole components or SMD? What kind of mosfets are you using? Are all of your connections solid? (because loose connections are the absolute devil and hard to diagnose)

> Is this fork or reverse engineering project not kosher with him / ratgdo?

Lol, not even slightly. No, Paul Wieland in no way sanctioned this project and actually seems to be a bit angry about it. If I'm not mistaken, Kaldek has actually been completely banned from the main repositories. 

I do find it more than a bit creepy that he can edit your comment, but the original is still there for people who want to see it (thank you, Github, for being generally excellent record keepers). Thankfully, though, Paul Wieland is not in control of the Home Assistant forum thread so I can and have linked to the RAT-RATGDO repository as I feel it is necessary or beneficial. He's not helping there anymore, anyways.

As for your problem, looks like it might be a bit tricky since you're using original-flavor RATGDO and Security+1.0. Security+1.0 is "supported" but not really truly compatible atm, so there have been a whole slew of issues cropping up with them. If things don't work out over there, maybe try making a new Discussion of it over here, and we can see if we can't muddle through a bit. I can't guarantee we'll be able to work out the problem, but maybe we'll all be able to learn something new.

Kaldek, 2024-06-13:
Hi folks, I exist!  Sorry, just a bit of a joke related to my day job consuming me all over again.  

You are right, Paul did ban me and a few others completely from his repo.  I do still feel that was excessive.  

Is it possible that some nefarious types have used the data in **this** repo to go and build + sell their own ratgdo clones?  Obviously, but they were going to do that anyway and the community still needed somewhere to help enthusiasts build their own.

dolfelt, 2024-06-14:
I'm moving to a new house in a few months, so maybe I just get new garage door openers that are truly security 2.0. I really do love the jackshaft openers, so hopefully the RJO101 openers are actually Security 2.0+ compatible. That might make my journey through all this more palatable.

Thank you @Kaldek for starting this initiative. I believe open source should truly be open source. If I wanted to buy hardware and have it supported by a company, I would (and I have), but forcing someone to buy unsupported hardware and then have open source software that only sort of supports the hardware seems disdainful at best.

closed 2024-09-05 by Kaldek