Kaldek/rat-ratgdo #99: Send command stop working until reboot

nodiaque, 2024-03-31:
Hello,

I have a chamberlain myq 2.0 with integrated wifi, installed in 2016. I've installed the rat-ratgdo this winter with an esp8386 (nodemcu). I'm currently on ratgdo 2.51 for hardware 2.50. I'm using the mqtt version since I'm using openhab that doesn't have ESPHome.

I tried to open my garage door this morning and it wasn't working. I've checked directly my nodemcu and when I send open, I see the led flash saying I received the command, but nothing happen. I did a manual open and I saw all of the information getting updated in mqtt like usual. I tried sending a close command and it didn't closed. I connected directly to the ratgdo wifi address and reboot. After the reboot is done, control is back. It's the second time this happen to me and I'm unsure how to solve this.

Thank you!

Kaldek, 2024-04-11:
Sorry mate, support is a little behind at the moment.  Did you use one of the schematics provided here?

nodiaque, 2024-04-12:
Hello, yes. But I'm on the MQTT firmware and from what I gather around, it seems to be a know issue with that firmware. I reflashed and now it's working. I hope I don't have to flash each 3 months or so.

nodiaque, 2024-04-17:
Well, after a week, it stopped working again. Tried power cycling and still no go. MQTT firmware is really a picky one

Kaldek, 2024-09-05:
That's super weird.  Unfortunately I don't use MQTT so there's little I can do to help.

closed 2024-09-05 by Kaldek